micropython/micropython #13965: micropython crashes with `print(">>> ", end='')` expression after 0.015 seconds

Kyuchumimo, 2024-03-02:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

rp2

### MicroPython version

MicroPython v1.22.2 on 2024-02-22; Raspberry Pi Pico with RP2040

### Reproduction

```py
import time

print('>>> ',end='')

while True:
    time.sleep(0.015)
    
    print("THIS SHOULD BE PRINTED!")
```

### Expected behaviour

">>> " string without new line printed on shell

### Observed behaviour

For some unknown reason, I can't print exactly what I want, which should be ">>> " without new line escape sequence (\n).

A workaround is to print ">>>  \b" (three greater-than signs followed by two spaces and a backspace escape sequence (\b)).

To avoid crashes you need to print something after the `print('>>> ',end='')` in a time span of less than 0.015 seconds, but as I said before, that is not the goal, i want to print exactly ">>> " without newline or any aditional characters, which I consider to be a bug.

### Additional Information

No, I've provided everything above.

robert-hh, 2024-03-02:
No problem here, and no reason why it should crash. If you use Thonny, it may be a Thonny problem.  Please try without Thonny. use a simple terminal emulator like Putty or screen or picocom.

Kyuchumimo, 2024-03-02:
Tested in PuTTY and it works correctly.
You are right, it is probably Thonny's problem that identifies ">>> " as an interactive prompt output and does something different with it.

closed 2024-03-02 by Kyuchumimo