micropython/micropython #14005: Different behavior on V1.22.2 over V1.19

Tangerino, 2024-03-02:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

ESP32

### MicroPython version

Using the official V1.22.2

### Reproduction

This is a very complete program using:
- Ethernet
- 4G
- 2 serial ports
- WiFI
- BLE

### Expected behaviour

Not panic attacks

### Observed behaviour

The same code used on 2K devices on V1.18 and V1.19

[image]

Now trying V1.22 (the official image) in few seconds running I get this.

But don't know where to look

Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed). 

Core  1 register dump:
PC      : 0x401ec634  PS      : 0x00060035  A0      : 0x800901c6  A1      : 0x3ffc03a0  
A2      : 0x3fffa4b8  A3      : 0x3fff057c  A4      : 0x00000006  A5      : 0x3ffc03e8  
A6      : 0x00000001  A7      : 0x3fffa59c  A8      : 0x8009a475  A9      : 0x3ffc0380  
A10     : 0x3fffa4b8  A11     : 0x3fffa3e4  A12     : 0x3fffa3ea  A13     : 0x3fffa3e4  
A14     : 0xfffffff7  A15     : 0x00060023  SAR     : 0x00000018  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x400973d4  LEND    : 0x400973f0  LCOUNT  : 0xffffffff  

Backtrace: 0x401ec631:0x3ffc03a0 0x400901c3:0x3ffc03c0 0x40083061:0x3ffc0430 0x4009c54b:0x3ffd5f20 0x4009c38e:0x3ffd5f40 0x4009f186:0x3ffd5f60 0x40091f76:0x3ffd5fe0 0x40149221:0x3ffd6020 0x40110824:0x3ffd6050 0x400ddeb6:0x3ffd6080 0x400e4481:0x3ffd60b0 0x400e4559:0x3ffd60d0 0x400ee426:0x3ffd60f0 0x400eed1d:0x3ffd6120 0x400f1cf2:0x3ffd6140 0x400f2456:0x3ffd6180 0x400f29f2:0x3ffd6210 0x400f5271:0x3ffd6270 0x400ef3e4:0x3ffd62a0 0x400ddeb6:0x3ffd62c0 0x400e4481:0x3ffd62f0 0x400e4559:0x3ffd6310 0x400edd29:0x3ffd6330 0x400ee146:0x3ffd6360 0x400ddeb6:0x3ffd6390 0x400e4481:0x3ffd63c0 0x400e4559:0x3ffd63e0 0x400861e1:0x3ffd6400 0x400de296:0x3ffd64a0 0x400e4657:0x3ffd64c0 0x400863b5:0x3ffd64f0 0x400de296:0x3ffd6590 0x400e4657:0x3ffd65b0 0x400863b5:0x3ffd65e0 0x400de296:0x3ffd6680 0x400e4657:0x3ffd66a0 0x400863b5:0x3ffd66d0 0x400de296:0x3ffd6770 0x400e4657:0x3ffd6790 0x400863b5:0x3ffd67c0 0x400de296:0x3ffd6860 0x400de2df:0x3ffd6880 0x400de346:0x3ffd68b0 0x400dde3e:0x3ffd68d0 0x400e4481:0x3ffd68f0 0x400e4559:0x3ffd6910 0x400861e1:0x3ffd6930 0x400ddf3c:0x3ffd69d0 0x400e4481:0x3ffd6a00 0x40086139:0x3ffd6a20 0x400ddf3c:0x3ffd6ac0 0x400e4481:0x3ffd6b10 0x400e4559:0x3ffd6b30 0x400861e1:0x3ffd6b50 0x400ddf3c:0x3ffd6bf0 0x400e4481:0x3ffd6c20 0x40086139:0x3ffd6c40 0x400ddf3c:0x3ffd6ce0 0x400e4481:0x3ffd6d40 0x400e4496:0x3ffd6d60 0x400f169a:0x3ffd6d80 0x400f1a85:0x3ffd6e10 0x400f1af2:0x3ffd6e30 0x400d798d:0x3ffd6e50

### Additional Information

Wifi connection issue as well, it seems something is wrong with the ESP libraries, right?

Tangerino, 2024-03-04:
My project has 25K LOC.
I did not find a way to reproduce the error, is just an observation that in V19 is all fine but in V1.2x I get this panic message.
Thank you for your time.

dpgeorge, 2024-03-06:
@Tangerino for the backtrace that you provide above, which exact firmware image did you download, that corresponds to that backtrace?

jimmo, 2024-03-07:
FYI @Tangerino, worth noting that this is a duplicate of the discussion post -- https://github.com/orgs/micropython/discussions/13738

The reason we keep asking about the exact firmware image is because that's how you decode the backtrace. What we should have asked for clarity is for the exact filename.

Assuming (because you mentioned SPIRAM) that it's `ESP32_GENERIC-SPIRAM-20240222-v1.22.2.elf`

This is the backtrace:

```
PC: 0x401ec634 is in prvGetFreeSize (/home/micropython/esp-idf-v5.0/components/esp_ringbuf/ringbuf.c:267).
BT-1: 0x400901c3 is in uart_rx_intr_handler_default (/home/micropython/esp-idf-v5.0/components/driver/uart.c:950).
BT-2: 0x40083061 is at /home/micropython/esp-idf-v5.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1118.
BT-3: 0x4009c54b is in spi_flash_hal_poll_cmd_done (/home/micropython/esp-idf-v5.0/components/hal/esp32/include/hal/spi_flash_ll.h:74).
BT-4: 0x4009c38e is in spi_flash_hal_read (/home/micropython/esp-idf-v5.0/components/hal/spi_flash_hal_common.inc:174).
BT-5: 0x4009f186 is in spi_flash_chip_generic_read (/home/micropython/esp-idf-v5.0/components/spi_flash/spi_flash_chip_generic.c:246).
BT-6: 0x40091f76 is in esp_flash_read (/home/micropython/esp-idf-v5.0/components/spi_flash/esp_flash_api.c:842).
BT-7: 0x40149221 is in esp_partition_read (/home/micropython/esp-idf-v5.0/components/esp_partition/partition_target.c:38).
BT-8: 0x40110824 is in esp32_partition_readblocks (/home/micropython/micropython-autobuild/ports/esp32/esp32_partition.c:174).
BT-9: 0x400ddeb6 is in fun_builtin_var_call (/home/micropython/micropython-autobuild/py/objfun.c:119).
BT-10: 0x400e4481 is in mp_call_function_n_kw (/home/micropython/micropython-autobuild/py/runtime.c:713).
BT-11: 0x400e4559 is in mp_call_method_n_kw (/home/micropython/micropython-autobuild/py/runtime.c:729).
BT-12: 0x400ee426 is in mp_vfs_blockdev_read_ext (/home/micropython/micropython-autobuild/extmod/vfs_blockdev.c:68).
BT-13: 0x400eed1d is in mp_vfs_lfs2_dev_read (/home/micropython/micropython-autobuild/extmod/vfs_lfsx.c:56).
BT-14: 0x400f1cf2 is in lfs2_bd_read (/home/micropython/micropython-autobuild/lib/littlefs/lfs2.c:116).
BT-15: 0x400f2456 is in lfs2_dir_fetchmatch (/home/micropython/micropython-autobuild/lib/littlefs/lfs2.c:1133).
BT-16: 0x400f29f2 is in lfs2_dir_find (/home/micropython/micropython-autobuild/lib/littlefs/lfs2.c:1516).
BT-17: 0x400f5271 is in lfs2_dir_open (/home/micropython/micropython-autobuild/lib/littlefs/lfs2.c:2664).
BT-18: 0x400ef3e4 is in mp_vfs_lfs2_ilistdir_func (/home/micropython/micropython-autobuild/extmod/vfs_lfsx.c:233).
BT-19: 0x400ddeb6 is in fun_builtin_var_call (/home/micropython/micropython-autobuild/py/objfun.c:119).
BT-20: 0x400e4481 is in mp_call_function_n_kw (/home/micropython/micropython-autobuild/py/runtime.c:713).
BT-21: 0x400e4559 is in mp_call_method_n_kw (/home/micropython/micropython-autobuild/py/runtime.c:729).
BT-22: 0x400edd29 is in mp_vfs_proxy_call (/home/micropython/micropython-autobuild/extmod/vfs.c:124).
BT-23: 0x400ee146 is in mp_vfs_ilistdir (/home/micropython/micropython-autobuild/extmod/vfs.c:431).
BT-24: 0x400ddeb6 is in fun_builtin_var_call (/home/micropython/micropython-autobuild/py/objfun.c:119).
BT-25: 0x400e4481 is in mp_call_function_n_kw (/home/micropython/micropython-autobuild/py/runtime.c:713).
BT-26: 0x400e4559 is in mp_call_method_n_kw (/home/micropython/micropython-autobuild/py/runtime.c:729).
BT-27: 0x400861e1 is in mp_execute_bytecode (/home/micropython/micropython-autobuild/py/vm.c:1042).
BT-28: 0x400de296 is in mp_obj_gen_resume (/home/micropython/micropython-autobuild/py/objgenerator.c:219).
BT-29: 0x400e4657 is in mp_resume (/home/micropython/micropython-autobuild/py/runtime.c:1408).
BT-30: 0x400863b5 is in mp_execute_bytecode (/home/micropython/micropython-autobuild/py/vm.c:1201).
BT-31: 0x400de296 is in mp_obj_gen_resume (/home/micropython/micropython-autobuild/py/objgenerator.c:219).
BT-32: 0x400e4657 is in mp_resume (/home/micropython/micropython-autobuild/py/runtime.c:1408).
BT-33: 0x400863b5 is in mp_execute_bytecode (/home/micropython/micropython-autobuild/py/vm.c:1201).
BT-34: 0x400de296 is in mp_obj_gen_resume (/home/micropython/micropython-autobuild/py/objgenerator.c:219).
BT-35: 0x400e4657 is in mp_resume (/home/micropython/micropython-autobuild/py/runtime.c:1408).
BT-36: 0x400863b5 is in mp_execute_bytecode (/home/micropython/micropython-autobuild/py/vm.c:1201).
BT-37: 0x400de296 is in mp_obj_gen_resume (/home/micropython/micropython-autobuild/py/objgenerator.c:219).
BT-38: 0x400e4657 is in mp_resume (/home/micropython/micropython-autobuild/py/runtime.c:1408).
BT-39: 0x400863b5 is in mp_execute_bytecode (/home/micropython/micropython-autobuild/py/vm.c:1201).
BT-40: 0x400de296 is in mp_obj_gen_resume (/home/micropython/micropython-autobuild/py/objgenerator.c:219).
BT-41: 0x400de2df is in gen_resume_and_raise (/home/micropython/micropython-autobuild/py/objgenerator.c:268).
BT-42: 0x400de346 is in gen_instance_send (/home/micropython/micropython-autobuild/py/objgenerator.c:295).
BT-43: 0x400dde3e is in fun_builtin_2_call (/home/micropython/micropython-autobuild/py/objfun.c:80).
BT-44: 0x400e4481 is in mp_call_function_n_kw (/home/micropython/micropython-autobuild/py/runtime.c:713).
BT-45: 0x400e4559 is in mp_call_method_n_kw (/home/micropython/micropython-autobuild/py/runtime.c:729).
BT-46: 0x400861e1 is in mp_execute_bytecode (/home/micropython/micropython-autobuild/py/vm.c:1042).
[... cut ...]
BT-59: 0x400e4496 is in mp_call_function_0 (/home/micropython/micropython-autobuild/py/runtime.c:687).
BT-60: 0x400f169a is in parse_compile_execute (/home/micropython/micropython-autobuild/shared/runtime/pyexec.c:123).
BT-61: 0x400f1a85 is in pyexec_file (/home/micropython/micropython-autobuild/shared/runtime/pyexec.c:686).
BT-62: 0x400f1af2 is in pyexec_file_if_exists (/home/micropython/micropython-autobuild/shared/runtime/pyexec.c:698).
BT-63: 0x400d798d is in mp_task (/home/micropython/micropython-autobuild/ports/esp32/main.c:145).
```

Tangerino, 2024-03-07:
> @Tangerino for the backtrace that you provide above, which exact firmware image did you download, that corresponds to that backtrace?

Official V1.22

Tangerino, 2024-03-07:
How the trace can help me?

The version is the official V1.22.2
and here is the Panic message
```
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed). 

Core  1 register dump:
PC      : 0x401ec630  PS      : 0x00060035  A0      : 0x800901c6  A1      : 0x3ffc03a0  
A2      : 0x3fffa864  A3      : 0x3fffa464  A4      : 0x00000001  A5      : 0x3ffc03e8  
A6      : 0x00000001  A7      : 0x3fffa948  A8      : 0x8009a475  A9      : 0x3ffc0380  
A10     : 0x3fffa864  A11     : 0x3fffa77a  A12     : 0x3fffa77b  A13     : 0x3fffa77a  
A14     : 0xfffffff7  A15     : 0x00060023  SAR     : 0x00000018  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x40097370  LEND    : 0x40097383  LCOUNT  : 0x00000000  

Backtrace: 0x401ec62d:0x3ffc03a0 0x400901c3:0x3ffc03c0 0x40083061:0x3ffc0430 0x4009c546:0x3ffd5f50 0x4009c38e:0x3ffd5f70 0x4009f186:0x3ffd5f90 0x40091f76:0x3ffd6010 0x4014921d:0x3ffd6050 0x40110824:0x3ffd6080 0x400ddeb6:0x3ffd60b0 0x400e4481:0x3ffd60e0 0x400e4559:0x3ffd6100 0x400ee426:0x3ffd6120 0x400eed1d:0x3ffd6150 0x400f1cf2:0x3ffd6170 0x400f2380:0x3ffd61b0 0x400f29f2:0x3ffd6240 0x400f5271:0x3ffd62a0 0x400ef3e4:0x3ffd62d0 0x400ddeb6:0x3ffd62f0 0x400e4481:0x3ffd6320 0x400e4559:0x3ffd6340 0x400edd29:0x3ffd6360 0x400ee146:0x3ffd6390 0x400ddeb6:0x3ffd63c0 0x400e4481:0x3ffd63f0 0x400e4559:0x3ffd6410 0x400861e1:0x3ffd6430 0x400de296:0x3ffd64d0 0x400e4657:0x3ffd64f0 0x400863b5:0x3ffd6520 0x400de296:0x3ffd65c0 0x400e4657:0x3ffd65e0 0x400863b5:0x3ffd6610 0x400de296:0x3ffd66b0 0x400e4657:0x3ffd66d0 0x400863b5:0x3ffd6700 0x400de296:0x3ffd67a0 0x400e4657:0x3ffd67c0 0x400863b5:0x3ffd67f0 0x400de296:0x3ffd6890 0x400de2df:0x3ffd68b0 0x400de346:0x3ffd68e0 0x400dde3e:0x3ffd6900 0x400e4481:0x3ffd6920 0x400e4559:0x3ffd6940 0x400861e1:0x3ffd6960 0x400ddf3c:0x3ffd6a00 0x400e4481:0x3ffd6a30 0x40086139:0x3ffd6a50 0x400ddf3c:0x3ffd6af0 0x400e4481:0x3ffd6b40 0x400e4559:0x3ffd6b60 0x400861e1:0x3ffd6b80 0x400ddf3c:0x3ffd6c20 0x400e4481:0x3ffd6c50 0x40086139:0x3ffd6c70 0x400ddf3c:0x3ffd6d10 0x400e4481:0x3ffd6d70 0x400e4496:0x3ffd6d90 0x400f169a:0x3ffd6db0 0x400f1a85:0x3ffd6e40 0x400f1af2:0x3ffd6e60 0x400d798d:0x3ffd6e80

ELF file SHA256: 83ca933c29f0205a

Rebooting...
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4728
load:0x40078000,len:14888
load:0x40080400,len:3368
entry 0x400805cc
```

Tangerino, 2024-03-07:
Let's reset the thread.

I downloaded the file `ESP32_GENERIC-SPIRAM-20240222-v1.22.2.bin`  from the MP web site

`MicroPython v1.22.2 on 2024-02-22; Generic ESP32 module with SPIRAM with ESP32`

I also downloaded the .map file but I did not find the address on it

```
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed). 

Core  1 register dump:
PC      : 0x401ec634  PS      : 0x00060035  A0      : 0x800901c6  A1      : 0x3ffc03a0  
A2      : 0x3fffb10c  A3      : 0x3fffae10  A4      : 0x00000001  A5      : 0x3ffc03e8  
A6      : 0x00000001  A7      : 0x3fffb1f0  A8      : 0x8009a475  A9      : 0x3ffc0380  
A10     : 0x3fffb10c  A11     : 0x3fffb03a  A12     : 0x3fffb03b  A13     : 0x3fffb03a  
A14     : 0x007be740  A15     : 0x00060023  SAR     : 0x0000001d  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x40097370  LEND    : 0x40097383  LCOUNT  : 0x00000000  

Backtrace: 0x401ec631:0x3ffc03a0 0x400901c3:0x3ffc03c0 0x40083061:0x3ffc0430 0x400826f1:0x3ffd5ef0 0x4009ba19:0x3ffd5f10 0x40091359:0x3ffd5f30 0x40091418:0x3ffd5f50 0x400923e7:0x3ffd5f80 0x400923ef:0x3ffd5fa0 0x400965c1:0x3ffd5fc0 0x40091f5a:0x3ffd5fe0 0x40149221:0x3ffd6020 0x40110824:0x3ffd6050 0x400ddeb6:0x3ffd6080 0x400e4481:0x3ffd60b0 0x400e4559:0x3ffd60d0 0x400ee426:0x3ffd60f0 0x400eed1d:0x3ffd6120 0x400f1cf2:0x3ffd6140 0x400f2380:0x3ffd6180 0x400f29f2:0x3ffd6210 0x400f5271:0x3ffd6270 0x400ef3e4:0x3ffd62a0 0x400ddeb6:0x3ffd62c0 0x400e4481:0x3ffd62f0 0x400e4559:0x3ffd6310 0x400edd29:0x3ffd6330 0x400ee146:0x3ffd6360 0x400ddeb6:0x3ffd6390 0x400e4481:0x3ffd63c0 0x400e4559:0x3ffd63e0 0x400861e1:0x3ffd6400 0x400de296:0x3ffd64a0 0x400e4657:0x3ffd64c0 0x400863b5:0x3ffd64f0 0x400de296:0x3ffd6590 0x400e4657:0x3ffd65b0 0x400863b5:0x3ffd65e0 0x400de296:0x3ffd6680 0x400e4657:0x3ffd66a0 0x400863b5:0x3ffd66d0 0x400de296:0x3ffd6770 0x400e4657:0x3ffd6790 0x400863b5:0x3ffd67c0 0x400de296:0x3ffd6860 0x400de2df:0x3ffd6880 0x400de346:0x3ffd68b0 0x400dde3e:0x3ffd68d0 0x400e4481:0x3ffd68f0 0x400e4559:0x3ffd6910 0x400861e1:0x3ffd6930 0x400ddf3c:0x3ffd69d0 0x400e4481:0x3ffd6a00 0x40086139:0x3ffd6a20 0x400ddf3c:0x3ffd6ac0 0x400e4481:0x3ffd6b10 0x400e4559:0x3ffd6b30 0x400861e1:0x3ffd6b50 0x400ddf3c:0x3ffd6bf0 0x400e4481:0x3ffd6c20 0x40086139:0x3ffd6c40 0x400ddf3c:0x3ffd6ce0 0x400e4481:0x3ffd6d40 0x400e4496:0x3ffd6d60 0x400f169a:0x3ffd6d80 0x400f1a85:0x3ffd6e10 0x400f1af2:0x3ffd6e30 0x400d798d:0x3ffd6e50

ELF file SHA256: 47e9095431738ab1

Rebooting...
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4728
load:0x40078000,len:14888
load:0x40080400,len:3368
entry 0x400805cc```

dpgeorge, 2024-03-07:
@Tangerino can you share some information about the code that is running?  Without that it's really hard to tell what the issue is.

From the backtrace my guess is that you are using `asyncio` and doing an `ilistdir()` of the filesystem in a nested chain of async functions.  And during the listing of the VfsLfs2 filesystem a UART RX interrupt comes in, at which point things fail.  What component in your system is using UART?

Tangerino, 2024-03-08:
Thank you @dpgeorge, you got it right.

I would like to know how do you read this information so I can try to figure out myself next time.

Indeed, I left the application running over 9 hours without any problem communicating to devices over modbus tcp (over lan).
Once I configured a modbus rtu (serial) device the board had a panic attack at the same address (0x401ec634).

The difference between the TCP and RTU, in the code, is just the communication layer, a small part so I hope to be able to pinpoint the problem. So something is wrong with the serial part.

I'll try to disable parts of the application and focus on the serial part.
I'll update you later.
For now, thank you very much.

p.s. Sometimes I miss some **application notes** telling users how to's, more complete examples, etc, a knowledge base (such as how to parse a map file).

And specially new features or breaking changes for new releases
You may say this is a pre-requisite for professionals working here, but it would be very helpful.

Cheers

Tangerino, 2024-03-09:
Because I saw https://github.com/micropython/micropython/issues/13428 I have created a small code to generate the Panic message.

In fact, after adding the ram_disk, it just happen when I interrupt the program, so I'm not sure if it counts.

And I'll keep trying. Not even sure if the serial is the bad guy but in my project, the serial communication generates data that is sent to the disk

Adding the code and the log just after
```python
import asyncio
import os
from time import ticks_ms

from machine import UART

class RAMBlockDev:
    def __init__(self, block_size, num_blocks):
        self.block_size = block_size
        self.data = bytearray(block_size * num_blocks)

    def readblocks(self, block_num, buf, offset=0):
        addr = block_num * self.block_size + offset
        for i in range(len(buf)):
            buf[i] = self.data[addr + i]

    def writeblocks(self, block_num, buf, offset=None):
        if offset is None:
            # do erase, then write
            for i in range(len(buf) // self.block_size):
                self.ioctl(6, block_num + i)
            offset = 0
        addr = block_num * self.block_size + offset
        for i in range(len(buf)):
            self.data[addr + i] = buf[i]

    def ioctl(self, op, arg):
        if op == 4:  # block count
            return len(self.data) // self.block_size
        if op == 5:  # block size
            return self.block_size
        if op == 6:  # block erase
            return 0

def build_ram_disk(block_size, num_blocks) -> bool:
    if hasattr(os, "VfsLfs2"):
        bdev = RAMBlockDev(block_size, num_blocks)
        os.VfsLfs2.mkfs(bdev)
        os.mount(bdev, '/ram')
        return True
    return False

def open_uart():
    return UART(1, 110, timeout_char=50, tx=20, rx=37)

async def async_read_frames(uart, msg_queue: list):
    buffer = bytearray(1024)
    mv = memoryview(buffer)
    while True:
        n = uart.any()
        if n >= 8:
            bytes_read = uart.readinto(mv)
            if bytes_read is None:
                print('Error reading UART - Timeout')
                break
            if bytes_read != n:
                print(f'Error reading UART - Incomplete frame. Expected {n} bytes, got {bytes_read}')
                break
            msg_queue.append(bytes(buffer[:bytes_read]))
        await asyncio.sleep(0.01)

def save_frame_to_ram_disk(frame: bytes):
    fn = f'/ram/frame_{ticks_ms()}.bin'
    with open(fn, 'wb') as f:
        f.write(frame)

async def async_handle_frames(msg_queue: list):
    while True:
        if len(msg_queue):
            frame = msg_queue.pop(0)
            print(f'Got {len(frame)} bytes: {list(frame)}')
            save_frame_to_ram_disk(frame)
        await asyncio.sleep(0.01)

async def disk_reader():
    while True:
        for fn in os.listdir('/ram'):
            with open(f'/ram/{fn}', 'rb') as f:
                print(f'File {fn} has {len(f.read())} bytes')
            os.remove(f'/ram/{fn}')
        await asyncio.sleep(5)

def main():
    loop = asyncio.get_event_loop()
    uart = open_uart()
    if not build_ram_disk(1024, 64):
        print('Error creating RAM disk')
        return
    msg_queue = []
    loop.create_task(async_read_frames(uart, msg_queue))
    loop.create_task(async_handle_frames(msg_queue))
    loop.create_task(disk_reader())
    loop.run_forever()

```

The log
```

MPY: soft reboot
MicroPython v1.22.2 on 2024-02-22; Generic ESP32 module with SPIRAM with ESP32
Type "help()" for more information.
>>> import main
>>> main.main()
Got 8 bytes: [201, 2, 130, 17, 0, 13, 0, 170]
File frame_21497.bin has 8 bytes

...
... log removed
...
Got 8 bytes: [153, 2, 130, 21, 0, 16, 0, 227]
File frame_69856.bin has 8 bytes
Got 8 bytes: [152, 2, 130, 21, 0, 16, 0, 226]
File frame_70856.bin has 8 bytes

────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "main.py", line 101, in main
  File "asyncio/core.py", line 1, in run_forever
  File "asyncio/core.py", line 1, in run_until_complete
  File "asyncio/core.py", line 1, in wait_io_event
KeyboardInterrupt: 

MPY: soft reboot
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed). 

Core  1 register dump:
PC      : 0x401ec634  PS      : 0x00060035  A0      : 0x800901c6  A1      : 0x3ffc03a0  
A2      : 0x3ffbd288  A3      : 0x3ffbb99c  A4      : 0x00000001  A5      : 0x3ffc03e8  
A6      : 0x00000001  A7      : 0x3ffbd36c  A8      : 0x8009a475  A9      : 0x3ffc0380  
A10     : 0x3ffbd288  A11     : 0x3ffbd214  A12     : 0x3ffbd21b  A13     : 0x3ffbd21a  
A14     : 0x00000010  A15     : 0x00060023  SAR     : 0x00000019  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x40097370  LEND    : 0x40097383  LCOUNT  : 0x00000000  

Backtrace: 0x401ec631:0x3ffc03a0 0x400901c3:0x3ffc03c0 0x40083061:0x3ffc0430 0x4009c606:0x3ffd6980 0x400965ae:0x3ffd69a0 0x40091f5a:0x3ffd69c0 0x40149221:0x3ffd6a00 0x40110824:0x3ffd6a30 0x400ddeb6:0x3ffd6a60 0x400e4481:0x3ffd6a90 0x400e4559:0x3ffd6ab0 0x400ee426:0x3ffd6ad0 0x400eed1d:0x3ffd6b00 0x400f1cf2:0x3ffd6b20 0x400f1d6b:0x3ffd6b60 0x400f1e36:0x3ffd6bb0 0x400f1ee8:0x3ffd6be0 0x400f4dbf:0x3ffd6c10 0x400ef435:0x3ffd6c60 0x400ee2f6:0x3ffd6da0 0x400f1ae7:0x3ffd6e30 0x400d7932:0x3ffd6e50

ELF file SHA256: 47e9095431738ab1

Rebooting...
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4728
load:0x40078000,len:14888
load:0x40080400,len:3368
entry 0x400805cc
MicroPython v1.22.2 on 2024-02-22; Generic ESP32 module with SPIRAM with ESP32
Type "help()" for more information.
>>> 

```

Tangerino, 2024-03-09:
A panic in a different situation, after stopping the program and importing the main module
```
9, 2]
Got 8 bytes: [130, 24, 0, 18, 0, 244, 128, 2]
Got 8 bytes: [130, 24, 0, 18, 0, 245, 127, 2]
Got 8 bytes: [130, 24, 0, 18, 0, 10, 126, 2]
Got 8 bytes: [130, 24, 0, 18, 0, 11, 125, 2]
File /ram/frame_88386.bin has 8 bytes
File /ram/frame_89396.bin has 8 bytes
File /ram/frame_90406.bin has 8 bytes
File /ram/frame_91406.bin has 8 bytes
File /ram/frame_92407.bin has 8 bytes
Got 8 bytes: [130, 24, 0, 18, 0, 8, 124, 2]
Got 8 bytes: [130, 24, 0, 18, 0, 9, 123, 2]
Got 8 bytes: [130, 24, 0, 18, 0, 14, 122, 2]
Got 8 bytes: [130, 24, 0, 18, 0, 15, 121, 2]

────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "main.py", line 102, in main
  File "asyncio/core.py", line 1, in run_forever
  File "asyncio/core.py", line 1, in run_until_complete
KeyboardInterrupt: 

MPY: soft reboot
MicroPython v1.22.2 on 2024-02-22; Generic ESP32 module with SPIRAM with ESP32
Type "help()" for more information.
>>> import main
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed). 

Core  1 register dump:
PC      : 0x401ec634  PS      : 0x00060035  A0      : 0x800901c6  A1      : 0x3ffc03a0  
A2      : 0x3ffbd288  A3      : 0x3ffbb99c  A4      : 0x00000001  A5      : 0x3ffc03e8  
A6      : 0x00000001  A7      : 0x3ffbd36c  A8      : 0x8009a475  A9      : 0x3ffc0380  
A10     : 0x3ffbd288  A11     : 0x3ffbd18d  A12     : 0x3ffbd1ca  A13     : 0x3ffbd1c9  
A14     : 0xc0000020  A15     : 0x00060023  SAR     : 0x0000001a  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x40097370  LEND    : 0x40097383  LCOUNT  : 0x00000000  

Backtrace: 0x401ec631:0x3ffc03a0 0x400901c3:0x3ffc03c0 0x40083061:0x3ffc0430 0x4009f19a:0x3ffd62b0 0x40091f76:0x3ffd6330 0x40149221:0x3ffd6370 0x40110824:0x3ffd63a0 0x400ddeb6:0x3ffd63d0 0x400e4481:0x3ffd6400 0x400e4559:0x3ffd6420 0x400ee426:0x3ffd6440 0x400eed1d:0x3ffd6470 0x400f1cf2:0x3ffd6490 0x400f2456:0x3ffd64d0 0x400f29f2:0x3ffd6560 0x400f4db0:0x3ffd65c0 0x400ef435:0x3ffd6610 0x400ee2f6:0x3ffd6750 0x400ffdd4:0x3ffd67e0 0x400ffde4:0x3ffd6800 0x400ffe29:0x3ffd6820 0x40100155:0x3ffd6840 0x400e45d6:0x3ffd6930 0x40086479:0x3ffd6970 0x400ddf3c:0x3ffd6a10 0x400e4481:0x3ffd6a60 0x400e4496:0x3ffd6a80 0x400e44f5:0x3ffd6aa0 0x400fff19:0x3ffd6ae0 0x40100237:0x3ffd6b10 0x400e45d6:0x3ffd6c00 0x40086479:0x3ffd6c40 0x400ddf3c:0x3ffd6ce0 0x400e4481:0x3ffd6d30 0x400e4496:0x3ffd6d50 0x400f169a:0x3ffd6d70 0x400f18d2:0x3ffd6e00 0x400d79a6:0x3ffd6e50

ELF file SHA256: 47e9095431738ab1

Rebooting...
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4728
load:0x40078000,len:14888
load:0x40080400,len:3368
entry 0x400805cc
MicroPython v1.22.2 on 2024-02-22; Generic ESP32 module with SPIRAM with ESP32
Type "help()" for more information.
>>> 
```

jimmo, 2024-03-14:
@Tangerino 

> I would like to know how do you read this information so I can try to figure out myself next time.

The crash dump you provided gives a list of addresses. With the .elf or the .map you can turn that into the original function names. This is why we wanted to know the exact version.

The IDF's `monitor` tool can do the address to name mapping automatically, otherwise e.g. https://github.com/tve/esp32-backtrace

> From the backtrace my guess is that you are using asyncio and doing an ilistdir() of the filesystem in a nested chain of async functions. And during the listing of the VfsLfs2 filesystem a UART RX interrupt comes in, at which point things fail. What component in your system is using UART?

Inferring this from the backtrace:
* It's likely asyncio due to the amount of generators involved (e.g. mp_obj_gen_resume)
* ilistdir (and the corresponding lfs functions) are listed there in the backtrace
* listdir must talk to flash (and you can see the flash access functions)
* It's a UART IRQ because (a) there's no reason for a UART function to be called in the middle of doing a flash operation, and (b) because the function is called `uart_rx_intr_handler_default`.

The next detail is the actual crash reason: `Cache disabled but cached memory region accessed`
* During the flash operation, the CPU cannot access code (because the code is on the same flash chip)
* All functions that operate while flash operations are in progress must therefore be in RAM (`iram`)
* Therefore `prvGetFreeSize` (part of the UART ring buffer implementation) must not be in RAM

And indeed, you can see that, `prvGetFreeSize`'s address is 0x401ec634 which is not in `iram`, it's in the main flash text region.

But this should not be the case. In our config we set `CONFIG_UART_ISR_IN_IRAM`, and although we also set `CONFIG_RINGBUF_PLACE_FUNCTIONS_INTO_FLASH`, we do not set `CONFIG_RINGBUF_PLACE_ISR_FUNCTIONS_INTO_FLASH` which is what the IDF deems to be the necessary functions required for the UART ISRs to be in RAM.

However, the IDF is incorrect here, because prvGetFreeSize (called by the UART ISR) is considered by the ringbuf linker script to be a "function" not an "ISR function", and therefore doesn't get placed in RAM.

i.e. this line https://github.com/espressif/esp-idf/blob/v5.0.4/components/esp_ringbuf/ringbuf.c#L1076 does not match https://github.com/espressif/esp-idf/blob/v5.0.4/components/driver/Kconfig#L212

I will raise an IDF bug. As a temporary workaround we can disable `CONFIG_RINGBUF_PLACE_FUNCTIONS_INTO_FLASH` although as per https://github.com/micropython/micropython/blob/master/ports/esp32/boards/sdkconfig.spiram#L12 it might not be possible as we will run out of `iram`.

jimmo, 2024-03-14:
Raised https://github.com/espressif/esp-idf/issues/13378

> As a temporary workaround we can disable `CONFIG_RINGBUF_PLACE_FUNCTIONS_INTO_FLASH` although as per https://github.com/micropython/micropython/blob/master/ports/esp32/boards/sdkconfig.spiram#L12 it might not be possible as we will run out of `iram`.

Unfortunately this is the case... Either we will need to move to a later IDF, wait for a fix from Espressif, or find a way to force these functions into IRAM from our CMake config (e.g. a custom linker script?).

Tangerino, 2024-03-19:
Very impressive

jimmo, 2024-03-21:
Espressif replied to the issue above, no indication yet as to whether there might be a fix.

In the meantime, here's an attempt at a workaround on our side: https://github.com/micropython/micropython/pull/14145

dpgeorge, 2024-03-22:
Fixed by workaround added in acbdbcd95e6d953b08f0642a844c1808323e0b7c

closed 2024-03-22 by dpgeorge

Tangerino, 2024-03-22:
Is this available already for download?
I downloaded the file `v1.23.0-preview.253.g71044a418` and got this error

```
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed). 

Core  1 register dump:
PC      : 0x401ebf18  PS      : 0x00060035  A0      : 0x800901be  A1      : 0x3ffc03c0  
A2      : 0x3fffa6b4  A3      : 0x3fff9ef8  A4      : 0x00000006  A5      : 0x3ffc0408  
A6      : 0x00000001  A7      : 0x3fffa798  A8      : 0x8009a435  A9      : 0x3ffc03a0  
A10     : 0x3fffa6b4  A11     : 0x3fffa5bb  A12     : 0x3fffa5c1  A13     : 0x3fffa5bb  
A14     : 0xfffffff7  A15     : 0x00060023  SAR     : 0x00000018  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x40097330  LEND    : 0x40097343  LCOUNT  : 0x00000000  

Backtrace: 0x401ebf15:0x3ffc03c0 0x400901bb:0x3ffc03e0 0x40083059:0x3ffc0450 0x4009c506:0x3ffd5f40 0x4009c34e:0x3ffd5f60 0x4009f146:0x3ffd5f80 0x40091f16:0x3ffd6000 0x40148f4d:0x3ffd6040 0x401109c8:0x3ffd6070 0x400ddf66:0x3ffd60a0 0x400e4501:0x3ffd60d0 0x400e45d9:0x3ffd60f0 0x400ee3ce:0x3ffd6110 0x400eecc1:0x3ffd6140 0x400f1cb6:0x3ffd6160 0x400f241a:0x3ffd61a0 0x400f29b6:0x3ffd6230 0x400f5235:0x3ffd6290 0x400ef388:0x3ffd62c0 0x400ddf66:0x3ffd62e0 0x400e4501:0x3ffd6310 0x400e45d9:0x3ffd6330 0x400edcd1:0x3ffd6350 0x400ee0ee:0x3ffd6380 0x400ddf66:0x3ffd63b0 0x400e4501:0x3ffd63e0 0x400e45d9:0x3ffd6400 0x400861d9:0x3ffd6420 0x400de31a:0x3ffd64c0 0x400e46d7:0x3ffd64e0 0x400863ad:0x3ffd6510 0x400de31a:0x3ffd65b0 0x400e46d7:0x3ffd65d0 0x400863ad:0x3ffd6600 0x400de31a:0x3ffd66a0 0x400e46d7:0x3ffd66c0 0x400863ad:0x3ffd66f0 0x400de31a:0x3ffd6790 0x400e46d7:0x3ffd67b0 0x400863ad:0x3ffd67e0 0x400de31a:0x3ffd6880 0x400de363:0x3ffd68a0 0x400de3ca:0x3ffd68d0 0x400dded2:0x3ffd68f0 0x400e4501:0x3ffd6910 0x400e45d9:0x3ffd6930 0x400861d9:0x3ffd6950 0x400ddfec:0x3ffd69f0 0x400e4501:0x3ffd6a20 0x40086131:0x3ffd6a40 0x400ddfec:0x3ffd6ae0 0x400e4501:0x3ffd6b30 0x400e45d9:0x3ffd6b50 0x400861d9:0x3ffd6b70 0x400ddfec:0x3ffd6c10 0x400e4501:0x3ffd6c40 0x40086131:0x3ffd6c60 0x400ddfec:0x3ffd6d00 0x400e4501:0x3ffd6d60 0x400e4516:0x3ffd6d80 0x400f165e:0x3ffd6da0 0x400f1a49:0x3ffd6e30 0x400f1ab6:0x3ffd6e50 0x400d798d:0x3ffd6e70

ELF file SHA256: 5835da01fbfa2b10

Rebooting...
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4728
load:0x40078000,len:14888
load:0x40080400,len:3368
entry 0x400805cc
```

dpgeorge, 2024-03-23:
You need a more recent build.  Try `v1.23.0-preview.273.g35b2edfc2`.

zcattacz, 2024-04-11:
Hi @Tangerino, just curious, has this issue been resolved? Is the immigration going smooth ?