micropython/micropython #14013: Ethernet fails on the Olimex ESP32-POE-ISO Revision L

PhantomPhoton, 2024-03-03:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

ESP32-POE-ISO Revision L

### MicroPython version

MicroPython v1.22.2 on 2024-02-22; Olimex ESP32 ETH with ESP32

### Reproduction

lan = network.LAN(mdc=machine.Pin(23), mdio=machine.Pin(18), power=machine.Pin(12),
                  phy_type=network.PHY_LAN8720, phy_addr=0,
                  ref_clk=machine.Pin(17), ref_clk_mode=machine.Pin.OUT)
lan.active(True)

### Expected behaviour

Expected ethernet to enable

### Observed behaviour

OSError: ethernet enable failed

### Additional Information

I've pulled the expected working code from https://docs.micropython.org/en/latest/esp32/quickref.html#lan

I know the Ethernet on the device works, when i flash it with a example program from arduino ide, it gets an ip and works correctly.

robert-hh, 2024-03-03:
It works here if I use a slightly reduced call:
```
import machine, network, time
lan=network.LAN(mdc=machine.Pin(23), mdio=machine.Pin(18), power=machine.Pin(12),
    phy_type=network.PHY_LAN8720, phy_addr=0)

lan.active(True)
```
Adding the settings for ref_clk_mode causes an error. That's stumping, because it used to work. Maybe related to  ESP IDF version change.
Note: When it failed once, the board must be reset with a power cycle.

robert-hh, 2024-03-03:
Strange. After have the board powered for a while, even your variant with specifying ref_clk works. That has to be further analyzed.

robert-hh, 2024-03-03:
There is a strange behaviour at the MDC line of LAN8720. At the first MDIO message after Reset, the level of the clock signal slowly ramps up. Frequency and rise/fall times are not affected.  Further messages are fine. This behavior can be repeated several times. So it's not that the probe was not on spot
First message MDC
[image]

Further messages MDC

[image]

robert-hh, 2024-03-03:
The ramping seems to be caused by the power mechanism. Reset of the LAN8720 is done be switching the power off/on. After enabling it takes some µs before the power is up. If the MDIO communication happens during that time, the MDIO pins have a low impedance and the supplied level is low. Switching GPIO12 to 1 before enabling the LAN seems to improve the behavior.

@PhantomPhoton:
You could try to use the following start-up code:
```
import machine, network, time
machine.Pin(12, machine.Pin.OUT, value=1)  # manually enable power
time.sleep_ms(1)
lan=network.LAN(mdc=machine.Pin(23), mdio=machine.Pin(18),
    phy_type=network.PHY_LAN8720, phy_addr=0)

lan.active(True)
```

robert-hh, 2024-03-03:
May I ask you to try the attached firmware with your board. It shows better power up ramping for the LAN8720.

[micropython.zip](https://github.com/micropython/micropython/files/14472752/micropython.zip)

PhantomPhoton, 2024-03-03:
The micropython.zip firmware just enters into a never ending loop of

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3f400020,len:287180
ets Jul 29 2019 12:21:46
```

I can't seem to get to a repl prompt to test anything with it.

That said, the startup code you gave me does work with the v1.22.2 firmware.

Thank you so very much for the quick and detailed investigation of the issue. I really appreciate your time and effort.

robert-hh, 2024-03-03:
Thank you for the feedback. The firmware works here, but when uploaded as part of the build process. So I may have sent you the wrong file. But if the python startup code works, then this is also a valuable information. The changed firmware just added the power Pin declaration and delay into the C code.

PhantomPhoton, 2024-03-03:
Thank you for the updated firmware. 

firmware.zip works with my initial code example.

robert-hh, 2024-03-03:
OK. Thanks for testing.

dpgeorge, 2024-03-08:
Fixed by cdfc6c159f6b51f67fd314a1dd4a61d5c18d5d93

closed 2024-03-08 by dpgeorge