micropython/micropython #14039: RP2: Can not build against XTAL value of 12.8MHz - solution

mungewell, 2024-03-06:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

rp2

### MicroPython version

```
commit 90e517862d055b8ec95a0c7067052c0ba6cd0df1 (HEAD -> master, origin/master, origin/HEAD)
Author: Damien George <damien@micropython.org>
Date:   Mon Mar 4 11:33:00 2024 +1100
```

### Reproduction

RP2 Build fails as `XOSC_MHZ` can not be float (ie 12.8).

Solution is:
```
diff --git a/lib/pico-sdk b/lib/pico-sdk
--- a/lib/pico-sdk
+++ b/lib/pico-sdk
@@ -1 +1 @@
-Subproject commit 6a7db34ff63345a7badec79ebea3aaef1712f374
+Subproject commit 6a7db34ff63345a7badec79ebea3aaef1712f374-dirty
diff --git a/ports/rp2/modmachine.c b/ports/rp2/modmachine.c
index 70b22dac0..e3209bdc7 100644
--- a/ports/rp2/modmachine.c
+++ b/ports/rp2/modmachine.c
@@ -125,7 +125,11 @@ STATIC void mp_machine_lightsleep(size_t n_args, const mp_obj_t *args) {
         }
     }
 
+    #ifdef XOSC_KHZ
+    const uint32_t xosc_hz = XOSC_KHZ * KHZ;
+    #else
     const uint32_t xosc_hz = XOSC_MHZ * 1000000;
+    #endif
 
     uint32_t my_interrupts = save_and_disable_interrupts();
     #if MICROPY_PY_NETWORK_CYW43
```

### Expected behaviour

Successful build.

### Observed behaviour

A build failure... :-(

### Additional Information

I want to use this XTAL as they (at least the one I found) are higher precision that the stock/passive XTAL on the Pico.

I also had to make the following changes in the pico-sdk to get the XTAL to work.
```
diff --git a/src/rp2040/hardware_regs/include/hardware/platform_defs.h b/src/rp2040/hardware_regs/include/hardware/platform_defs.h
index 5d635ae..40f41f7 100644
--- a/src/rp2040/hardware_regs/include/hardware/platform_defs.h
+++ b/src/rp2040/hardware_regs/include/hardware/platform_defs.h
@@ -43,12 +43,22 @@
 // override the defaults which live in src/rp2_common/hardware_clocks/include/hardware/clocks.h
 // Please see the comments there about calculating the new PLL setting values.
 #ifndef XOSC_KHZ
-#define XOSC_KHZ _u(12000)
+//#define XOSC_KHZ _u(12000)
+#define XOSC_KHZ _u(12800)
+
+#define PLL_SYS_VCO_FREQ_KHZ                (960 * KHZ)
+#define PLL_SYS_POSTDIV1                    4
+#define PLL_SYS_POSTDIV2                    2
+
+#define PLL_USB_VCO_FREQ_KHZ                (1152 * KHZ)
+#define PLL_USB_POSTDIV1                    6
+#define PLL_USB_POSTDIV2                    4
 #endif
 
 // PICO_CONFIG: SYS_CLK_KHZ, The system operating frequency in kHz, type=int, default=125000, advanced=true, group=hardware_base
 #ifndef SYS_CLK_KHZ
-#define SYS_CLK_KHZ _u(125000)
+//#define SYS_CLK_KHZ _u(125000)
+#define SYS_CLK_KHZ _u(120000)
 #endif
 
 // PICO_CONFIG: USB_CLK_KHZ, USB clock frequency. Must be 48MHz for the USB interface to operate correctly, type=int, default=48000, advanced=true, group=hardware_base
```

dpgeorge, 2024-03-06:
I'm happy to change `XOSC_MHZ` to `XOSC_KHZ`, that is worth doing.

But we can't change the pico-sdk to support different crystals.  To get a solution that doesn't need changes to the pico-sdk you could try something like this in CMakeLists.txt:
```
target_compile_definitions(${MICROPY_TARGET} PRIVATE XOSC_KHZ=12800)
```
Note you'll also need to change the PLL settings to get the other clocks working on the RP2040.

mungewell, 2024-03-06:
Great.

Totally agree that the pico-sdk is their own thing, adding the specific settings for the PLLs works for me as-is.