micropython/micropython #14093: Unsigned conversion overflow for QSTRs during build

ironss-iotec, 2024-03-14:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

Applies to all ports and boards

### MicroPython version

Micropython v1.22

### Reproduction

* Add a new ext_mod with lots of QSTRs

### Expected behaviour

Build succeeds

### Observed behaviour

Build creates the following error messages

    micropython-v1.22.2/py/scope.c:36:22: warning: unsigned conversion from 'int' to 'unsigned char' changes value from '256' to '0' [-Woverflow]
    36 |     [SCOPE_LAMBDA] = MP_QSTR__lt_lambda_gt_,
    micropython-v1.22.2/py/scope.c:46:5: note: in expansion of macro 'MP_STATIC_ASSERT'
    46 |     MP_STATIC_ASSERT(MP_QSTR__lt_lambda_gt_ <= UINT8_MAX);
      |     ^~~~~~~~~~~~~~~~

and a few similar errors.

### Additional Information

In `py/scope.c`, a comment says that "these low numbered qstrs should fit in 8 bits.  See assertions below." However, during the QSTR generation, there is nothing to force this.

It seems that there is a way to do this in `py/makeqstrdefs.py`: there is a list of operators `operator_qstr_list` that must have low values.

My solution was to add these values to this list.

    "<module>",
    "<lambda>",
    "<listcomp>",
    "<dictcomp>",
    "<setcomp>",
    "<genexpr>",

However, the name of the list and the comment above it are now incorrect, as they refer to 'operators'. 

Please indicate whether it would be better to create _another_ list with additional QSTRs to keep below 255, and add another list to the test around line 410 in `print_qstr_data`, changing

    pool = 0 if qstr in operator_qstr_list else 1

to something like

    pool = 0 is qstr in operator_qstr_list or qstr in more_low_qstr_list else 1

ironss-iotec, 2024-03-17:
@jimmo I don't have a convenient setup to do a PR. Can you process a patch file? It does the things you mentioned

* changes name of the list from `operator_qstr_list` to `unsorted_qstr_list`
* adds those qstrs to the list
* adds or changes comments in `py/makeqstrdata.py`, `py/objtype.c`, `py/qstr.c` and `py/scope.c`.

[makeqstr.patch.txt](https://github.com/micropython/micropython/files/14625678/makeqstr.patch.txt)

jimmo, 2024-03-17:
@ironss-iotec Yep no worries, thanks for the patch file.

jimmo, 2024-03-26:
PR here: https://github.com/micropython/micropython/pull/14180

PR #14180 (2024-03-26): py/makeqstrdata.py: Ensure that scope names get low qstr values.
Originally implemented in a patch file provided by @ironss-iotec.

Fixes #14093
closed 2024-03-26 by dpgeorge