micropython/micropython #14097: rp2: code sometimes hangs when doing `bluetooth.BLE().active(True)`

dpgeorge, 2024-03-15:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

rp2 port with RPI_PICO_W

### MicroPython version

RPI_PICO_W-20240105-v1.22.1.uf2

### Reproduction

Run the application at https://github.com/Gaudon/zephyr-cycling:
- install `RPI_PICO_W-20240105-v1.22.1.uf2` to a Pico W board
- copy all files from the above repo to the board (omit `firmware` directory)
- copy `config/template-primary.txt` or `config/template-secondary.txt` to the board at `config/config.txt` (they can be changed to assign different GPIO if needed)
- reset the board to run the code in `main.py`

### Observed behaviour

Code sometimes hangs at `ble.active(True)`.

dpgeorge, 2024-03-15:
@Gaudon can you please clarify a few points:
- Do you use a SWD debugger attached to the Pico W?  Or some other kind of debugger?
- What are your exact `config.txt` files for the primary and secondary cases?

Gaudon, 2024-03-15:
@dpgeorge I use the [MicroPico](https://marketplace.visualstudio.com/items?itemName=paulober.pico-w-go) VSCode extension.

For the case of reproduction, you only need a single primary pico (how I reproduced it earlier) and changing nothing from the template is suitable to achieve this. 

Simply rename the `template-primary.txt` to `config.txt` 

or 

create a new file called `config.txt` with the contents found in `template-primary.txt` in the config/ directory.