micropython/micropython #14115: Micropython fails to build with gcc 14 snapshots

stratakis, 2024-03-17:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

Unix port

### MicroPython version

1.21 and 1.22.1

### Reproduction

Building on Fedora with the latest GCC 14 snapshot and Python 3.12.2  we get various test failures. From https://bugzilla.redhat.com/show_bug.cgi?id=2259215

Various tests FAIL in micropython when built with GCC 14 snapshots (such as gcc-14.0.1-0.2.fc40 in rawhide).

I believe the bug is in micropython-1.21.0/py/nlrx64.c (nlr_jump).
Preprocessed the function is:
__attribute__((noreturn)) void nlr_jump(void *val) {
    nlr_buf_t **_top_ptr = &(mp_thread_get_state()->nlr_top); nlr_buf_t *top = *_top_ptr; if (top == ((void *)0)) { nlr_jump_fail(val); } top->ret_val = val; nlr_call_jump_callbacks(top); (void)top; *_top_ptr = top->prev;
    __asm volatile (
        "movq   %0, %%rcx           \n"
        "movq   72(%%rcx), %%r15    \n"
        "movq   64(%%rcx), %%r14    \n"
        "movq   56(%%rcx), %%r13    \n"
        "movq   48(%%rcx), %%r12    \n"
        "movq   40(%%rcx), %%rbx    \n"
        "movq   32(%%rcx), %%rsp    \n"
        "movq   24(%%rcx), %%rbp    \n"
        "movq   16(%%rcx), %%rax    \n"
        "movq   %%rax, (%%rsp)      \n"
        "xorq   %%rax, %%rax        \n"
        "inc    %%al                \n"
        "ret                        \n"
        :
        : "r" (top)
        :
        );
    __builtin_unreachable();
}
The bug is that the inline asm doesn't use "memory" clobber nor say uses "m" (top) to explain to the compiler that the inline asm reads from memory, so starting with https://gcc.gnu.org/r14-1982 GCC dead store eliminates the *_top_ptr = top->prev; store - the inline asm doesn't indicate it reads any memory and then is __builtin_unreachable () so nothing later will need the store either.

Reproducible: Always

### Expected behaviour

Tests pass

### Observed behaviour

[build.log](https://github.com/micropython/micropython/files/14628191/build.log)

### Additional Information

No, I've provided everything above.

dpgeorge, 2024-03-18:
Thanks for the report.

Does adding the relevant memory hints to the inline-asm statement fix the compiler warning?

We will need to fix this probably for all architectures, not just x86-64.

In the meantime, if you want to get the build going without error, you can safely use the following option to make:
```
$ make CFLAGS_EXTRA=-DMICROPY_NLR_SETJMP=1
```
(Although trying that locally it does introduce a separate warning in `nativeglue.c`...)

stratakis, 2024-03-18:
With either approach it looks to be working, although I'm no assembly expert so I'm not sure of the ramifications here:

```
diff --git a/py/nlrx64.c b/py/nlrx64.c
index 6b7d026..d1ad91f 100644
--- a/py/nlrx64.c
+++ b/py/nlrx64.c
@@ -123,7 +123,7 @@ NORETURN void nlr_jump(void *val) {
         "ret                        \n" // return
         :                           // output operands
         : "r" (top)                 // input operands
-        :                           // clobbered registers
+        : "memory"                  // clobbered registers
         );
 
     MP_UNREACHABLE
```

or

```
diff --git a/py/nlrx64.c b/py/nlrx64.c
index 6b7d026..d662f89 100644
--- a/py/nlrx64.c
+++ b/py/nlrx64.c
@@ -122,7 +122,7 @@ NORETURN void nlr_jump(void *val) {
         "inc    %%al                \n" // increase to make 1, non-local return
         "ret                        \n" // return
         :                           // output operands
-        : "r" (top)                 // input operands
+        : "m" (top)                 // input operands
         :                           // clobbered registers
         );
```

On Fedora we only build for x86_64 so I can't test other architectures.

projectgus, 2024-03-18:
FWIW I think for nlr_jump the memory clobber is probably safer as it also ensures any other dirty registers from previous code are written back to memory before jumping. Thanks for looking into this, @stratakis

dpgeorge, 2024-03-18:
@projectgus I guess we should add `"memory"` to all the nlr implementations with inline asm?

projectgus, 2024-03-18:
I think so, definitely for `nlr_jump`. For `nlr_push` we should probably also double check. We might be able to use some subtler method as this doesn't need to flush all dirty registers (I don't think), but then again there might not be much performance difference and `"memory"` is a lot simpler to understand.

I can look into this more if you'd like?

dpgeorge, 2024-03-19:
See #14126 for a fix, which adds `"memory"`.

dpgeorge, 2024-03-25:
Fixed by 35f3f0a87db2580041dd0f7dfd4361df48887796

closed 2024-03-25 by dpgeorge

stratakis, 2024-03-25:
Thanks for the fix!