micropython/micropython #14125: In a manifest file, freeze(".", []) freezes all files in the directory, rather than none of them

ironss-iotec, 2024-03-19:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

All

### MicroPython version

v1.22.2

### Reproduction

* Create a manifest file with the line, `freeze(<dirname>, [])`, where `dirname` is a directory that contains multiple `.py` files
* Build in the usual way, perhaps deleting `frozen_content.c` to ensure that it gets rebuilt

### Expected behaviour

`frozen_content.c` does not contain any `.py` files from `dirname`

### Observed behaviour

`frozen_content.c` contains ALL `.py` files from `dirname`

### Additional Information

The comment above `freeze()` in `manifestfile.py` says that 
* if `script` is `None`, then all files in the directory are frozen.
* if `script` is an iterable, then all the items in the iterable are frozen.

However, if the iterable is empty, then all files in the directory are frozen, rather than none of them.

The reason for this failure is that `_freeze()` (in `manifestfile.py`, around line 286) has the test `if files:` rather than `if files is not None:`, as an empty iterable is falsey. So execution falls through to the search-the-directory path, rather than the iterate-through-the-list case.

This is important because convenient to structure a manifest file as

    freeze('.', [
        'file1.py',
        'file2.py',
    ])

Then it is easy to remove files temporarily from the manifest by commenting out individual lines. However, if you happen to comment out all of the lines in the list, then you suddenly get all of the files in the subdirectory, rather than none of them.

The fix is a one-line change

    diff --git a/tools/manifestfile.py b/tools/manifestfile.py
    index aa85c4650..ca84edd22 100644
    --- a/tools/manifestfile.py
    +++ b/tools/manifestfile.py
    @@ -282,7 +282,7 @@ class ManifestFile:
         def _search(self, base_path, package_path, files, exts, kind, opt=None, strict=False):
             base_path = self._resolve_path(base_path)
     
    -        if files:
    +        if files is not None:
                 # Use explicit list of files (relative to package_path).
                 for file in files:
                     if package_path:

dpgeorge, 2024-03-19:
Thanks for the detailed report.

Your fix sounds good!  Feel free to make a PR for that, otherwise I will get to it.

jimmo, 2024-03-21:
@ironss-iotec FYI instead of using

```py
freeze('.', [
    'file1.py',
    'file2.py',
])
```

I would recomend using

```
module('file1.py')
module('file2.py')
```

(In general, `freeze` should be considered deprecated, use `module` or `package`, specifying `base_path` as necessary)

dpgeorge, 2024-03-25:
See #14170 for a fix.

dpgeorge, 2024-03-25:
Fixed by 51d05c442afca147d225f99fcec5f66b89b1c525

closed 2024-03-25 by dpgeorge