micropython/micropython #14190: samd.Flash() hangs with an ISSI QSPI flash on the board -- with fix

ironss-iotec, 2024-03-26:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

samd51 port on a custom board with ISSI QSPI flash

### MicroPython version

v1.22.2, modified to add support for ISSI QSPI flash

### Reproduction

* Add support for ISSI IS25LPWP064D QSPI flash in drivers/memory/external_flash_devices.h -- details and a link to the datasheet in additional information
* Add 'IS25LPWP064D' to the #define list in `mpconfigboard.h` (eg. `#define MICROPY_HW_QSPIFLASH IS25LPWP064D`)
* Build system, etc
* Run Micropython code to create a flash device `fdev = samd.Flash()`
* System hangs in `samd_qspiflash_make_new()` in `samd_qspiflash.c`, around line 340
```
while (read_status2() & 0x80) {
}
```

### Expected behaviour

samd.Flash() returns and allows flash operations.

### Observed behaviour

System hangs forever

### Additional Information

This is a mix between a bug and a new feature: there is a bug in the existing code that (perhaps) only becomes relevant when used with a specific, new hardware device.

The problem is that `samd_qspiflash_make_new()` always calls function `read_status2()`, but the configuration for that flash device says it has only a single status register (`flash_device->single_status_byte=true`, so that function should not be called.

Other places in the code check `flash_device->single_status_byte` before calling `read_status2()`.

The fix is easy: don't try to read the second status byte if the device does not have one:

```
index 054bfd231..c00ff5f0d 100644
--- a/ports/samd/samd_qspiflash.c
+++ b/ports/samd/samd_qspiflash.c
@@ -337,9 +337,13 @@ STATIC mp_obj_t samd_qspiflash_make_new(const mp_obj_type_t *type, size_t n_args
     // The write in progress bit should be low.
     while (read_status() & 0x01) {
     }
-    // The suspended write/erase bit should be low.
-    while (read_status2() & 0x80) {
+
+    if (!flash_device->single_status_byte) {
+        // The suspended write/erase bit should be low.
+        while (read_status2() & 0x80) {
+        }
     }
+
     run_command(QSPI_CMD_ENABLE_RESET);
     run_command(QSPI_CMD_RESET);
     // Wait 30us for the reset
```

There are other flash devices that have only a single status byte. Presumably, the `read_status2()` operation is benign on other devices (or perhaps those QSPI devices have not been tested recently).

On the ISSI flash, the `QSPI_CMD_READ_STATUS2` command (0x35) does another operation called `QPIEN`; this is a new mode of operation `QPI ` mode, different from QSPI mode. The result is that any further operation with the flash device returns all-1s (0xFF) until the `QPIDI` command (disable QPI) or the device is reset (power-cycle, hardware reset pin or software reset command). The function returns 0xFF, the "write/erase bit" is never low, so the loop cycles endlessly.

Datasheet is available from https://www.issi.com/WW/pdf/25LP-WP064D.pdf

The configuration for the ISS flash is 

```
diff --git a/drivers/memory/external_flash_device.h b/drivers/memory/external_flash_device.h
index 037984460..b5cf4a887 100644
--- a/drivers/memory/external_flash_device.h
+++ b/drivers/memory/external_flash_device.h
@@ -442,6 +442,23 @@ typedef struct {
         .single_status_byte = false, \
 }
 
+// Settings for the ISSI devices
+#define IS25LPWP064D { \
+        .total_size = (1 << 23), /* 8 MiB */ \
+        .start_up_time_us = 5000, \
+        .manufacturer_id = 0x9D, \
+        .memory_type = 0x60, \
+        .capacity = 0x17, \
+        .max_clock_speed_mhz = 80, \
+        .quad_enable_bit_mask = 0x40, \
+        .has_sector_protection = false, \
+        .supports_fast_read = true, \
+        .supports_qspi = true, \
+        .supports_qspi_writes = true, \
+        .write_status_register_split = false, \
+        .single_status_byte = true, \
+}
+
 // Settings for a GENERIC device with the most common setting
 #define GENERIC { \
         .total_size = (1 << 21), /* 2 MiB */ \
```

I have tested these changes on our custom board. It would be nice to add support for the ISSI QSPI flash to Micropython, but understand that it is difficult to test against all hardware.

I don't have a convenient way to make a PR, but could take the time to set up a public clone if essential.

robert-hh, 2024-03-27:
Thank you for your report. It#s a good find. The change to the constructor seems to be easily added, and if you cannot make a PR for it. I can add that to one of my open "service packs". The additional ISSI QSPI flash data could be added as well, but I have no means to test that.

robert-hh, 2024-03-27:
So I tested that change with a few SAMD51 boards and added a commit to an open SAMD "service pack" PR. None of the boards I have for testing have `single_status_byte = true`, so I cannot cross-check that behavior. Maybe it never worked.
I can add the definitions for the ISSI device as another commit.

dpgeorge, 2024-03-28:
Should be fixed by d9b9e88899df86fec5679138f5f60ed62df0975f, and ISSI flash definition added in 4662a71f44bd8a0f8ff96d12e43f69be3bff8488

closed 2024-03-28 by dpgeorge