micropython/micropython #14200: Watch mode for mpremote

dustinlacewell, 2024-03-27:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) regarding this feature, and didn't find any.

### Description

Please consider adding a "watch" mode for mpremote, which will watch a root directory for changes and sync those changes to the micropython filesystem. This would make me much more efficient during heavy cycles of development.

### Code Size

I think the efficiency gains for MicroPython developers would be great.

### Implementation

- [ ] I intend to implement this feature and would submit a Pull Request if desirable.
- [X] I hope the MicroPython maintainers or community will implement this feature.
- [ ] I would like to [Sponsor](https://github.com/sponsors/micropython#sponsors) development of this feature.

peterhinch, 2024-03-28:
An alternative approach (which has been suggested) is to implement `rsync`, as per rshell. This is probably easier to implement given that mpremote must run on Windows and Apple systems.

dustinlacewell, 2024-03-29:
Thanks. I don't quite understand the suggestion (I do know what rsync is), could you elaborate?

peterhinch, 2024-03-29:
`rsync` is a Linux utility which copies files from a source directory tree to a matching destination tree. It only copies a file if the source differs from the destination. You might have a large project with multiple files where one file has been updated. `rsync` will rapidly find that file and copy it to the destination. 

See [the rshell doc](https://github.com/dhylands/rshell?tab=readme-ov-file#rsync) which brings that capability to MicroPython targets.

dustinlacewell, 2024-03-30:
> `rsync` is a Linux utility which copies files from a source directory tree to a matching destination tree. It only copies a file if the source differs from the destination. You might have a large project with multiple files where one file has been updated. `rsync` will rapidly find that file and copy it to the destination.
> 
> See [the rshell doc](https://github.com/dhylands/rshell?tab=readme-ov-file#rsync). This has brought that capability to MicroPython targets.

Oh sorry, I _do_ know what rsync is, but I didn't understand what you were suggesting. I see now that you were just observing that an alternative would be for mpremote to instead implement rsync like rshell does.

I think that synchronization and watching for changes are separate concerns, though related. No point in watching for changes, if you can't then sync those changes, hehe. 

I opened an issue with rshell, recommending they add watch functionality too.

While I appreciate and am impressed by the mounting feature, it is a bit heavy handed especially when I'm working on different projects on different devices throughout the day. 

Anyway, just a suggestion, thanks for the replies!

andrewleech, 2024-03-31:
For what it's worth I've got a kind of watch and sync in this PR https://github.com/micropython/micropython/pull/8231 though it's more about building and syncing a manifest than a folder directly.

This was a POC feature that would need updating to rebase onto master and discussion around whether the use of manifest like this is the best way forward for mpremote but I found it very useful on the project I was working on at the time.

mattytrentini, 2024-11-02:
> Also, for development, you can try mounting a local directory. That way, changes are immediately "synced", without wearing down flash.

Watch should be a little different though; I think any filesystem changes (within the folder that's mounted) should trigger a soft-reset. The mount will be preserved but files can be re-imported - and if there's a `main.py` in the root of the mount then it should be automatically imported, though that's not the current behaviour.

I've started implementing this with [watchdog](https://pythonhosted.org/watchdog/) to monitor for filesystem changes...

peterhinch, 2024-11-02:
> I think any filesystem changes (within the folder that's mounted) should trigger a soft-reset.

The target may or may not be running Python code at the time. I can envisage use-cases where you don't want a reset in the latter case.

Perhaps someone could explain the purpose of this feature. It seems a lot of effort to save issuing `ctrl-c ctrl-d`.

mattytrentini, 2024-11-03:
> The target may or may not be running Python code at the time. I can envisage use-cases where you don't want a reset in the latter case.

So you'll know not to save your file. Or don't use the watch option. 

But I'd argue that if you're using mount then the vast majority of the time you want to restart when you modify files in your mounted folder.
 
> Perhaps someone could explain the purpose of this feature. It seems a lot of effort to save issuing `ctrl-c ctrl-d`.

It's a _very_ common use case in many other languages/tools (particularly web-based languages - node and typescript for example). Many developers expect it. And for those coming from the CircuitPython world it's very similar to their USB mass-storage development model; save, restart.

"Why do I have to Ctrl+C/Ctrl+D? What does that do?" are questions I've had to answer enough that I don't really want to any more. 😜

peterhinch, 2024-11-03:
Thanks for the explanation. So long as "watch" is an option it sounds good.