micropython/micropython #14203: I2C4 on STM32F7 not working anymore

zsquareplusc, 2024-03-28:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

stm32

### MicroPython version

MicroPython v1.23.0-preview.160.g9e5b6972c.dirty on 2024-03-28; EDT with STM32F777
(git master)

### Reproduction

>>> machine.I2C(4)
I2C(4, scl=B6, sda=B7, freq=18000000, timingr=0x00000000)

clock settings are wrong

### Expected behaviour

>>> machine.I2C(4)
I2C(4, scl=B6, sda=B7, freq=321428, timingr=0x70330309)

### Observed behaviour

as commented here https://github.com/micropython/micropython/commit/5aec051f9f0e1be9750ca4f08478275f298087a3#r140318499 that commit breaks I2C4 clock initialization on STM32F7 (probably others too)

dpgeorge, 2024-03-28:
See #14202 for a fix.

dpgeorge, 2024-05-04:
Fixed by a521df27dc483c5376ef7e348aa9fbbd0c4e67a9

closed 2024-05-04 by dpgeorge