micropython/micropython #14217: ESP32-S3 forcing use of USB OTG

kdschlosser, 2024-03-30:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

ESP32-S3

### MicroPython version

MicroPython: 1.22.2
board: ESP32_GENERIC_S3
board_variant: SPIRAM_OCT

### Reproduction

compile for the ESP32-S3 and try and use the pins for I2C. Using them results in a timeout or if doing a scan it takes the scan more than a minute to complete and returns an empty list.

### Expected behaviour

for I2C to work using pins 19 and 20

### Observed behaviour

I2C attached to pins 19 and 20 does not work I believe this issue is caused by the forced use of USB OTG. I have tried to turn it off by setting `CONFIG_USB_OTG_SUPPORTED=n` in the `ports/esp32/boards/ESP32_GENERIC_S3/sdkconfig.board` file and also removing `boards/sdkconfig.usb` from the `ports/esp32/boards/ESP32_GENERIC_S3/mpconfigboard.cmake` file. For whatever reason `CONFIG_USB_OTG_SUPPORTED` still gets set to `y` when being compiled.

Normally this wouldn't be an issue but because of this code in MicroPython

```c
void mp_task(void *pvParameter) {
    volatile uint32_t sp = (uint32_t)esp_cpu_get_sp();
    #if MICROPY_PY_THREAD
    mp_thread_init(pxTaskGetStackStart(NULL), MICROPY_TASK_STACK_SIZE / sizeof(uintptr_t));
    #endif
    #if CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG
    usb_serial_jtag_init();
    #elif CONFIG_USB_OTG_SUPPORTED
    usb_init();
    #endif
    #if MICROPY_HW_ENABLE_UART_REPL
    uart_stdout_init();
    #endif
    machine_init();
```

Pin 19 and 20 get forced into use for OTG instead of being something that user is able to selectively turn on. I believe that the ESP-IDF is overriding the setting `CONFIG_USB_OTG_SUPPORTED=n` which is why I am not able to turn it off and because MicroPython is coded in a manner that forces those pins into being used for USB OTG at runtime results in my inability to use those pins for anything else.

I believe that this kind of a thing should be able to be turned on or off at the users discretion and it should be able to be done at runtime and not compile time. The `CONFIG_USB_OTG_SUPPORTED` is simply a setting that should be used to have code compiled and not be used to determine if it should run code. That macro as it implies if whether or not a board supports OTG not that it should be used. 

### Additional Information

I have not managed to grasp why the OTG is having repl data sent to it.

dpgeorge, 2024-03-30:
What hardware are you using?  I'm guessing that you have a board that does not have a USB connector on GPIO19/20, and instead uses a UART REPL.

> For whatever reason `CONFIG_USB_OTG_SUPPORTED` still gets set to `y` when being compiled.

It looks like this config setting is always enabled if the MCU supports USB, which is ESP32-S2 and ESP32-S3.  You cannot turn this option off.

So MicroPython should probably not use this option to control whether `usb_init()` is called.  Instead maybe it should use `CONFIG_TINYUSB_CDC_ENABLED`.

kdschlosser, 2024-03-30:
If the only purpose for using the USB feature is to output repl data, which is what it looks like then this being set is what should e checked.

CONFIG_ESP_CONSOLE_USB_CDC 

I also put into question the complexity of the code being used because if the user is wanting the repl data output to the USB then why not use `esp_tusb_init_console` which redirects the output from the uart to the USB instead if having this code.

```
#if CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG
    usb_serial_jtag_tx_strn(str, len);
    did_write = true;
    #elif CONFIG_USB_OTG_SUPPORTED
    usb_tx_strn(str, len);
    did_write = true;
    #endif
    #if MICROPY_HW_ENABLE_UART_REPL
    uart_stdout_tx_strn(str, len);
    did_write = true;
    #endif
```

Which kind of makes no sense because of this...
```
#ifndef MICROPY_HW_ENABLE_UART_REPL
#define MICROPY_HW_ENABLE_UART_REPL (!CONFIG_USB_OTG_SUPPORTED && !CONFIG_ESP_CONSOLE_USB_CDC && !CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG)
#endif
```

In reality why would someone want to occupy 4 pins that have the purpose of doing the same thing essentially?

The USB OTG should be exposed to the user in MicroPython if it is available and it is not being used for repl output. There are quite a few things it can be used for. Things like connecting a mouse, keyboard or joystick are a few examples. Just because a board has a second USB port doesn't mean that the USB should be locked to only being able to be used for the purposes of sending repl data.

kdschlosser, 2024-03-30:
I have a waveshare display board that has 2 USB ports and the REPL is available over the uart.  I would imagine it is because of 
`#define MICROPY_HW_ENABLE_UART_REPL         (1)` being set in mpconfigboard.h for the S3. So as a default there is a loss of 4 pins instead of 2.

kdschlosser, 2024-03-30:
The board I am working with is a Sunton display board and no it doesn't have a second USB port.

The waveshare board has a way of turning on and off the use of the second USB PORT by using an IO expander.

vshymanskyy, 2024-08-11:
I have also faced this, and I was able to fix it by introducing an additional `#define MICROPY_HW_ENABLE_CDC_REPL`, which then replaces `CONFIG_USB_OTG_SUPPORTED` in many cases.

This can be found here as part of a larger set of changes:
https://github.com/micropython/micropython/compare/master...vshymanskyy:micropython:develop

If anyone's interested, I can create a PR with this specific change.

kdschlosser, 2024-08-11:
The ESP-IDF internally has the capability to redirect the stdout to the OTG. There is no need to have micropython specifically writing the REPL to the OTG. It should continue to write it to the UART and it will redirect if the config setting has been set to do it. That same config setting is what MicroPython should be using to determine if it needs to set the pins up for OTG use or not. If that redirect is not enabled then use the UART and if it is then use the OTG pins. Right now what is happening is MicroPython is setting up the UART used or not and also setting up the OTG whether or not it is used. By doing this it causes the pins to become unusable for any other purpose.

kdschlosser, 2024-08-27:
however it is wanting to be done to solve the issue.

Question about the efuses. If the USB has not been disabled via the efuse there is still a choice whether or not to use the USB correct? Once the efuse is set to disable something that's it it can never be enabled again correct? If that is the case and even tho the efuse has not been disabled for USB that there is a choice whether or not to use the USB there would be no reason to ever disable the efuse, doing so would permanently change the ESP32 and it is not something I believe most people would end up doing. Especially if there is another option to free the use of the pins.

From my understanding this is what can and cannot be done with the USB pins.

* JTAG and USB Serial Console: This can be done
* JTAG and CDC other than serial console: This cannot be done
* JTAG and OTG: this cannot be done
* CDC and OTG: this cannot be done
* USB serial console and OTG: this cannot be done
* USB Serial Console and CDC: this can be done

I do not now know how the ESPIDF is set up and if these things are able to be enabled or disabled at runtime or if it needs to be done at compile time. I have not looked. I also do not know if the above list is 100% correct or not as I have not tested it specifically myself. In an ideal world these things would be able to be enabled or disabled at runtime instead of compile time. Python is a runtime programming language which is why it would be ideal to be able to turn them on or off at runtime.

As we all know the single largest limitation of the ESP32 is the number of pins it has available. Having pins that are set at compile time removes the ability to share pins. an example would be the UART0 pins. when the ESP32 first starts up a control sequence can be sent out the serial pins and if that sequence is responded to and the response is correct then the serial stays active. If not the serial disables and the pins can be used for something else like a sensor input. This would allow a project to have a port where a sensor could be plugged in or when wanting to flash or access the REPL the sensor could be unplugged and an FTDI UART to USB device would be plugged in instead. If the USB pins are able to be used in the same manner that would be ideal. specifically having to assign pins that are always locked so they provide a serial console and cannot be used for any other purpose is bonkers. The serial console is something that might get used once in a blue moon so dedicating pins only to provide that ability can cause some issues.

 I have personally had this come up where I was 1 pin short and ended up having to add an IO expander and moving pins around to be able to connect the IO expander. I was not using WiFi so the WebREPL was not an option.

projectgus, 2024-08-27:
> Question about the efuses. If the USB has not been disabled via the efuse there is still a choice whether or not to use the USB correct? 

Never again on that chip. Burning the USB disable efuse would mean the two USB pins on that chip are permanently regular GPIO and can't be switched back to USB functions. I also haven't actually confirmed this works with MicroPython, I just believe it would. It's *absolutely not the proper fix* for this issue.

kdschlosser, 2024-08-28:
agreed, having to make a permanent change to the ESP32 is definitely not a solution. I had thought that is how the efuses work but I was not 100% sure.

With as many users as MicroPython has I am sure that the folks can be asked over at espressif to alter the IDF's code so that the USB relevant code be changed to being runtime if it is not already made that way. I am thinking that some portions might be runtime changeable but not all of it. as an example, I would not believe that USB OTG and USB CDC could be enabled at the same time. They may not conflict with each other unless both are started but I can see where the decision might be made to only compile the code for one or the other and if both are enabled then there would be a compiler error. Once again, I do not know too much about that portion of the ESP-IDF and what can and cannot be done and I am making assumptions. I would have to dive into it to see what is possible.

Out of question, Right now how MicroPython runs is on Core 0. Core 1 gets used for Bluetooth and WiFi. Would the USB stuff also reside on core 1 or would it end up on core 0? I don't commonly use Bluetooth or WiFi so I end up with one of the cores mostly sitting there idle. If the USB related code was able to run on core core 1 how would that work with the REPL?

vshymanskyy, 2024-09-26:
I can confirm that after https://github.com/micropython/micropython/pull/15727 was merged, I am able to disable `MICROPY_HW_USB_CDC` in `mpconfigboard.h` of my board:
```c
// Disable USB CDC as the same pins are used by I2C0
#define MICROPY_HW_USB_CDC                  (0)

#define MICROPY_HW_I2C0_SCL                 (19)
#define MICROPY_HW_I2C0_SDA                 (20)
```

After that, i can use pins 19 and 20 for I2C as usual:
```py
>>> board.i2c().scan()
[72, 80, 81]
```

kdschlosser, 2024-09-26:
it should be defaulted to to off not on...

vshymanskyy, 2024-09-26:
Why? On most boards, you may actually want the CDC to be enabled by default.

kdschlosser, 2024-09-26:
on the majority of the boards USB-CDC is NOT being used. Most boards still use the UART. Having MicroPython shanghai 2 pins by default when the majority do not have those 2 pins connected to USB is an insane concept to me. Have a look at the ESP32-S3 devkits. They have BOTH the UART and the CDC connected. So you are going to make users plug into one port to flash the firmware and then have to unplug the device when done and plug into the second USB port in order to get the REPL to show.. WOW is that going to cause a large number of people opening issues for the REPL not working. 

That being said.. The name `MICROPY_HW_USB_CDC` should not be used. You are not enabling hardware USB CDC for the purposes of the user being able to use it. you are enabling REPL over the USB-CDC.  The macros should be named as to where the REPL output should be sent to. 

It is possible to have more than one device at a time so binding the CDC so it is only able to be used for the REPL is not a fantastic thing to do. 

https://docs.espressif.com/projects/esp-idf/en/v5.2.3/esp32s3/api-reference/peripherals/usb_device.html?highlight=cdc#overview

USB-CDC should be exposed to the interpreter and that should be able to be turned on and off using a macro named something along the lines of what is being used right now to turn on and off the REPL over the USB-CDC. where the REPL is output to should not be etched in stone at compile time. It should be something that is able to be changed while the program is running. It would not be a large amount of memory use to be able to do this. what should be able to be done is there should be a function to set what the repl is supposed to write to and read from. this could be a a UART instance, a CDC instance or anything that supports being read from or written to. Files, UART, CDC, WiFi, SPI, I2C, etc.... it shouldn't matter what it is that is being used so long as it can be written to and read from. This could be 2 completely different things. If you want output to go to a display but you want the input to come from USB-OTG you can make that happen by being able to set the REPL to read from one thing but output to something completely different.

andrewleech, 2024-09-27:
> on the majority of the boards USB-CDC is NOT being used. Most boards still use the UART.

On every S2 and S3 board I own, _only_ USB CDC is wired up, UART pins are not connected to anything.
Even if I had a board that wired both I would _never_ use the UART port, I consider it vastly inferior to the native USB in both speed and reliability. So I personally would prefer the native USB port to be enabled by default and UART opt-in only, rather than waste pins on it be default.

Keep in mind most of this code is community contributed and under active development / change. I for one have been working on improving the consistency of repl / stdio interfaces across all the ports recently with the consolidation of stdio handling - this has all been done in my own personal time around my young family needs. This is what I do when there's features that don't work the way I want them to, I raise PR's.

Regardless, the long term fix here is to use the dupterm interface/module for both CDC and UART repl, it's already designed to support dynamic attach/detach of streams to stdio so then users can decide at runtime.

This is how it works currently on the stm port and I'm in progress making it possible on all others that use TinyUSB etc.

kdschlosser, 2024-09-27:
really?

https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/hw-reference/esp32s3/user-guide-devkitc-1.html

https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-4.3

2 examples right there one of them being the devkit made by espressif for the S3. Those both use UART. In fact if you search for S3 devkits you are going to find the large majority of them have the USB to Serial chips on them and they are connected to TXD0/RXD0. Some of them do not have a USB port for the USB-CDC and some of them do.

kdschlosser, 2024-09-27:
It's not about what you use. It's about what is currently being made by the vast majority of the board manufacturers. The majority of them still utilize the UART and some of them provide the USB-CDC but not all. 

You as a single person doing something is not the majority.

vshymanskyy, 2024-09-27:
> on the majority of the boards USB-CDC is NOT being used. Most boards still use the UART. Having MicroPython shanghai 2 pins by default when the majority do not have those 2 pins connected to USB is [...]

I'm not sure what your comment is based on, but I think there are some inaccuracies. The REPL is available on both UART and CDC. Additionally, it might be worth considering:

a) Creating your own board definition if the default doesn’t suit your use case—this is fairly straightforward.

b) Contributing your suggestions through a Pull Request. It could help improve things for everyone.

vshymanskyy, 2024-09-27:
This discussion is too heated

kdschlosser, 2024-09-27:
Why on earth would someone want the REPL to occupy 4 pins instead of 2?? 

How does one flash firmware to the ESP32-S3 when it is in USB-CDC mode. It cannot be done unless it has been written into the firmware that is running on the ESP. Has this been done? I have not seen any code that indicates that this has been done. Using the UART when flashing I plug in the USB cable and run the esptool flash utility, that's it nothing more needs to be done. It flashes no matter what firmware is running on the ESP. Is this able to be done with the ESP over CDC?...

Now... the USB on the ESP is limited to 6 ports and 5 ports of those being in/out ports. to use the CDC_ACM that is going to consume 3 of those ports. That means a user is only able to plug in 1 additional device even if the REPL is not being used. 3 of the available ports has been shanghaied and cannot be used for any other purpose. Writing the driver so it is able to be accessed from python code would mean the user would have the ability to start the CDC_ACM for the REPL and check to see if there is a connection. this could be set with a timeout period of say 30 seconds and after that 30 seconds has expired then it would shut down the CDC_ACM freeing the use of the 3 ports and making them available for other uses. a simple reset of the device would restart the process allowing a Serial host to be connected to get repl output.

If there is no device connected there is mno reason to keep the ports tied up. I believe there is a way to get a signal that a device has been connected to the USB and testing is able to be done to see what kind of a device it is and the proper port is able to be made at that time. NONE of this is being done currently. more than 1/2 of the capability of the USB is being taken control over and is not able to be used for any other purpose. This is done WRONG. When dealing with the UART the pins were already occupied by being connected to a dedicated UART to USB chip and were not able to be used for any other purpose so always having the REPL connected to this pins was not a HUGE deal. With the USB however this doesn't need to be done.

kdschlosser, 2024-09-27:
It's not a board definition issue. It's a core coding issue.

vshymanskyy, 2024-09-27:
The discussion is too heated.
Overall, it it can be fixed on the board definition level, it's a board definition issue. It makes perfect sense for Generic boards to expose as many REPLs as needed for the first-time user.

kdschlosser, 2024-09-27:
This is not meant to be a heated conversation. It is meant to provide information so that people will be able to look past their own singular wants and needs. There are many other folks out there that use MicroPython and this kind of a thing has come up many many many times in a variety of different ways. search open and closed issues for stdout/stdin or disabling the REPL and you will see how many times people have wanted the ability to change the REPL output/input at runtime. It's more than just me..

andrewleech, 2024-09-27:
@kdschlosser when I read your messages they sound very heated to me, even if that's not your intention.
I was presenting my experiences with many esp boards because your expectations sounded to me like your very singular experience. 

Yes the Espressif Dev kit exposes both, as does the waveshare one - devkits often try to expose everything to let users decide what to use. The majority of other smaller esp32 boards I see on the market only expose USB,  not UART.

I always use esptool to flash my S3 boards via the native USB port. There were bugs in the silicon and some other platforms that have hurt the reliability of this in the early days, but that was never the intention of Espressif and PR's I've submitted both here and to esptool have resolved some of these issues.

Regardless it's absolutely now a moot point because as @vshymanskyy said the port repl configuration can be enabled / disabled in board profile. The Dev kids with both ports wired should expose repl on both to avoid confusion, boards that only expose UART or USB can configure just that port. Unused port pins can be user configured for other goals. 
If a user with a dual port board does not want repl on one of those ports they can use a different board profile or compile their own - if the repl was missing on either port by default _that_ would be confusing for many users.

projectgus, 2024-10-01:
Well said, @andrewleech. As far as I'm concerned most of this issue discussion is generating far more heat than light. Time spent here is time not spent improving any code.

Going back to the assessment I wrote in [this comment](https://github.com/micropython/micropython/issues/14217#issuecomment-2311757836) then can re-summarise the status quo as follows:

## Compile time USB enable/disable

This is now possible, and thanks @vshymanskyy for confirming and describing how: https://github.com/micropython/micropython/issues/14217#issuecomment-2377049715 :tada: :tada:

Thanks also @andrewleech for putting in the work for this. :heart: 

## Runtime USB pins

What I said in the comment linked above hasn't changed. It should be possible to reconfigure a USB pin at runtime and have it become a normal pin (maybe with a check that it's not currently being used as an enumerated USB device). However this is not currently possible.

This needs someone to do the work, submit the PR, and would resolve the remaining part of this issue.

<details><summary>There should be almost no behavioural difference between compile-time and runtime disabling of native USB (click for details).</summary>
<p>
<ol>
<li>When any ESP32-S* chips boots from ROM then it will enable the USB port, which - without a USB host present - means USB D- pin is Hi-Z but USB D+ pin has a 1.5k pullup to 3.3V.</li>
<li>When MicroPython starts running and USB was disabled at compile time then during MicroPython startup USB D+ switches back to Hi-Z and both USB pins are then available for normal IO use.</li>
<li>If we enable runtime disabling then the event described in (2) is deferred until MicroPython code does something with the pin, but the behaviour will be otherwise identical. i.e. The external impact is limited to "USB D+ pin has a pullup for a time and may get confused if it sees a particular sequence of input transitions", the only thing which changes is the time period this applies for.</li>
</ol>
</p>
<p>
Preventing the situation described in (1) and disabling the USB port at all times requires burning an efuse to permanently disable USB on that chip. There's nothing MicroPython can do to change this as it's ROM bootloader behaviour.
</p>
</details>

## Summary

* Profiles for boards which have no native USB port can now disable native USB at compile time.
* The generic boards will always have USB enabled so they work in the most number of places.
* Once runtime disabling is implemented then the USB pins should be reusable on these boards, as well.

vshymanskyy, 2024-10-17:
After https://github.com/micropython/micropython/pull/15108 got merged, my board stopped building.
See #16035

trikhonghu, 2025-02-01:
Any solutions for this problem? I am working on ESP32-S3 and I want to use GPIO19 20 for GPIO purpose. But I can't turn off the USB function

andrewleech, 2025-02-02:
> Any solutions for this problem? I am working on ESP32-S3 and I want to use GPIO19 20 for GPIO purpose. But I can't turn off the USB function

As described above in https://github.com/micropython/micropython/issues/14217#issuecomment-2384616932 / https://github.com/micropython/micropython/issues/14217#issuecomment-2377049715
You should be able to disable MICROPY_HW_USB_CDC in mpconfigboard.h of your board profile and recompile the firmware:

```
// Disable USB CDC to allow using the pins as regular gpio
#define MICROPY_HW_USB_CDC
```

closed 2025-03-03 by kdschlosser

Ombenavides, 2025-05-22:
Hello everyone, 

I designed a PCB and used ESP32C3 MIN 1U as a main MCU. I connected pins 18 and 19 as general GPIO without knowing this problem and after days of trying and trying I still cannot solve the issue. I have tried many times by changing mpconfigboard.h and recompiling the firmware like 20 times and still I get the invalid pin error.

MY mpconfigboard.h is like this: 
`#define MICROPY_HW_BOARD_NAME               "ESP32C3 module"
#define MICROPY_HW_MCU_NAME                 "ESP32C3"

// Enable UART REPL for modules that have an external USB-UART and don't use native USB.
#define MICROPY_HW_ENABLE_UART_REPL         (1)
#define MICROPY_HW_USB_CDC                  (0) `

As you can see I disabled the USB_CDC and also enabled the UART because I am using a CP2102 as USB 'UART converter. 
I also tried to modify the sdkconfig without any good output. 

If anyone can help me I would be really appreciated.