micropython/micropython #14224: i hate the access point system

IAmDazen, 2024-04-01:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

pico w

### MicroPython version

Micropython RP2040 Board CDC COM3

### Reproduction

my pico w's access point is locked with the wrong password, it still has a password when i remove it I CANNOT FIX THIS AND THE SSID IS WRONG IT IS SET AS PICO2F6F NOT PicoServer AS IT SHOULD AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA JUST USING ACCESS POINTS BREAK IT

### Expected behaviour

expected to create AP and create webserver from there

### Observed behaviour

no exception, just broken

### Additional Information

No, I've provided everything above.

robert-hh, 2024-04-01:
Setting the AP name and password works for me if I do the BEFORE setting the AP active.

jonnor, 2024-08-24:
Hi and thank you for the bug report. Can you provide a minimal code example which demonstrates this problem? And which version of MicroPython are you using?

projectgus, 2024-11-13:
Can confirm this behaviour - the PICO-W requires you to set `active(False)` *before* configuring the AP config or it doesn't change the actual Wi-Fi behaviour. Whereas the docs say to set `active(True)` before configuring, and the esp8266 and esp32 ports require that.

Fix incoming.

PR #16226 (2024-11-20): extmod/network_cyw43: Allow configuring active AP interface.
### Summary

Configuring the AP for cyw43 writes to some buffers that are only sent to the modem when the interface is brought up. This means you can't configure the AP after calling active(True), the new settings seem to be accepted but the radio doesn't change.

This is different to the documented WLAN behaviour which says to call active(True) before configuring the interface. The esp8266 ~~and esp32~~ ports require this, even. *EDIT: only esp8266 requires this, esp32 works in either order*

Fix this by bouncing the AP interface af [... cut]
closed 2024-11-20 by dpgeorge