micropython/micropython #14228: StateMachine doesn't seem to implement buffer protocol which breaks DMA

tdunning, 2024-04-02:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

rp2

### MicroPython version

3.4.0; MicroPython v1.22.2 on 2024-02-22, Raspberry Pico W

### Reproduction

This code defines a `StateMachine` and demonstrates that it doesn't seem to be a buffer:
```
def PioCopy():
    pull()
    mov(isr, osr)
    push()

# Run the PIO code

sm = StateMachine(0, PioCopy, freq=5000)
sm.active(1)

# results in TypeError: object with buffer protocol required
memoryview(sm)
```

The impact of this is that DMA attempts against a StateMachine fail. For instance, this code:
```
def DmaPut(src, sm: StateMachine):
    d = rp2.DMA()
    c = d.pack_ctrl(size=1)
    d.config(read=src, write=sm, count=10000, ctrl=c, trigger=True)
    return d
```
results in this error:
```
TypeError: can't convert StateMachine to int
```
There doesn't appear to be an easy work around:
```
>>> p = rp2.PIO(0)
>>> memoryview(p)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: object with buffer protocol required
>>> 
```
Calculating the address of the appropriate FIFO and entering that explicitly may work, but the documentation indicates that `StateMachine` and `PIO` both implement the buffer protocol.

### Expected behaviour

Expected `StateMachine` and `PIO` to implement buffer so that they can work with `DMA()`

### Observed behaviour

Neither type implements the required protocol and DMA objects reject them as input or output.

### Additional Information

No, I've provided everything above.

fdufnews, 2024-04-02:
Encountered the same problem and was scratching my head why I am so stupid that I can't make the DMA working.
This reassures me if it is a Micropython problem.

tdunning, 2024-04-02:
My current work-around is to define addresses by hand:
```
pio0_base = 0x50200000
pio_txfifo = 0x10
pio_rxfifo = 0x20
sm_tx = pio0_base + pio_txfifo
sm_rx = pio0_base + pio_rxfifo
```
Then when I define the dma, I do this:
```
    d = rp2.DMA()
    c = d.pack_ctrl(size=1, inc_write=False)
    d.config(read=src, write=sm_tx, count=256, ctrl=c, trigger=True)
```
(this code is broken because it doesn't have any back-pressure but that's not the point)

This code is clear not as nice as it should be. At the very least, it would be *really* nice to have a `.registers` field that returns a `memoryview` of the registers to avoid address errors.

fdufnews, 2024-04-02:
> My current work-around is to define addresses by hand:
I made it the same way and it works.
I will update my code when the bug will be corrected

dpgeorge, 2024-04-03:
The buffer protocol was implemented for `StateMachine` in commit 1da45e887af1b5e6e608a13e292aa343f5b9f976

That commit is not yet part of any release, but will be part of the upcoming 1.23.0 release.

dpgeorge, 2024-04-03:
If you need this feature now, you can use the preview builds (towards the bottom of the download page).

tdunning, 2024-04-03:
Awesome

fdufnews, 2024-04-03:
Thanks a lot