micropython/micropython #14252: SAMD51 on Wio Terminal : No file transfer/write + _boot.py error  for firmware v1.21 & v1.22 (working on v1.20)

mchobby, 2024-04-05:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

SAMD51 : Wio Terminal (SeeedStudio)

### MicroPython version

MicroPython v1.22.2 on 2024-02-22; Wio Terminal D51R with SAMD51P19A
Also MicroPython v1.21 for same plateform 

### Reproduction

Make Soft Reboot show the error 
```
MicroPython v1.22.2 on 2024-02-22; Wio Terminal D51R with SAMD51P19A
MPY: soft reboot
Traceback (most recent call last):
  File "_boot.py", line 14, in <module>
OSError: 28
```
Making a file copy with `mpremote fs cp ...`  or with Thonny IDE fails with an error message

### Expected behaviour

Being able to copy files to the MicroPython file system.

### Observed behaviour

Error message related to file handling

```
$ mpremote fs cp readme.md :readme.md
cp readme.md :readme.md
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 19] ENODEV
```

### Additional Information

**Works** on MicroPython v1.20 for Wio Terminal
**Fails** on MicroPython v1.21 for Wio Terminal
**Fails** on MicroPython v1.22 for Wio Terminal

ricksorensen, 2024-04-05:
This is working for me using 1.23 preview that I built locally:
```
>>> 
MPY: soft reboot
MicroPython 8433ddff7 on 2024-03-20; Wio Terminal D51R with SAMD51P19A
Type "help()" for more information.
>>> import sys
>>> sys.implementation
(name='micropython', version=(1, 23, 0, 'preview'), _machine='Wio Terminal D51R with SAMD51P19A', _mpy=7686)
```
exit repl
```
xxxx:.../mybuilder$ mpremote fs ls :
ls :
xxxx:.../mybuilder$ mpremote fs cp bldwio :bldwio.txt
cp bldwio :bldwio.txt
xxxx:.../mybuilder$ mpremote fs ls
ls :
         245 bldwio.txt
rick@ricklinux2:.../mybuilder$ mpremote
Connected to MicroPython at /dev/ttyACM0
Use Ctrl-] or Ctrl-x to exit this shell

>>> import os
>>> os.listdir('.')
['bldwio.txt']
>>> 
```
1.22.2 also works:
```
MPY: soft reboot
MicroPython v1.22.2 on 2024-02-22; Wio Terminal D51R with SAMD51P19A
Type "help()" for more information.
>>> 
```

I have no custom `boot.py` nor `main.py`, but I see you have an error reported by `_boot.py`

robert-hh, 2024-04-05:
I cannot replicate the behavior. All firmware versions run fine.  The error code tells, that there is a problem with the file system.
There was a major change between V1.20 and V1.21 to the flash driver setup. You can try to re-format the flash drive in V1.22.2 using the code from _boot.py:
```
import vfs
import samd

bdev = samd.Flash()
fs_type = vfs.VfsLfs2 if hasattr(vfs, "VfsLfs2") else vfs.VfsLfs1
fs_type.mkfs(bdev, progsize=256)
fs = fs_type(bdev, progsize=256)
```

mchobby, 2024-04-05:
@robert-hh 
A previous version of MicroPython was running on the terminal. Your assumption looks right !
Unfortunately, **vfs is NOT included** in the SAMD build for Wio Terminal (for MicroPython v1.22).

Dominique

robert-hh, 2024-04-05:
Sorry. That changed between V1.22.2 and the actual preview. Use `os` instead of `vfs`:
```
import os
import samd

bdev = samd.Flash()
fs_type = os.VfsLfs2 if hasattr(os, "VfsLfs2") else os.VfsLfs1
fs_type.mkfs(bdev, progsize=256)
fs = fs_type(bdev, progsize=256)
```

mchobby, 2024-04-05:
@robert-hh 
Give the same OSError: 28 as _boot.py, here the result of the commands.
```
>>> import os
>>> import samd
>>> bdev = samd.Flash()
>>> fs_type = os.VfsLfs2 if hasattr(os, "VfsLfs2") else os.VfsLfs1
>>> fs_type
<class 'VfsLfs2'>
>>> fs_type.mkfs(bdev, progsize=256)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 28
```
which is `#define MP_ENOSPC            (28) // No space left on device`

mchobby, 2024-04-05:
looking a bit more on file-system level with the reference/filesystem.rst , I wrote the following:
```
>>> import samd
>>> bdev = samd.Flash()
>>> bdev.ioctl( 4, 0 ) # Block_count 
1024
>>> bdev.ioctl( 5, 0 ) # block_size 
4096
```
So I have 1024 blocks of 4096 bytes each (as stated in . So a total of 4194304 bytes (4 Megabytes) as expected.

I just wonder about `fs_type.mkfs(bdev, progsize=256)` call trying to make a FileSystem with block size `progsize` of 256 bytes onto a Flash splitted in block_size of 4096 (ioctl returning `BLOCK_SIZE` defines from samd_flash.c ).

### Interesting discovery

I presently do not understand why the code can't format the external Flash. 
The file System is empty on V1.21 & v1.22. But can still mounted when going back to V1.20.
By checking the ioctl on V1.20 I did realize that it may runs on the MCU internal Flash ?!?!
```
MicroPython v1.20.0 on 2023-04-26; Wio Terminal D51R with SAMD51P19A
Type "help()" for more information.
>>> dir()
['bdev', '__name__', 'sys', 'samd', 'fs_type']
>>> fs_type
<class 'VfsLfs2'>
>>> bdev.ioctl(4, 0) # block count
85
>>> bdev.ioctl(5, 0) # block size
1536
```
So a total of 105060 bytes (102 KB of filesystem)

mchobby, 2024-04-05:
Based on previous discovery I decided to explore the external Flash memory.
I Installed back v1.22 on my Wio Terminal and wrote some scripts.

### Exploring the 4 MB flash on V1.22
Based on the BLOCK_SIZE = 4096, I did wrote the following script to detect altered block (containing at least one byte <> 0xFF).
```
from samd import Flash
bdev = Flash()
data = bytearray( 4096 )
for i in range( 1024 ):
    bdev.readblocks(i,data)
    for _byte in data:
        if _byte != 0xff:
            print( "Block %i has data" % i )
            break
```
Then I explored the blocks and ONLY the block 0 contains some data.
```
>>> bdev.readblocks( 0, data )
>>> data
bytearray(b'\x01\x00\x00\x00\xf0\x0f\xff\xf7hipphafq\x0f\xe0\x00\x10\x01\x00\x00\x00
\x00\x10\x00\x00\x00\x00\x00\x00\xff\x00\x00\x00\xff\xff\xff\x7f\xfe\x01\x00\x00\x7f
\xef\xf8\x10\x00\x01\x00\x00\x9e\x17\x17\x01\x0f\xf0\x00\x88\x89\x19\x91n\xff\xff\xff
\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff
\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff
\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff
\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff
...
```
Other modified blocks only contains 0x00.
* The block 0 of 4096 bytes contains some data and filled with 0xFF up to the end of the block.
* The blocks 1, 2, 3, 4 of 4096 bytes each are filled with 0x00.
* The blocks 5 to 1023 of 4096 bytes each are filled with 0xFF (Flash not modified yet).

mchobby, 2024-04-06:
@robert-hh 
Still working with V1.22

### Flash Resetting - the hard way
I decided to reinitialize the flash content with factory default (0xFF on each bytes).
```
from samd import Flash
bdev = Flash()
data = bytearray( 4096 )
for i in range(4096):
    data[i] = 0xFF
bdev.writeblocks(0,data)
bdev.writeblocks(1,data)
bdev.writeblocks(2,data)
bdev.writeblocks(3,data)
bdev.writeblocks(4,data)
```
Then I did reused this script to check that every byte is at 0xFF (so completely un-initialized eeprom).
```
for i in range( 1024 ):
    bdev.readblocks(i,data)
    for _byte in data:
        if _byte != 0xff:
            print( "Block %i has data" % i )
            break
```
### new formatting attempt
Then I reused the suggested code (from _boot.py) to format the flash content.
**It still result in OSError: 28**

```
=== import os
=== import samd
=== bdev = samd.Flash()
=== fs_type = os.VfsLfs2 if hasattr(os, "VfsLfs2") else os.VfsLfs1
=== 
>>> fs_type
<class 'VfsLfs2'>
>>> fs_type.mkfs(bdev, progsize=256)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 28
```
### Exploring Flash content
Only the block 0 contains data (other than 0xFF).
Here some excerpt of the content
```
>>> bdev.readblocks(0,data)
>>> data
bytearray(b'\x01\x00\x00\x00\xf0\x0f\xff\xf7hipphafq\x0f\xe0\x00\x10\x01\x00\x00\x00\x00\x10\x00\x00\x00\x00\x00\x00\xff\x00\x00\x00\xff\xff\xff\x7f\xfe\x01\x00\x00p\x1f\xff\xe6\x98x\x87\x08\xff\xff\xff\xff\xff\xff
......
\xff\xff\x00\x00\x00\x00\xe1\x00\x99\x9e\xff\xff\xff
...
\xff\xff\xff\x00\x00\x00\x00\xe1\x00\x99\x9e\xff\xff\xff\xff\xff
...
\xff\xff\xff\xff\x00\x00\x00\x10\x16\x90\x0f\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff
...
\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff')
```

robert-hh, 2024-04-06:
Yes, v1.20 used the internal flash for the file system. With v1.21 the file system was moved to the external flash, if present.
Your observations are strange. An attempt to overwrite the flash using ioctl.writeblocks failed! That looks like a problem between MCU and flash. Since os.statvfs() (and ioctl) return the proper number of blocks, some communication to the flash chip work. But this control information is exchanged in single bit mode, while bulk data uses QSPI mode.
And yes, I have two WIO terminal devices. One regularly used, the other never used. I unpacked it and tested it. Works flawlessly, like the one I used for testing when making the SAMD firmware.
The labels on the package are the same as your's. 

Comparing the data from block 0 of my device and your device, there are some similarities in the structure, but interesting differences in the content. e.g, at byte 8 you data shows "hippafs", my data show "littlefs". Comparing the differences it looks like a problem with bit D2, which reads as 0 when it should be 1, but not always.

This is the first part of what I read back from block 0:
```
bytearray(b'\x97\x01\x00\x00\xf0\x0f\xff\xf7littlefs/\xe0\x00\x10\x01\x00\x02\x00
\x00\x10\x00\x00\x00\x04\x00\x00\xff\x00\x00\x00\xff\xff\xff\x7f\xfe\x03\x00\x00
\x7f\xef\xfc\x10\x00\x01\x00\x00\xdeWW\x01\x0f\xf0\x00\xccX$\xda\x16\xff\xff
\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff
\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff
\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff
\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff
\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff
\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff
\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff
\xff\xff\xff\xff\xff\xff\xff\xff\x10\x1f\xf8\xc4@\x00\x00\x07main.py \x00\x00\x07
\x7f\xef\xf8\x08\x00\x01\x00\x00\xdeWW\x01\x0f\xf0\x00\xd5_\
```

mchobby, 2024-04-06:
Well noticed with the problem about the bit D2 !
I did reset the first block with 0xFF, then I wrote your header (containing "littlefs").
IF I do read back the block 0 THEN "littlefs" is again replaced with "hipphafq".

**I ordered a new Wio Terminal.**

@@robert-hh ,
**Many thanks for your great support**, I would never put the hardware in balance such a way.
It is the first time I do have such surprising issue with Flash Memory or MCU.

robert-hh, 2024-04-06:
There is indeed a string indication that the hardware is the problem. I would here open the case and look for the flash chip and how it is assembled. The case has two hidden screws at the back side below the rubber feet, if you want to open it.
I checked also the flash clock rate. But that is 24Mhz, while the flash chip is rated at 133MHz.

robert-hh, 2024-04-07:
You could try the attached firmware. It does not switch the external flash to QSPI mode. At least I hope it works as intended.

[firmware.zip](https://github.com/micropython/micropython/files/14895953/firmware.zip)

mchobby, 2024-04-07:
Hi Robert,
I did open the can are replace the chip with a Winbond 4MB QSPI (very close in spec) coming from a computer Bios. I can flash... but firmware doesn't even starts. So I replaced the Original Flash.

The firmware you give me (excluding the QSPI mode) and the error stays identical in any case. 
It is a good idea to checks the clocks... I will do also

robert-hh, 2024-04-07:
The firmware does not need the QSPI chip to boot. But it tries to do enable the QSPI chip. Maybe there is a wait w/o timeout. I should check that.
The problem can be at any place in the connection MCU/flash. The clock should be present, since it works in singe data line mode of the QSPI interface. Only quad data line QSPI mode fails. 
I looked into the QSPI pins features whether the device can be used with a regular SPI flash driver. But that's not possible. The serial port com assignments do not match.

mchobby, 2024-04-10:
Hi Robert,
The brand new "Wio Terminal" works perfectly!
Thank very much for your support
Dominique

closed 2024-04-10 by mchobby

robert-hh, 2024-04-10:
Thank you for the confirmation.