micropython/micropython #14255: [shared] [webassembly] pyexec_event_repl_process_char unable to understand unicode

WebReflection, 2024-04-05:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

webassembly, linux shell

### MicroPython version

latest

### Reproduction

Open a MicroPython REPL or visit this page (which is half patched, but not fully): https://webreflection.github.io/coincident/test/micropython.html

try to type in it the following:

```python
print("µpython")
```

on a native shell you'll see `python` instead of `µpython`, on the Web REPL you see even less because the count goes off due `replProcessChar` (even the Asyncify one) and this is the tip of the iceberg ... now try a combined emoji:

```python
print("👩‍❤️‍👨")
```

... see emptiness or awkward results ...

Most emoji are indeed just broken out of the box **unless** you ask for these as an `input(...)`:

```python
fam = input("> ")
# type 👩‍❤️‍👨
print(fam) # 👩‍❤️‍👨
fam # '\U0001f469\u200d\u2764\ufe0f\u200d\U0001f468'
```

Coincidentally, if you explicitly go into "*REPL paste mode*" (`\5`) you can past anything you like then get out (`\4`) and see all code pasted had no issues in being processed, just like the `input(...)` case.

Related PR that fixes at least the output side of affairs https://github.com/pyscript/pyscript/pull/2018 but it cannot fix users' typing on the terminal somehow as `replProcessChar` misses chars in the process (and yes, it has no `linebuffer` but it's the same with `linebuffer`, the issue is within the code behind `replProcessChar` to me).

### Expected behaviour

if I type the following in the REPL I expect things to just work and output the correct result:

```python
print("µpython")
# µpython
print("👩‍❤️‍👨")
# 👩‍❤️‍👨
```

### Observed behaviour

if I type the following in the REPL this happens instead:

```python
print("µpython")
# python or thon
print("👩‍❤️‍👨")
# ... nothing, awkward state
```

### Additional Information

Pinging @dpgeorge as I've done already in Discord but this looks and feels like a broader issue with REPL because it's possible to reproduce it via native Linux port.

felixdoerre, 2024-04-07:
> because it's possible to reproduce it via native Linux port.

I think I can elaborate on what does not work/what is missing: The micropython-implemented "readline" functionality does not handle unicode characters at all (maybe this also explains why "raw" mode and "paste" mode works). You can see it in this implementation: After escape-sequences are handled, any non-ascii character is ignored (which is, how the first few chars of those emojis would look like):

https://github.com/micropython/micropython/blob/master/shared/readline/readline.c#L279

I'm not sure, where the webassembly port exactly interfaces with the rest of micropython, because it seems like I cannot interact with readline (For example, it does not react to Ctrl-R, and Tab-completion, Ctrl-C seems to work, maybe thats implemented explicitly, and the other special characters are not input completely?)

Probably extending that statement to parse multi-byte UTF-8 characters and handle them correctly, would solve this problem at its root.

(I'm also interested in unicode input working in other instances of the micropython-REPL)

dpgeorge, 2024-04-07:
I think this is a duplicate of #2789.

jonnor, 2024-10-01:
Yes I also hope that fixing it in the simple case will be accepted as well. As long as it is an improvement over the current state and not have other drawbacks. Closing as duplicate of https://github.com/micropython/micropython/issues/2789.