micropython/micropython #14274: btree creashing with ESP-IDF 5.2

xyzzy42, 2024-04-10:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

esp32-s3

### MicroPython version

MicroPython v1.23.0-preview.322.g5114f2c1e on 2024-04-09; Generic ESP32S3 module with 
v1.22.0-323-g5114f2c1e

### Reproduction

```
import btree
db = btree.open(open("database.db","w+b"))
db.close()
db[b'1']
```

### Expected behaviour

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 22] EINVAL
```

### Observed behaviour

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x420310ce  PS      : 0x00060f30  A0      : 0x82030b61  A1      : 0x3fcbb250  
A2      : 0x3fcaa46c  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x3fcbb250  A8      : 0xe266e3dd  A9      : 0xe266e3dd  
A10     : 0x3c181650  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3fca393c  
A14     : 0x3fcbb470  A15     : 0x3fcbb490  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xe266e3f1  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0xffffffff  

Backtrace: 0x420310cb:0x3fcbb250 0x42030b5e:0x3fcbb290 0x4202ea32:0x3fcbb2d0 0x4202d961:0x3fcbb320 0x42021a0d:0x3fcbb370 0x4200b1dc:0x3fcbb3b0 0x40378089:0x3fcbb3d0 0x4200d653:0x3fcbb470 0x420149d1:0x3fcbb4c0 0x420149e6:0x3fcbb4e0 0x42022713:0x3fcbb500 0x42022a08:0x3fcbb590 0x42006238:0x3fcbb5d0

### Additional Information

The expected behavior occurs with ESP-IDF v5.1.3, while the crash is with ESP-IDF v5.2 and v5.2.1.

I tested a number of Micropython versions and ESP-IDF versions.  The consistent factor is that something changed between esp-idf v5.2 and v5.1.3 that causes this crash.

gampam2000, 2024-09-14:
As I found out in https://github.com/micropython/micropython/issues/15841 Micropython only supports the exact ESP-IDF versions that are documented here:
https://github.com/micropython/micropython/blob/master/ports/esp32/README.md

5.1.3 and 5.2.1 are not supported by Micropython. You need to check out the specific tag. The most recent version supported is the esp-idf release tag v5.2.2. I hope that helps.

xyzzy42, 2024-09-14:
v5.2.0 *is* in the supported list.

robert-hh, 2024-09-14:
@xyzzy42 Please check out the version of the esp-idf with the tag v5.2. `git checkout v5.2`.
It has the commit hash `11eaf41b37267ad7709c0899c284e3683d2f0b5e`. The version you used has a different commit hash, starting with `5114f2c1e`.

xyzzy42, 2024-09-16:
I tried a number of idf versions, all since 5.1.3 crash.  Seems unlikely that versions both newer and older would be broken, but one version in the middle would be ok.

But I tested again anyway.

Micropython: v1.23.0-0-ga61c446c0
ESP-IDF: v5.2-0-g11eaf41b37

```
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== import btree
=== db = btree.open(open("database.db","w+b"))
=== db.close()
=== db[b'1']
===
0

A fatal error occurred. The crash dump printed below may be used to help
determine what caused it. If you are not already running the most recent
version of MicroPython, consider upgrading. New versions often fix bugs.

To learn more about how to debug and/or report this crash visit the wiki
page at: https://github.com/micropython/micropython/wiki/ESP32-debugging

MPY version : v1.23.0 on 2024-09-15
IDF version : v5.2
Machine     : Generic ESP32S3 module with ESP32S3

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4203119e  PS      : 0x00060f30  A0      : 0x82030c31  A1      : 0x3fcbb2a0
A2      : 0x3fcaa484  A3      : 0x00000001  A4      : 0x00000001  A5      : 0x3c181384
A6      : 0x00000001  A7      : 0x3fcbb2a0  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x00000000  A11     : 0x00000008  A12     : 0x00000000  A13     : 0x3c181390
A14     : 0x00000059  A15     : 0x00000008  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000014  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000

Backtrace: 0x4203119b:0x3fcbb2a0 0x42030c2e:0x3fcbb2e0 0x4202eb02:0x3fcbb320 0x4202da31:0x3fcbb370 0x42021a89:0x3fcbb3c0 0x4200b228:0x3fcbb400 0x40378089:0x3fcbb420 0x4200d6c7:0x3fcbb4c0 0x42014
a45:0x3fcbb520 0x42014a5a:0x3fcbb540 0x4202278f:0x3fcbb560 0x42022a84:0x3fcbb5f0 0x4200626c:0x3fcbb630
```

dpgeorge, 2024-09-16:
The reproduction code posted above is:
```py
import btree
db = btree.open(open("database.db","w+b"))
db.close()
db[b'1']
```
This is attempting to access the database after closing it.

This (access after close) originally had undefined behaviour because it wasn't being handled correctly.  That issue was reported in #12543 and fixed in #12949.

But note that this fix has not yet landed in a stable release.  Also, support for ESP IDF 5.2.2 has not yet landed in a stable release.  So the situation is:
- If you are on MicroPython v1.23.0 then the bug is not fixed and any `OSError` that you may get from accessing a closed database is not correct behaviour and cannot be relied upon.  Different IDF versions may give different results because, eg, C stack layout is different.  Also, it's recommended to use IDF 5.0.5 when using this MicroPython version (that's what the official firmware builds used).
- If you are on the latest master then the bug is fixed and you can use IDF 5.2.2.

@xyzzy42 can you please try latest master MicroPython and see if the bug is fixed?

xyzzy42, 2024-09-16:
Tested with v1.23.0-310-gf1bdac375 and it works as expected, raising a ValueError on access to a closed database.  Also tested with 6db91dfefb1a7ba0026106d8f0c6589630e9a012 which fixes it as expected.

xyzzy42, 2024-09-16:
Fixed by  #12949