micropython/micropython #14288: raw f-strings

andrewrk, 2024-04-12:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) regarding this feature, and didn't find any.

### Description

Here is a tracking issue for the implementation of raw f-strings.

```
Traceback (most recent call last):
  File "yt_dlp/__init__.py", line 268
SyntaxError: raw f-strings are not supported
```

### Code Size

Parsers are pretty small if you don't overengineer them, and it could easily be configured at build-time. Also, Micropython has use cases outside of microcontrollers, such as my music player project. 

### Implementation

- [ ] I intend to implement this feature and would submit a Pull Request if desirable.
- [X] I hope the MicroPython maintainers or community will implement this feature.
- [ ] I would like to [Sponsor](https://github.com/sponsors/micropython#sponsors) development of this feature.

andrewrk, 2024-04-12:
related: https://github.com/yt-dlp/yt-dlp/issues/9674

wis, 2024-05-31:
I would love to see this in MicroPython, but since I don't have interpreter or compiler building experience I have no idea why the docs say:

> # Raw f-strings are not supported[¶](https://docs.micropython.org/en/latest/genrst/core_language.html#raw-f-strings-are-not-supported)
> Cause: MicroPython is optimised for code space.

(I mean why it mention this to be the reason)

I'd really appreciate it if any of the maintainers can shed some light on the reason why _especially_ Raw f-strings would increase the build size of MicroPython. Andrew, you too are welcome to, I'm sure you'd be able to, as the freaking creator of the Zig programming language.

but for now, building on the suggestion of dirkf, I suggest you do what I'll be doing: use `.format()` with `**locals()` or `**globals()` as the argument.

``` python 
food = "Hummus"
template = r""" I like {food}
It is yummy.""".format(**globals())
```

I initially tried using `.format_map()`, but found that it is not implemented in MicroPython, which is surprising considering it was [_"Added in version 3.2."_](https://docs.python.org/3/library/stdtypes.html#str.format_map) (python docs link), while MicroPython aims for complete compatibility with Python 3.4.

andrewrk, 2024-05-31:
Well, my use case was running ytdlp inside micropython but that's not going to happen (see linked issue above), so I actually have no more stake in this issue. I solved my problem by building CPython with the zig build system and so I provide that runtime environment to ytdlp in my music player.

I'll leave the issue open but I'm unsubscribing now. Cheers!

andrewrk, 2024-05-31:
Oops, pressed the wrong button

dpgeorge, 2024-06-06:
Implementing raw f-strings was actually pretty easy, and did not increase code size.  See #15218.

dpgeorge, 2024-06-06:
Raw f-strings have been implemented in 3c8089d1b10683ee31ddbaeebd0b18c47bf6d09d

@andrewrk FYI

closed 2024-06-06 by dpgeorge

andrewrk, 2024-06-06:
Nice job