micropython/micropython #14293: ESP32C3 BLE get panic when some 10-20k byte arrays is allocated otherwhere before

m24h, 2024-04-12:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

ESP32C3

### MicroPython version

MicroPython v1.22.2 on 2024-02-22; LOLIN_C3_MINI with ESP32-C3FH4

### Reproduction

`MicroPython v1.22.2 on 2024-02-22; LOLIN_C3_MINI with ESP32-C3FH4
Type "help()" for more information.
>>> import micropython as m
>>> m.mem_info()
stack: 1012 out of 14336
GC: total: 64000, used: 11392, free: 52608, max new split: 139264
 No. of 1-blocks: 200, 2-blocks: 29, max blk sz: 94, max free sz: 3278
>>> a=bytes(20000)
>>> b=bytes(20000)
>>> import bluetooth
>>> m.mem_info()
stack: 1012 out of 14336
GC: total: 139968, used: 122640, free: 17328, max new split: 65536
 No. of 1-blocks: 84, 2-blocks: 13, max blk sz: 4747, max free sz: 531
>>> ble=bluetooth.BLE()
>>> ble.active()
False
>>> ble.active(True)
Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4205e2ba  RA      : 0x4205f9fa  SP      : 0x3fca8760  GP      : 0x3fc97c00  
TP      : 0x3fc6e5f4  T0      : 0x0000201d  T1      : 0x00000000  T2      : 0x42067bc6  
S0/FP   : 0x3fc9f868  S1      : 0x000141eb  A0      : 0x3fca0ab0  A1      : 0x00000000  
A2      : 0x00000000  A3      : 0x000049be  A4      : 0x00000002  A5      : 0x00000001  
A6      : 0x00000000  A7      : 0xac400000  S2      : 0x000149bb  S3      : 0x000007d0  
S4      : 0x3fc9ae5c  S5      : 0x00000003  S6      : 0x3fffffff  S7      : 0x3fca99fd  
S8      : 0x00000030  S9      : 0x3c141000  S10     : 0x3c141000  S11     : 0x3c141000  
T3      : 0x00000000  T4      : 0x00014204  T5      : 0x0000201e  T6      : 0x42068332  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
MHARTID : 0x00000000  

Stack memory:
3fca8760: 0x000000c0 0x3fcaa3d0 0x3fc9f738 0x42027a5a 0x3fffffff 0x3ffffffe 0x40000000 0x3fca98c0
3fca8780: 0x000000c0 0x3fca895c 0x3fca99fe 0x4201bce6 0x000000c0 0x3fca895c 0x3fca99fe 0x40382f12
3fca87a0: 0x3fca8954 0x00000008 0x3fca8968 0x3fca8940 0x3c14888c 0x3fc9ae5c 0x3fca8964 0x00000000
3fca87c0: 0x00002dda 0x3fca8920 0x00000000 0x3fca895c 0x00000020 0x3fca8920 0x3fca89c8 0x4203f348
3fca87e0: 0x403828fa 0x3fca89a0 0x3fc9aeb8 0x00000000 0x00000000 0x00000000 0x00000005 0x00000000
3fca8800: 0x3fca8940 0x00000030 0x3c141000 0x3c141000 0x3c141000 0x3fca87a0 0x000000a8 0x000000a7
3fca8820: 0x00000030 0x3fcaa7f0 0x00000032 0x3fcaa804 0x3fca8920 0x00000002 0x00000002 0x4203f860
3fca8840: 0x00000000 0x000000a6 0x00000000 0x00000001 0x00000000 0x3fca8920 0x00000001 0x4203f348
3fca8860: 0x3fca8954 0x00000001 0x00000001 0x3fcaa80c 0x3c141000 0x3c141000 0x00000001 0x3fca886c
3fca8880: 0x00000030 0x3fcaa81c 0x00000032 0x3c141000 0x3c141000 0x3c141000 0x00000001 0x00000000
3fca88a0: 0x00000030 0x3fcaa81c 0x00000032 0x3fcaa830 0x3fca8920 0x00000002 0x00000002 0x4203f860
3fca88c0: 0x00000002 0x000000e9 0x3fca89c0 0x00000004 0x00000000 0x3fcaa848 0x3fca8920 0x4203db6e
3fca88e0: 0x00000000 0x00000001 0x3fca89c0 0x00000004 0x00000002 0x00000005 0x00000014 0x00000005
3fca8900: 0x3fcaa3d0 0x0000000f 0x3fca98e0 0x3c141000 0x3c141000 0x3c141000 0x00000030 0x3fca8940
3fca8920: 0x00000000 0x00000005 0x00000000 0x00000000 0x00000000 0x3fc9aeb8 0x3fca89a0 0x4200f29a
3fca8940: 0x3fca98e0 0x3fca99fc 0x3fca8950 0x00000005 0x3fc9aeb8 0x3fcaa1d0 0x3fcaa2e0 0x3c1490bc
3fca8960: 0x3fca98b0 0x0000001e 0x3fca89a0 0x4200f1ec 0x3c14b000 0x3c14b000 0x3c142000 0x3fcb8be8
3fca8980: 0x3c14b000 0x3c14b000 0x3c14b000 0x3c143000 0x420291f8 0x420291c2 0x3fca98e0 0x420290b0
3fca89a0: 0x00003800 0x00000011 0x3fca8b30 0x00000001 0x3c171d39 0xffffffff 0x3c1404ac 0x00000002
3fca89c0: 0x3fcaa848 0x3fcaa7d0 0x00000000 0x00000000 0x4202902c 0x00000004 0x420291c2 0x420291f8
3fca89e0: 0x3c143000 0x3c14b000 0x3c14b000 0x3c14b000 0x3fcb8be8 0x3c142000 0x3c141000 0x3c141000
[... 97 lines cut ...]
Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403807a8
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0xe8c
load:0x403cc710,len:0x6ec
load:0x403ce710,len:0x2b24
entry 0x403cc710

### Additional Information

It's strange why the GC report seems to be enlarged after applying for two pieces of memory.

GC: total: 64000, used: 11392, free: 52608, max new split: 139264
 No. of 1-blocks: 200, 2-blocks: 29, max blk sz: 94, max free sz: 3278
>>> a=bytes(20000)
>>> b=bytes(20000)
>>> import bluetooth
>>> m.mem_info()
stack: 1012 out of 14336
GC: total: 139968, used: 122640, free: 17328, max new split: 65536
 No. of 1-blocks: 84, 2-blocks: 13, max blk sz: 4747, max free sz: 531

m24h, 2024-04-12:
If I did not allocate variables 'a' and 'b', or call BLE.active(True) before that, everything seems to be OK.

m24h, 2024-04-12:
A similar issue seems to be happening on the WLAN. 
If more than 20-30k RAM has been allocated before, activing AP may cause panic, STA will report a password error when connecting (the password is actually correct)

agatti, 2024-08-24:
It seems the bug in question is gone?  Even when allocating 140k it still doesn't crash:

```
MicroPython v1.24.0-preview.224.g6c3dc0c0b on 2024-08-22; ESP32C3 module with ESP32C3
Type "help()" for more information.
>>> import micropython as m
>>> m.mem_info()
stack: 516 out of 15360
GC: total: 64000, used: 1872, free: 62128, max new split: 147456
 No. of 1-blocks: 30, 2-blocks: 6, max blk sz: 18, max free sz: 3847
>>> a=bytes(20000)
>>> b=bytes(20000)
>>> c=bytes(20000)
>>> d=bytes(20000)
>>> e=bytes(20000)
>>> f=bytes(20000)
>>> g=bytes(20000)
>>> import bluetooth
>>> m.mem_info()
stack: 516 out of 15360
GC: total: 208000, used: 141840, free: 66160, max new split: 5632
 No. of 1-blocks: 33, 2-blocks: 5, max blk sz: 1251, max free sz: 3749
>>> ble=bluetooth.BLE()
>>> ble.active()
False
>>> ble.active(True)
E (52499) BLE_INIT: Malloc failed
E (52499) BLE_INIT: Malloc failed
E (52549) BLE_INIT: controller init failed

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 116] ETIMEDOUT
>>>
```

jonnor, 2024-09-01:
@m24h can you re-test on latest master, that the issue has been fixed? There are builds available on the Downloads page.