micropython/micropython #14302: I need a support module for a library with camera operation,

jasonccs, 2024-04-14:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) regarding this feature, and didn't find any.

### Description

I need a library with camera operation, which includes camera models such as OV2640, OV5640, OV7725, etc. I hope to have a library that can take photos and obtain video streams. The latest version currently does not have such a library. MicroPython v1.22.2 on 2024-02-22; Generic ESP32 module with ESP32

### Implementation

- [X] I intend to implement this feature and would submit a Pull Request if desirable.
- [X] I hope the MicroPython maintainers or community will implement this feature.
- [X] I would like to [Sponsor](https://github.com/sponsors/micropython#sponsors) development of this feature.

jonnor, 2024-07-03:
Hi @jasonccs and thank you for this feature request.
There are many libraries available for MicroPython which implements support for cameras. A list can be found at https://awesome-micropython.com/#camera 
ESP32 and OV2640 in particular has several alternatives.

mattytrentini, 2024-07-04:
As @jonnor has said, there are some libraries available for this purpose so I think the _issue_ can be considered resolved - but I'll convert this to a discussion since the solution could include a comprehensive set of camera drivers.