micropython/micropython #14307: [webassembly][pyscript] Unable to process over 88Kb of code

WebReflection, 2024-04-15:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

webassemply

### MicroPython version

latest for pyscript

### Reproduction

We bootstrap our virtual env via FS operations and we port that whole operation via strings (as in Python strings) but we reached a memory limit so that everything breaks. Under 60Kb of single *run* we don't have issues but beyond that it's breaking.

### Expected behaviour

MicroPython should be able to run up to 256Kb of code or more to me or we should be able to raise that limit. We tried to explicitly increase the `heapsize` without any success.

### Observed behaviour

`runPython(moreThan88KCode)` fails, it doesn't on Pyodide, it doesn't if we manually remove some line to evaluate from that code.

### Additional Information

This file contains our bootstrap logic which is automatically generated and it travels across workers or projects to be sure our namespace is up and running by the time users' code executes.

It served us well as logic to date and we never had issues with MicroPython but a recent branch that would like to bring more is failing hard on it.

[micropython-limit.txt](https://github.com/micropython/micropython/files/14979983/micropython-limit.txt)

WebReflection, 2024-04-15:
This file, once renamed as `.html` will show the limit. This is without PyScript around.

Remove the line line 47 and see no error ... please note: you need to remove that line, as even commenting it, will still show the *memory out of bound* error, just to be clear it's not the code the issue, rather how much input one can *run*.

[micropython-limit.html.txt](https://github.com/micropython/micropython/files/14980167/micropython-limit.html.txt)

WebReflection, 2024-04-16:
@dpgeorge I don't know if this is useful but in this line maybe there's not enough memory allocated https://github.com/micropython/micropython/blob/53d0050255ef92df5adc9255380f170eb33bd2c9/ports/webassembly/api.js#L134 but I also wonder if this should be `JSFLAGS += -s ALLOW_MEMORY_GROWTH=1` instead https://github.com/micropython/micropython/blob/53d0050255ef92df5adc9255380f170eb33bd2c9/ports/webassembly/variants/pyscript/mpconfigvariant.mk#L1

dpgeorge, 2024-04-16:
Thanks for the report.

The problem is an overflow of the Emscripten C stack.  When using `Module.ccall(...)` and passing an argument (from JS to C) as a type "string", Emscripten will use `stringToUTF8OnStack()` to allocate space for the string on the C side, on the C heap.  This is nothing to do with MicroPython, it's internal Emscripten behaviour.

When passing really big strings from JS to C, the allocation for this string on the C heap fails.  That's the error that you see.

The fix is for MicroPython to explicitly allocate memory for the string that's to be passed from JS to C, and to use `stringToUTF8()` instead of `stringToUTF8OnStack()`.

WebReflection, 2024-04-16:
> This is nothing to do with MicroPython, it's internal Emscripten behaviour.

I trust your opinion and experience but we don't have the issue on Pyodide so maybe they do that behind the scene already? If you find a single minute, and a single minute is enough to explain, please expand on that `3 * 4` `Module._malloc` as it buffles me why that's needed, although maybe it's just to allocate enough memory for those pointers used then via `ccall(...)`?

Thanks!

Despite that clarification, anything *we* can do on our side to prevent that error from happening? 🤔

dpgeorge, 2024-04-17:
> If you find a single minute, and a single minute is enough to explain, please expand on that `3 * 4` `Module._malloc` as it buffles me why that's needed, although maybe it's just to allocate enough memory for those pointers used then via `ccall(...)`?

That allocation is for the proxy to pass back the Python result from C to JS.  It's a small 3-word array that contains the type of the proxy as the first entry (eg int, str, list, dict, function, etc) and then additional information in the other two entries.

I should make these magic numbers constants...

> Despite that clarification, anything _we_ can do on our side to prevent that error from happening?

Probably not.  The only suggestion is not to pass large strings to `runPython`!

> any reason me playing around _heapsize_ never produced a different result?

Because it's a stack overflow, not running out of heap.  Apparently the Emscritpen stack size can only be increased when building via the `-sSTACK_SIZE=xxx` option.