micropython/micropython #14311: docs: machine.UART.irq arguments are incorrect

AsensioL, 2024-04-16:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) and didn't find any that matched.

### Documentation URL

https://docs.micropython.org/en/latest/library/machine.UART.html

### Description

The `machine.UART.irq` entry on the public docs has the following arguments:
`UART.irq(trigger, priority=1, handler=None, wake=machine.IDLE)`

However, looking at _extmod/machine_uart.c_, and in practice, `machine.UART.irq` behaves more like:
`UART.irq(handler=None, trigger=0, hard=False)`.
Note that the position of `handler` is wrong, `priority` and `wake` should be removed and `hard` needs to be added.

I would like to update _docs/library/machine.UART.rst_ to reflect the correct arguments.
But maybe other platforms handle `machine.UART.irq` differently, and I only use MicroPython on STM32 devices.

Please, let me know if that's the case and, if not, I'll proceed to fix and create a merge request.

robert-hh, 2024-04-16:
Thank you for the report. There are already two PRs open about this topic: #14040 and #14041. So this will be fixed.

AsensioL, 2024-04-17:
That work look great. I only searched on the issues section.
Closing this.

closed 2024-04-17 by AsensioL