micropython/micropython #14329: Arduino Nano Connect flash file system not present in 1.23 preview versions

RetiredWizard, 2024-04-19:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

RP2040 port, Arduino Nano RP2040 Connect with RP2040

### MicroPython version

MicroPython v1.23.0-preview.324.gd11ca092f on 2024-04-17; Arduino Nano RP2040 Connect with RP2040

### Reproduction

1. Install any of the 1.23 preview versions from preview.319 - preview.324
2. connect and press ctrl-D

### Observed behaviour

MPY: soft reboot
Traceback (most recent call last):
  File "_boot_fat.py", line 12, in <module>
OSError: [Errno 19] ENODEV
MicroPython v1.23.0-preview.324.gd11ca092f on 2024-04-17; Arduino Nano RP2040 Connect with RP2040
Type "help()" for more information.

### Additional Information

No, I've provided everything above.

RetiredWizard, 2024-04-19:
I did try the flash_nuke.uf2 however it didn't seem to do what I expected because after installing the flash_nuke and trying the preview versions, I then re-installed v1.22 and all my micropython files were still on the flash file system.

scruss, 2024-05-09:
Can confirm same behaviour here. Flash is only accessible via host computer interface (MSC) unless the emulated drive is ejected.

RetiredWizard, 2024-11-04:
I am no longer seeing this issue with the nano connect on version 1.24.0 or preview.6 of 1.25.0.

closed 2024-11-04 by RetiredWizard

dpgeorge, 2024-11-04:
> I am no longer seeing this issue with the nano connect on version 1.24.0 or preview.6 of 1.25.0.

Thanks for confirming that this is fixed.  I'm not sure what was wrong.

I did test 1.23.0-preview.324.gd11ca092f on a Nano Connect but could not reproduce the problem reported here.