micropython/micropython #14339: [RP2] machine.soft_reset() hangs when i2s.deinit() is not called

marecl, 2024-04-20:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

RPI_PICO

### MicroPython version

MicroPython v1.23.0-preview.324.gd11ca092f.dirty on 2024-04-20; Raspberry Pi Pico with RP2040

### Reproduction

1. Initialize I2S device
2. Call `machine.soft_reset()`

### Expected behaviour

machine.soft_reset() resets machine

### Observed behaviour

device hangs, requiring hardware reset or reconnecting USB

### Additional Information

This issue is work-in-progress. I've observed it during development of my application. At this time please treat this like a general observation.
Firmware is slightly customized with added MSC in board configuration. It has it's own issues but I didn't observe any connection.
I've observed that `main.c` calls I2S init on startup, but doesn't deinit it during soft reset.

marecl, 2024-04-21:
Solved in mentioned PR. Waiting for feedback on the issue.

marecl, 2024-04-21:
#14345

dpgeorge, 2024-04-22:
Thanks for the detailed report and reproduction.  I can confirm the issue on a Pico.  And the same issue exists on the stm32 port (tested with a PYBv1.0), and I assume it also exists on the mimxrt port.

The fix would be to enable finalisers on I2S objects for all the ports, following how esp32 does it.  That would also fix other cases of I2S instances being GC'd without `.deinit()` being called on them by the user.

robert-hh, 2024-04-22:
The mimxrt port de-inits I2S on soft reset. But as for enabling finalizers: during soft reset `gc_sweep_all()` is called. I had the impression that this call clears each and every object from the heap, whether created with finalizer or not.

robert-hh, 2024-04-22:
The i2s object contains usually pointers to buffers, allocated with m_new(). Assuming that `gc_sweep_all()` does not free objects in a specific order it may happen, machine_is2_deinit() as part of calling __del__ with it is called after the buffers have already been cleared. 
What happens is  m_free() is called on already free'd object?

dpgeorge, 2024-04-23:
> The mimxrt port de-inits I2S on soft reset

Ah, so it does!  It could instead use finalisers, that would have the same effect.

> during soft reset `gc_sweep_all()` is called. I had the impression that this call clears each and every object from the heap, whether created with finalizer or not.

Yes, `gc_sweep_all()` clears all sections of the GC heap that have been allocated, marking them as free.  As part of this it runs finalisers on those bits of memory that are freed and that have a finaliser associated with them.  So you must allocate the memory with `mp_obj_malloc_with_finaliser()` and have a `__del__` method on that type (otherwise how does the GC know what to call to finalise it?).

> Assuming that `gc_sweep_all()` does not free objects in a specific order it may happen, machine_is2_deinit() as part of calling **del** with it is called after the buffers have already been cleared.
> What happens is m_free() is called on already free'd object?

That's a very good point.  If you call `m_free()` on an object that's already been free'd then an assertion will fail (if assertions are enabled).

So this needs some more thought... maybe we need a `gc_run_all_finalisers()` which executes all finalisers first, and then run `gc_sweep_all()`.

dpgeorge, 2024-08-07:
The rp2 port I2S deinit was fixed in d2e33fe3096eec60d7017b0f17c4ddb6910a4d0b

The stm32 port still needs fixing, and probably should make esp32 and mimxrt port behave the same way (use a deinit function rather than finalisers, due to issues with finalisation mentioned above).