micropython/micropython #14340: Codecov CI action (`coverage`) failing due to API issues

neuschaefer, 2024-04-20:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

unix

### MicroPython version

_(Perhaps this is tracked internally already, but a public issue has the advantage that people can subscribe to it and be notified after the resolution of the problem.)_

The `coverage` CI action is currently failing in many recently submitted/updated pull requests, for example #14332.
The [log output](https://github.com/micropython/micropython/actions/runs/8764664484/job/24054900615?pr=14332) around the error is as follows:

```
[2024-04-20T11:25:54.342Z] ['info'] Pinging Codecov: https://codecov.io/upload/v4?package=github-action-3.1.6-uploader-0.7.3&token=*******&branch=ctrl-l&build=8764664484&build_url=https%3A%2F%2Fgithub.com%2Fmicropython%2Fmicropython%2Factions%2Fruns%2F8764664484%2Fjob%2F24054900615&commit=88acf8a13f718210230a1b2d71b217dc382156db&job=unix+port&pr=14332&service=github-actions&slug=micropython%2Fmicropython&name=&tag=&flags=&parent=
[2024-04-20T11:25:54.342Z] ['verbose'] Passed token was 0 characters long
[2024-04-20T11:25:54.342Z] ['verbose'] https://codecov.io/upload/v4?package=github-action-3.1.6-uploader-0.7.3&branch=ctrl-l&build=8764664484&build_url=https%3A%2F%2Fgithub.com%2Fmicropython%2Fmicropython%2Factions%2Fruns%2F8764664484%2Fjob%2F24054900615&commit=88acf8a13f718210230a1b2d71b217dc382156db&job=unix+port&pr=14332&service=github-actions&slug=micropython%2Fmicropython&name=&tag=&flags=&parent=
        Content-Type: 'text/plain'
        Content-Encoding: 'gzip'
        X-Reduced-Redundancy: 'false'
[2024-04-20T11:25:54.617Z] ['error'] There was an error running the uploader: Error uploading to [https://codecov.io:](https://codecov.io/) Error: There was an error fetching the storage URL during POST: 404 - {'detail': ErrorDetail(string='Unable to locate build via Github Actions API. Please upload with the Codecov repository upload token to resolve issue.', code='not_found')}
[2024-04-20T11:25:54.618Z] ['verbose'] The error stack is: Error: Error uploading to [https://codecov.io:](https://codecov.io/) Error: There was an error fetching the storage URL during POST: 404 - {'detail': ErrorDetail(string='Unable to locate build via Github Actions API. Please upload with the Codecov repository upload token to resolve issue.', code='not_found')}
    at main (/snapshot/repo/dist/src/index.js)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
[2024-04-20T11:25:54.618Z] ['verbose'] End of uploader: 1474 milliseconds
Error: Codecov: Failed to properly upload: The process '/home/runner/work/_actions/codecov/codecov-action/v3/dist/codecov' failed with exit code 255
```

This message in particular stands out:

> Please upload with the Codecov repository upload token to resolve issue.

### Reproduction

- Submit a pull request that triggers CI checks

### Expected behaviour

- CI only shows errors related to the PR.

### Observed behaviour

see above

### Additional Information

- The `coverage` flow in .github/workflows/ports_unix.yml still uses `codecov/codecov-action@v3`, but v4 is available
- The [codecov-action changelog](https://github.com/codecov/codecov-action/blob/main/CHANGELOG.md) mentions as a recent breaking change: "Tokenless uploading is unsuported"
- The `coverage` flow does not seem to pass a codecov token as a secret

dpgeorge, 2024-04-20:
Thanks for raising this issue.

I have noticed this a lot, but also I notice that it does work sometimes, so that's really confusing, because I would think that the codecov settings are correct and it always works, or they are incorrect and it never works.

Still, your PR #14341 to update to codecov-action@v4 seems like a good thing to do.

closed 2024-04-21 by dpgeorge

neuschaefer, 2024-04-22:
The workflow seems to work fine in PRs now, but for the master branch, there's still an error:

```
/home/runner/work/_actions/codecov/codecov-action/v4/dist/codecov -v create-commit --git-service github -Z
info - 2024-04-21 23:50:05,471 -- ci service found: github-actions
debug - 2024-04-21 23:50:05,474 -- versioning system found: <class 'codecov_cli.helpers.versioning_systems.GitVersioningSystem'>
debug - 2024-04-21 23:50:05,476 -- versioning system found: <class 'codecov_cli.helpers.versioning_systems.GitVersioningSystem'>
warning - 2024-04-21 23:50:05,479 -- No config file could be found. Ignoring config.
debug - 2024-04-21 23:50:05,479 -- No codecov_yaml found
debug - 2024-04-21 23:50:05,479 -- Starting create commit process --- {"commit_sha": "f76cf2940288f699b8c6346beabdf4de65ddfa31", "parent_sha": null, "pr": null, "branch": "master", "slug": "micropython/micropython", "token": null, "service": "github", "enterprise_url": null}
Error: Codecov token not found. Please provide Codecov token with -t flag.
Error: Codecov: Failed to properly create commit: The process '/home/runner/work/_actions/codecov/codecov-action/v4/dist/codecov' failed with exit code 1
```

dpgeorge, 2024-04-22:
I have now added `CODECOV_TOKEN` as a secret for this repository in the GitHub settings.  That should get it working.

dpgeorge, 2024-04-22:
Looks like this is now working for the master branch on push.  At least, the previous two pushes have worked.