micropython/micropython #14350: Build micropython for ESP32 ports & ESP-IDF v5.06. failed with static-assert

aso-c, 2024-04-22:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

esp32 port, all ESP32-board variants

### MicroPython version

MicroPython v1.22.2-dirty on 2024-04-18; Generic ESP32 module with ESP32; ESP-IDF v.5.0.6

### Reproduction

make

### Expected behaviour

Build micrpopython port for ESP32, simply make. ;))

### Observed behaviour

Build process failed by the static assert with message:

"Synchronize WIFI_AUTH_XXX constants with the ESP-IDF. Look at esp-idf/components/esp_wifi/include/esp_wifi_types.h"

### Additional Information

I discover this trouble and detect, that Espressif upgrade it's code in all 5.x.y branches.
Now:
- location in file: - **ports/esp32/network_common.c**
- line: 171
- source code:
```
#if ESP_IDF_VERSION > ESP_IDF_VERSION_VAL(5, 1, 1)
_Static_assert(WIFI_AUTH_MAX == 11, "Synchronize WIFI_AUTH_XXX constants with the ESP-IDF. Look at esp-idf/components/esp_wifi/include/esp_wifi_types.h");
#else
_Static_assert(WIFI_AUTH_MAX == 10, "Synchronize WIFI_AUTH_XXX constants with the ESP-IDF. Look at esp-idf/components/esp_wifi/include/esp_wifi_types.h");
#endif
```
- Solition: update code as below:
#if ESP_IDF_VERSION > ESP_IDF_VERSION_VAL(5, 1, 1)
_Static_assert(WIFI_AUTH_MAX == 11, "Synchronize WIFI_AUTH_XXX constants with the ESP-IDF. Look at esp-idf/components/esp_wifi/include/esp_wifi_types.h");
**#elif ESP_IDF_VERSION > ESP_IDF_VERSION_VAL(5, 0, 4)
_Static_assert(WIFI_AUTH_MAX == 11, "Synchronize WIFI_AUTH_XXX constants with the ESP-IDF. Look at esp-idf/components/esp_wifi/include/esp_wifi_types.h");**
#else
_Static_assert(WIFI_AUTH_MAX == 10, "Synchronize WIFI_AUTH_XXX constants with the ESP-IDF. Look at esp-idf/components/esp_wifi/include/esp_wifi_types.h");
#endif

Upd: I'm detect, that in ESP-IDF, branch v5.2.x - WIFI_AUTH_MAX == 13

gampam2000, 2024-04-25:
Hi, i have the same Error with ESP-IDF, branch v5.2 and Micropython v1.22-release branch

I changed it to WIFI_AUTH_MAX == 13 and this specific  error disappeared.

But i get a lot of other errors:
$micropython/ports/esp32/uart.c: In function 'uart_stdout_init':
$micropython/ports/esp32/uart.c:60:34: warning: passing argument 2 of 'uart_hal_get_sclk' from incompatible pointer type [-Wincompatible-pointer-types]
   60 |     uart_hal_get_sclk(&repl_hal, &sclk); // To restore SCLK after uart_hal_init() resets it
      |                                  ^~~~~
      |                                  |
      |                                  uart_sclk_t * {aka soc_periph_uart_clk_src_legacy_t *}
In file included from $micropython/ports/esp32/uart.c:38:
$esp-idf/components/hal/include/hal/uart_hal.h:209:67: note: expected 'soc_module_clk_t *' but argument is of type 'uart_sclk_t *' {aka 'soc_periph_uart_clk_src_legacy_t *'}
  209 | void uart_hal_get_sclk(uart_hal_context_t *hal, soc_module_clk_t *sclk);
      |                                                 ~~~~~~~~~~~~~~~~~~^~~~
In file included from $esp-idf/components/esp_system/include/esp_system.h:12,
                 from $micropython/ports/esp32/mpconfigport.h:10,
                 from $micropython/py/mpconfig.h:91,
                 from $micropython/py/mpstate.h:31,
                 from $micropython/py/runtime.h:29,
                 from $micropython/ports/esp32/uart.c:30:
$micropython/ports/esp32/uart.c:70:24: error: 'uart_periph_signal' undeclared (first use in this function)
   70 |         esp_intr_alloc(uart_periph_signal[MICROPY_HW_UART_REPL].irq,
      |                        ^~~~~~~~~~~~~~~~~~
$esp-idf/components/esp_common/include/esp_err.h:105:30: note: in definition of macro 'ESP_ERROR_CHECK'
  105 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
$micropython/ports/esp32/uart.c:70:24: note: each undeclared identifier is reported only once for each function it appears in
   70 |         esp_intr_alloc(uart_periph_signal[MICROPY_HW_UART_REPL].irq,
      |                        ^~~~~~~~~~~~~~~~~~
$esp-idf/components/esp_common/include/esp_err.h:105:30: note: in definition of macro 'ESP_ERROR_CHECK'
  105 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
[ 83%] Building C object esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/__/__/pins.c.obj
make[3]: *** [esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/build.make:2869: esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/__/uart.c.obj] Error 1
make[3]: *** Waiting for unfinished jobs....
[ 83%] Building C object esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/__/modespnow.c.obj
[ 83%] Building C object esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/__/__/frozen_content.c.obj
make[3]: Leaving directory '$micropython/ports/esp32/build-ESP32_GENERIC_S3'
make[2]: *** [CMakeFiles/Makefile2:5321: esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/all] Error 2
make[2]: Leaving directory '$micropython/ports/esp32/build-ESP32_GENERIC_S3'
make[1]: *** [Makefile:130: all] Error 2
make[1]: Leaving directory '$micropython/ports/esp32/build-ESP32_GENERIC_S3'
make failed with exit code 2, output of the command is in the $micropython/ports/esp32/build-ESP32_GENERIC_S3/log/idf_py_stderr_output_846328 and $micropython/ports/esp32/build-ESP32_GENERIC_S3/log/idf_py_stdout_output_846328
-e See https://github.com/micropython/micropython/wiki/Build-Troubleshooting
make: *** [Makefile:66: all] Error 1
make: Leaving directory '$micropython/ports/esp32'

Did you get it to build? If yes which ESP-IDF version did you use? Any suggestions are appreciated.

gampam2000, 2024-04-25:
I made a complete clean install of esp-idf and micropython because I hade so many errors.

I used the following versions:
$ git clone -b v5.2 --recursive https://github.com/espressif/esp-idf.git
cd esp-idf
esp-idf$ git submodule update --init --depth 1
esp-idf$ sh install.sh esp32,esp32s3,esp32c3
esp-idf$ idf_tools.py install-python-env
esp-idf$ source export.sh
cd ..

$ git clone -b v1.22-release https://github.com/micropython/micropython.git
$ cd micropython
micropython$ git submodule update --init --depth 1
micropython$ make -j2 -C mpy-cross/
micropython$ make -C ports/esp32 BOARD=ESP32_GENERIC_S3

This fails with the above mentioned errors.

gampam2000, 2024-04-26:
I got it working with the esp-idf v 5.2 and micropython master branch.

However with the latest esp-idf master branch I got similar errors to #13197

aso-c, 2024-04-26:
> Did you get it to build? If yes which ESP-IDF version did you use? Any suggestions are appreciated.

In my current projects I'm use ESP-IDF v.5.0.6 now as the main working version, and with this branch & my liltle chahges in micropyton - build was sucessfull.
(Updating to the following branches I'm planned in the future.)

Fresh build now:

```idf.py -D MICROPY_BOARD=ESP32_GENERIC -D MICROPY_BOARD_DIR=[...]/job/ESP/micropython/src/micropython/ports/esp32/boards/ESP32_GENERIC  -B build-ESP32_GENERIC build || (echo -e "See \033[1;31mhttps://github.com/micropython/micropython/wiki/Build-Troubleshooting\033[0m"; false)
Executing action: all (aliases: build)
Running cmake in directory /[...]/job/ESP/micropython/src/micropython/ports/esp32/build-ESP32_GENERIC
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DMICROPY_BOARD=ESP32_GENERIC -DMICROPY_BOARD_DIR=/[...]/job/ESP/micropython/src/micropython/ports/esp32/boards/ESP32_GENERIC -DCCACHE_ENABLE=0 /[...]/job/ESP/micropython/src/micropython/ports/esp32"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /usr/bin/git (found version "2.34.1") 
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/share/esp/espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/share/esp/espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/share/esp/espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
Processing 2 dependencies:
[1/2] espressif/mdns (1.1.0)
[2/2] idf (5.0.6)
-- Project sdkconfig file /[...]/job/ESP/micropython/src/micropython/ports/esp32/build-ESP32_GENERIC/sdkconfig
Loading defaults file /[...]/job/ESP/micropython/src/micropython/ports/esp32/build-ESP32_GENERIC/sdkconfig.combined...
/[...]/job/ESP/micropython/src/micropython/ports/esp32/build-ESP32_GENERIC/sdkconfig.combined:140 CONFIG_BT_NIMBLE_TASK_STACK_SIZE was replaced with CONFIG_BT_NIMBLE_HOST_TASK_STACK_SIZE
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: /usr/share/esp/espressif/python_env/idf5.0_py3.10_env/bin/python (found version "3.10.12") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "micropython" version: v1.22.2-dirty
-- Adding linker script /[...]/job/ESP/micropython/src/micropython/ports/esp32/build-ESP32_GENERIC/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /usr/share/esp/esp-idf-v5.0.6/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /usr/share/esp/esp-idf-v5.0.6/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /usr/share/esp/esp-idf-v5.0.6/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /usr/share/esp/esp-idf-v5.0.6/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /usr/share/esp/esp-idf-v5.0.6/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /usr/share/esp/esp-idf-v5.0.6/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /usr/share/esp/esp-idf-v5.0.6/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /usr/share/esp/esp-idf-v5.0.6/components/soc/esp32/ld/esp32.peripherals.ld
**********************************************************
Including User C Module(s) from file: 
Current list DIR is: /[...]/job/ESP/micropython/src/micropython/py
**********************************************************
Found User C Module(s): 
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_netif esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump espressif__mdns esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main_esp32 mbedtls mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /usr/share/esp/esp-idf-v5.0.6/components/app_trace /usr/share/esp/esp-idf-v5.0.6/components/app_update /usr/share/esp/esp-idf-v5.0.6/components/bootloader /usr/share/esp/esp-idf-v5.0.6/components/bootloader_support /usr/share/esp/esp-idf-v5.0.6/components/bt /usr/share/esp/esp-idf-v5.0.6/components/cmock /usr/share/esp/esp-idf-v5.0.6/components/console /usr/share/esp/esp-idf-v5.0.6/components/cxx /usr/share/esp/esp-idf-v5.0.6/components/driver /usr/share/esp/esp-idf-v5.0.6/components/efuse [... cut ...]
/usr/share/esp/espressif/python_env/idf5.0_py3.10_env/bin/python ../../../../../../../../../usr/share/esp/esp-idf-v5.0.6/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size 4MB --flash_freq 40m 0x1000 build-ESP32_GENERIC/bootloader/bootloader.bin 0x8000 build-ESP32_GENERIC/partition_table/partition-table.bin 0x10000 build-ESP32_GENERIC/micropython.bin
or run 'idf.py -p (PORT) flash'
bootloader  @0x001000    23072  (    5600 remaining)
partitions  @0x008000     3072  (    1024 remaining)
application @0x010000  1692416  (  339200 remaining)
total                  1757952
```

aso-c, 2024-04-26:
> After checking out a different idf-version, did you update the submodules in the esp-idf directory and did you re-run install.sh?

Hi!
I started using version 5.0.6 in my C/C++ work projects before I started study micropython, so I've had ESP-IDF v.5.0.6 installed and properly configured for quite some time before it test.
Other version of ESP-IDF I'm not testing.

aso-c, 2024-04-26:
> is there an esp-idf version & micropython version that works currently?

With other versions of ESP-IDF later than v.5.0.6 - I have not tested micropython, sorry.

projectgus, 2024-04-29:
> is there an esp-idf version & micropython version that works currently?

You can find the list of supported ESP-IDF versions in the esp32 port README:
https://github.com/micropython/micropython/tree/master/ports/esp32/#setting-up-esp-idf-and-the-build-environment

projectgus, 2024-04-29:
(@aso-c  thanks for reporting this! For now the quickest way forward will be to switch to a supported ESP-IDF version. You can have more than one version installed at the same time in different directories, if necessary. We'll keep this issue open until 5.0.6 is supported.)