micropython/micropython #14391: docs: MicroPython Internals / Optimizations Typo

newcomb-luke, 2024-04-28:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) and didn't find any that matched.

### Documentation URL

https://docs.micropython.org/en/latest/develop/optimizations.html#variables

### Description

Extremely minor typo that anyone who modifies the docs for a real reason can fix.

Under the Variables section it says "Local variables on the other hand are are stored on the Python value stack..."

mattytrentini, 2024-04-28:
Thanks @newcomb-luke! I've read the paragraph at least a dozen times and never noticed. :)

It would have been fine to submit a PR yourself by the way; it could be merged on it's own or integrated into a bigger merge..

closed 2024-04-28 by mattytrentini

dpgeorge, 2024-04-29:
Fixed by b1ac266bb5cd0fcfa53539cb3230e7b3f323a09b