micropython/micropython #14413: [webassembly] [pyscript] Unable to create a File from the virtual FileSystem

WebReflection, 2024-05-02:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

webassembly

### MicroPython version

3.4.0; MicroPython v1.23.0-preview.334.g9c7f0659e on 2024-04-25

### Reproduction

```html
<!doctype html>
<html lang="en">
    <head>
        <meta charset="UTF-8">
        <meta name="viewport" content="width=device-width, initial-scale=1.0">
        <script type="module">
            const base = 'https://cdn.jsdelivr.net/npm/@micropython/micropython-webassembly-pyscript@latest';
            const { loadMicroPython } = await import(`${base}/micropython.mjs`);
            const interpreter = await loadMicroPython({ url: `${base}/micropython.wasm` });

            // sync or async has the same No.2 issue
            // async also has the No.1 issue though
            interpreter.runPythonAsync(`
                import js, jsffi

                async def after_fetch(request):
                    name = "page.html"
                    txt = await request.text()

                    # for report purpose, stored as text
                    with open(f"./{name}", "w") as dest:
                        dest.write(txt)
                        dest.close()

                    # but read as generic binary data
                    with open(f"./{name}", "rb") as source:
                        data = source.read()
                        source.close()

                        buffer = js.Uint8Array.new(len(data))
                        for pos, b in enumerate(data):
                            buffer[pos] = b

                        details = jsffi.to_js({"type": "text/html"})

                        # Issue No.2 - stringToUTF8Array expects a string (got symbol)
                        file = js.File.new([buffer], name, details)

                        tmp = js.URL.createObjectURL(file)
                        dest = js.document.createElement("a")
                        dest.setAttribute("download", name)
                        dest.setAttribute("href", tmp)
                        dest.click()
                        js.setTimeout(lambda: js.URL.revokeObjectURL(tmp), 10000)

                # Issue No.1 - this fails: silent "nothingness"
                # await after_fetch(await js.fetch(js.location.href))

                # this works though ... next ...
                js.fetch(js.location.href).then(after_fetch)
            `);
        </script>
    </head>
</html>
```

### Expected behaviour

With just `File` references coming directly from *JS* world, we have zero issues in creating downloads in the wild:

```html
<!doctype html>
<html lang="en">
    <head>
        <meta charset="UTF-8">
        <meta name="viewport" content="width=device-width, initial-scale=1.0">
        <script type="module">
            const base = 'https://cdn.jsdelivr.net/npm/@micropython/micropython-webassembly-pyscript@latest';
            const { loadMicroPython } = await import(`${base}/micropython.mjs`);
            const interpreter = await loadMicroPython({ url: `${base}/micropython.wasm` });
            interpreter.runPython(`
                import js

                def on_change(event):
                    file, = event.currentTarget.files
                    tmp = js.URL.createObjectURL(file)
                    dest = js.document.createElement("a")
                    dest.setAttribute("download", file.name)
                    dest.setAttribute("href", tmp)
                    dest.click()
                    js.setTimeout(lambda: js.URL.revokeObjectURL(tmp), 10000)

                js.document.querySelector('input[type="file"]').onchange = on_change
            `);
        </script>
    </head>
    <body>
        <input type="file">
    </body>
</html>
```

Select a generic file and see it downloaded too.

**However**, when files are pre-stored or added to the virtual File System, no matter which dance we do to convert these, the moment we try to create a new `File` reference, we always get the same error:

```
Aborted(Assertion failed: stringToUTF8Array expects a string (got symbol))
micropython.mjs:1 Uncaught (in promise) RuntimeError: Aborted(Assertion failed: stringToUTF8Array expects a string (got symbol))
    at abort (micropython.mjs:1:11816)
    at assert (micropython.mjs:1:8180)
    at stringToUTF8Array (micropython.mjs:1:19921)
    at stringToUTF8 (micropython.mjs:1:65962)
    at micropython.mjs:1:66768
    at string (micropython.mjs:1:66800)
    at Object.ccall (micropython.mjs:1:67374)
    at Object.get (micropython.mjs:1:100012)
    at js_reflect_construct (micropython.mjs:1:79716)
    at micropython.wasm:0x35fb4
```

### Observed behaviour

We are incapable of downloading files stored in the virtual file system but actually we are incapable of creating any *file* at all due very same error.

### Additional Information

Somehow there's another hidden bug in this report: we are also incapable of `await`ing something top level with no warnings or issues whatsoever or no errors at all anywhere if we don't force the `js.fetch(url).then(...)` instead of just `await js.fetc(url)` first.

dpgeorge, 2024-05-06:
Thanks for the detailed report and reproduction.

I can reproduce both issues.  For "issue no 1", where you say `this fails: silent "nothingness`, that's because you need to `await` the call to `runPythonAsync()`:
```js
await interpreter.runPythonAsync(...)
```

dpgeorge, 2024-05-06:
I have tried to improve error messages (Python exceptions) when using `runPythonAsync()`; see #14434.

As for issue number 2: the problem there is that MicroPython isn't implementing the JavaScript iterator protocol, and the `File.new(...)` call passes in an iterable as the first argument, which fails.  The actual error is that it tries to lookup `Symbol.iterator` and convert that to a string, which gives the cryptic error message.

Iterators are implemented by #14435.

WebReflection, 2024-05-06:
@dpgeorge thanks for looking at both ... about this though:

> that's because you need to await the call to runPythonAsync():

form *JS* point of view, that's never necessary ... not awaited promises just do what they need to do and eventually exit once finished / fulfilled.

I understand you want / need to move the Promise dance only when awaited, which might be convenient, but we need to explain this in our docs because that's not how it works usually on JS side of affair.

Not a big deal, but maybe if it's not too much work one day we'd love to have async stuff just running without needing to explicitly await it.

dpgeorge, 2024-05-06:
> form _JS_ point of view, that's never necessary ... not awaited promises just do what they need to do and eventually exit once finished / fulfilled.

I don't understand how to do it differently, how to make it automatically run the promise?  How does Pyodide work in this respect?  From what I can tell, it also requires that you `await` on the result of calling `runPythonAsync()`.

WebReflection, 2024-05-06:
@dpgeorge like I've said, not a blocker ... a nice to have **if** we want to have it ... here a JS example:

```html
<!DOCTYPE html>
<script type="module">
  const later = async doStuff => new Promise(resolve => {
    setTimeout(
      () => {
        doStuff();
        resolve();
      },
      1000
    );
  });

  let i = 0;
  const stuff = () => { console.log('stuff done', i++); };
  
  // you could await it
  await later(stuff);
  
  // or you can just invoke it
  later(stuff);
</script>
```

You'll see both `stuff done 0` and `stuff done 1` later ... true that JS never stops executing there, but why an *async* program in MicroPython would exit when there are scheduled tasks?

> How does Pyodide work in this respect?

I haven't investigated *how* but it does:

```html
<!doctype html>
<html lang="en">
  <head>
    <meta charset="UTF-8">
    <meta name="viewport" content="width=device-width,initial-scale=1.0">
    <script type="module">
      const base = 'https://cdn.jsdelivr.net/npm/pyodide@latest';
      const { loadPyodide } = await import(`${base}/pyodide.mjs`);
      const interpreter = await loadPyodide();
      const { textContent } = document.querySelector('script[type=pyodide]');
      interpreter.runPythonAsync(textContent);
    </script>
    <script type="pyodide" async>
      i = 0

      async def later():
        import js
        global i
        await js.Promise.new(lambda ok, err: js.setTimeout(ok, 1000))
        print(i)
        i = i + 1

      # you can await it ...
      await later()

      # or you can just call it
      later()
    </script>
  </head>
</html>
```

WebReflection, 2024-05-06:
**random hint** any time an *async* function is invoked you actually add a tasks counter + 1 and only once all tasks have been completed you can exit the program execution.

WebReflection, 2024-05-06:
@dpgeorge last one from me:

> how to make it automatically run the promise?

in *JS*, I am not sure about *Python* there, the moment you invoke an *async* function that already starts the task to complete it ... so I think, in your words, any async method / function automatically run the promise ... if anyone is awaiting it that's the returned value, if someone addresses it (in *JS* at least) that resolves once completed ... if nobody is awaiting it but still it's an explicit invoke, the code gotta do behind the scene what's gonna do and eventually exit the program only after ... if this makes sense ... happy to elaborate more on that, if needed, though!

dpgeorge, 2024-05-07:
Thanks @WebReflection for the info about promises.  MicroPython's behaviour here should be fixed by #14438, ie you no longer need to `await` the result of `runPythonAsync()`.

dpgeorge, 2024-05-07:
This issue is now fixed.

closed 2024-05-07 by dpgeorge