micropython/micropython #14416: docs: Fix typos in README.md

tharuka-pavith, 2024-05-03:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) and didn't find any that matched.

### Documentation URL

https://github.com/micropython/micropython/blob/master/README.md

### Description

I have found a few typos in the README.md. Following is an example,

> Select ports have support for ` _thread` module (multithreading), `socket` and `ssl` for networking, and `asyncio`. Note that only a subset of Python 3 functionality is implemented for the data types and modules.

Here, `Select` should be changed to `Selected`. 
I would like to work on these typos and make a contribution to MicroPython.

ma261065, 2024-06-03:
While I'm sure there are typos in the documentation, you picked a controversial use of English as your example.

Many people now use "select" to imply a carefully chosen subset, such as "The awards were attended by a select group of celebrities". 

One piece of guidance I have seen says:

_Use "select" as an adjective to connote a desirable group of objects. Use "selected" as adjective to connote a neutral (even if purposeful) subsetting of objects from a larger universe._

So while I agree that "selected" would be the better choice here, perhaps we would be better off using more common language such as "**Some** ports have support for  _thread module" and avoid the use of "advertising speak" for technical documentation.

dpgeorge, 2024-06-26:
Yes, the word "select" as used here is correct English.  But it would be better to simplify the language so it doesn't confuse (eg it could be confused as the `select` module, since the previous sentence was listing module names).

dpgeorge, 2024-06-26:
Fixed in 206dc09456727415248c8d10f60b0121d391d38c

closed 2024-06-26 by dpgeorge