micropython/micropython #14429: Incorrect static const data behavior in native module on esp32.

BrianPugh, 2024-05-05:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

esp32

### MicroPython version

MicroPython v1.22.2 on 2024-02-22; Generic ESP32 module with ESP32

### Reproduction

I have a medium-complex native module that performs lossless data compression/decompression.
My code as a native modules has been tested to work on `x64` and `armv6m`. I'm currently testing on esp32 and my code is not longer working; I have narrowed it down to the following:

I have a few `static const` [variables defined here](https://github.com/BrianPugh/tamp/blob/6dc73962c79574309e28fc10cacd84544a665a68/tamp/_c_src/tamp/compressor.c#L24). If I remove the `static`, the code works fine. However, I tried adding `static` to `examples/natmod/features1`, and that code works fine with a `static`. So I have no idea what's going on here.

I found this out because `write_to_bit_buffer` was being invoked with `n_bits=32` at one point, and I traced this down to `huffman_bits[0] == 32`, but obviously the value should be the constant `0x2`.

### Expected behaviour

If my library is operating correctly, it should have the following behavior:

```
>>> import tamp
>>> tamp.compress(b"foo")
b'X\xb3\x04\x18'
>>> tamp.decompress(tamp.compress(b"foo"))
bytearray(b'foo')
```

### Observed behaviour

Accessing the RO data results in incorrect values.

### Additional Information

No, I've provided everything above.

jonnor, 2024-07-03:
Could this be related to https://github.com/micropython/micropython/issues/6592 ?

dpgeorge, 2024-07-15:
> Could this be related to #6592 ?

Yes, highly likely it's the same issue.