micropython/micropython #14430: mpy_ld.py: Unsupported relocation for libm __ieee754_rem_pio2f

jonnor, 2024-05-05:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

esp32/xtensawin and armv7emsp

### MicroPython version

MicroPython 1.22.2 from git

### Reproduction

1. Have a dynamic native module using sinf/cosf functions

Note, you have to include the object files from libm and libgcc manually in the Makefile. This means extracting the .o files from the archives, and then adding them to SRC_O make variable. See this for an example:
https://github.com/emlearn/emlearn-micropython/blob/bd85ae1fade99b34a626c9fcaae0aa4f2ec87803/src/emlfft/Makefile

```
ifeq ($(ARCH), xtensawin)
    LIBM_O := lib_a-sf_cos.o lib_a-sf_sin.o lib_a-ef_rem_pio2.o lib_a-kf_cos.o lib_a-kf_sin.o lib_a-sf_fabs.o lib_a-kf_rem_pio2.o lib_a-sf_scalbn.o lib_a-sf_floor.o
endif
ifeq ($(ARCH), armv7emsp)
    LIBM_O := libm_a-sf_cos.o libm_a-sf_sin.o libm_a-ef_rem_pio2.o libm_a-kf_cos.o libm_a-kf_sin.o libm_a-sf_fabs.o libm_a-kf_rem_pio2.o libm_a-sf_scalbn.o libm_a-sf_floor.o
endif

ifeq ($(ARCH), armv7emsp)
	SOFTFP_O := _arm_addsubdf3.o _arm_muldivdf3.o _arm_muldivdf3.o _arm_truncdfsf2.o
endif
ifeq ($(ARCH), xtensawin)
	SOFTFP_O := _floatsidf.o _muldf3.o _divdf3.o _truncdfsf2.o
endif
```

2. Try to build the module

### Expected behaviour

Linking succeeds without errors, module is functional.

Alternatively: Error and/or documentation provides hints as to the possible reasons why this could happen, and ways of resolving.

### Observed behaviour

Exception when linking. With ARCH=xtensawin

```
  File "/home/jon/projects/micropython/tools/mpy_ld.py", line 572, in do_relocation_text
    assert 0, r_info_type
AssertionError: 11
```

On ARCH=armv7emsp
```
  File "/home/jon/projects/micropython/tools/mpy_ld.py", line 572, in do_relocation_text
    assert 0, r_info_type
AssertionError: 2
```

### Additional Information

arm-none-eabi-gcc --version
arm-none-eabi-gcc (Arch Repository) 13.2.0

xtensa-esp32-elf-gcc --version
xtensa-esp32-elf-gcc (crosstool-NG esp-2022r1) 11.2.0

Error log for xtensawin. ARM is the same, just the relocation value differs

```
python3 /home/jon/projects/micropython/tools/mpy_ld.py '-vvv' --arch xtensawin --qstrs build/emlfft.config.h -o build/emlfft.native.mpy _floatsidf.o _muldf3.o _divdf3.o _truncdfsf2.o lib_a-sf_cos.o lib_a-sf_sin.o lib_a-ef_rem_pio2.o lib_a-kf_cos.o lib_a-kf_sin.o lib_a-sf_fabs.o lib_a-kf_rem_pio2.o lib_a-sf_scalbn.o lib_a-sf_floor.o build/fft.o
qstr vals: __del__, emlfft, new, run
sections:
  00000004 GOT size=140
  00000090 LIT size=204
  0000015c .text size=61
  0000019c .text size=511
  0000039c .text size=531
  000005b0 .text size=164
  00000654 .text size=141
  000006e4 .text size=152
  0000077c .text size=559
  00000000 .rodata size=920
  000009ac .text size=180
  00000a60 .text size=147
  00000af4 .text size=11
  00000b00 .text size=1355
  00000398 .rodata size=56
  0000104c .text size=220
  00001128 .text size=165
  000011d0 .text size=992
  00000000 .bss size=80
  000003d0 .rodata.str1.1 size=168
  000015b0 .text.unlikely size=6
  00000478 .data.rel.ro size=40
  000004a0 .data.rel.ro.local size=24
GOT: 00000004
  00000000 mp_fun_table -> .external+00000000
  00000004 mp_native_qstr_table -> .external.qstr_table+00000000
  00000008 mp_fun_table+0x10c -> .external+00000000
  0000000c __ieee754_rem_pio2f -> .text+0000077c
  00000010 __kernel_cosf -> .text+000009ac
  00000014 __kernel_sinf -> .text+00000a60
  00000018 fabsf -> .text+00000af4
  0000001c __kernel_rem_pio2f -> .text+00000b00
  00000020 scalbnf -> .text+0000104c
  00000024 floorf -> .text+00001128
  00000028 eml_error_valid -> .text+00001214
  0000002c __floatunsidf -> .text+0000015c
  00000030 __muldf3 -> .text+00000280
  00000034 __divdf3 -> .text+0000048c
  00000038 __truncdfsf2 -> .text+000005b0
  0000003c cosf -> .text+00000654
  00000040 sinf -> .text+000006e4
  00000044 eml_fft_fill -> .text+00001248
  00000048 check_extract_array -> .text+000014a0
  0000004c eml_fft_forward -> .text+0000134c
  00000050 .rodata+0x0 -> .rodata+00000000
  00000054 .rodata+0x80 -> .rodata+00000080
  00000058 .rodata+0x2c -> .rodata+000003c4
  0000005c .rodata.str1.1+0x0 -> .rodata.str1.1+000003d0
  00000060 eml_error_strs -> .data.rel.ro.local+000004a0
  00000064 .rodata.str1.1+0x13 -> .rodata.str1.1+000003e3
  00000068 .rodata.str1.1+0x1c -> .rodata.str1.1+000003ec
  0000006c .rodata.str1.1+0x32 -> .rodata.str1.1+00000402
  00000070 .rodata.str1.1+0x45 -> .rodata.str1.1+00000415
  00000074 .data.rel.ro+0x20 -> .data.rel.ro+00000498
  00000078 .data.rel.ro+0x10 -> .data.rel.ro+00000488
  0000007c .data.rel.ro+0x18 -> .data.rel.ro+00000490
  00000080 .data.rel.ro+0x0 -> .data.rel.ro+00000478
  00000084 fft_type -> .bss+00000010
  00000088 fft_locals_dict_table -> .bss+00000000
LIT: 00000090
  00000000 = 00080000
  00000004 = 7ff00000
  00000008 = 001fffff
  0000000c = 38000000
  00000010 = 7fffffff
  00000014 = 3f490fd8
  00000018 = 00000000
  0000001c = 7f7fffff
  00000020 = 4016cbe3
  00000024 = 3fc90f80
  00000028 = 3fc90fd0
  0000002c = 37354443
  00000030 = 37354400
  00000034 = 2e85a308
  00000038 = 43490f80
  0000003c = 3f22f984
  00000040 = 3f000000
  00000044 = 2e85a300
  00000048 = 248d3132
  0000004c = 43800000
  00000050 = 3e900000
  00000054 = 3f800000
  00000058 = 31ffffff
  0000005c = ad47d74e
  00000060 = 310f74f6
  00000064 = b493f27c
  00000068 = 37d00d01
  0000006c = bab60b61
  00000070 = 3d2aaaab
  00000074 = 3e999999
  00000078 = 2f2ec9d3
  0000007c = b2d72f34
  00000080 = 3638ef1b
  00000084 = b9500d01
  00000088 = 3c088889
  0000008c = be2aaaab
  00000090 = 3e2aaaab
  00000094 = 3b800000
  00000098 = 3e000000
  0000009c = 41000000
  000000a0 = 4c000000
  000000a4 = ffff3cb0
  000000a8 = 0da24260
  000000ac = 7149f2ca
  000000b0 = f149f2ca
  000000b4 = 807fffff
  000000b8 = 8da24260
  000000bc = 33000000
  000000c0 = 007fffff
  000000c4 = 54442d18
  000000c8 = 401921fb
_floatsidf.o: .text relocations via .rela.text:
_muldf3.o: .text relocations via .rela.text:
  00000274 LIT -> 00000090
  00000283 LIT -> 00000094
  0000029d LIT -> 00000098
_divdf3.o: .text relocations via .rela.text:
  000003f8 LIT -> 00000090
  0000047d LIT -> 00000090
  0000048f LIT -> 00000094
  000004aa LIT -> 00000098
_truncdfsf2.o: .text relocations via .rela.text:
  000005b3 LIT -> 0000009c
  000005f4 LIT -> 00000094
lib_a-sf_cos.o: .text relocations via .rela.text:
  00000657 LIT -> 000000a0
  0000065f LIT -> 000000a4
  00000664 LIT -> 000000a8
  0000066a LIT -> 000000ac
  00000683 GOT __ieee754_rem_pio2f -> 00000010
Traceback (most recent call last):
  File "/home/jon/projects/micropython/tools/mpy_ld.py", line 1076, in <module>
    main()
  File "/home/jon/projects/micropython/tools/mpy_ld.py", line 1072, in main
    do_link(args)
  File "/home/jon/projects/micropython/tools/mpy_ld.py", line 1043, in do_link
    link_objects(env, len(native_qstr_vals))
  File "/home/jon/projects/micropython/tools/mpy_ld.py", line 833, in link_objects
    do_relocation_text(env, sec.addr, r)
  File "/home/jon/projects/micropython/tools/mpy_ld.py", line 572, in do_relocation_text
    assert 0, (r_info_type, s.name, s.entry)
AssertionError: (11, '__ieee754_rem_pio2f', Container({'st_name': 10, 'st_value': 0, 'st_size': 0, 'st_info': Container({'bind': 'STB_GLOBAL', 'type': 'STT_NOTYPE'}), 'st_other': Container({'visibility': 'STV_DEFAULT'}), 'st_shndx': 'SHN_UNDEF'}))

```

jonnor, 2024-05-05:
I am aware that I am pushing the mpy_ld.py / dynamic module support quite hard here, compared to usual. But some tips as to what might be the issue here would be very welcomed. So that maybe we can improve the support. 

I am providing here the .o files for the `__ieee754_rem_pio2f` (for xtensawin and armv7emsp). I have poked a bit with readelf/objdump, but don't really know how to interpret the relocations - and why this symbol is different from any others...
[micropython-reloc-rem_pio2.zip](https://github.com/micropython/micropython/files/15212504/micropython-reloc-rem_pio2.zip)

jonnor, 2024-05-05:
Renamed issue to highlight relevance. I am pretty sure this would happen whenever sin/cos or sinf/cosf are used. It would happen also if there just exists some code calling these functions - even these code paths are not actually used. Because there seems to be no dead-code stripping done by mpy_ld.py.

agatti, 2024-08-18:
@jonnor Does this patch the issue for Xtensa/Xtensawin?

```diff
--- mpy_ld.py.orig      2024-08-18 20:44:32.381793792 +0200
+++ mpy_ld.py   2024-08-18 20:52:20.521190910 +0200
@@ -69,6 +69,7 @@
 R_386_GOTOFF = 9
 R_386_GOTPC = 10
 R_ARM_THM_CALL = 10
+R_XTENSA_ASM_EXPAND = 11
 R_XTENSA_DIFF32 = 19
 R_XTENSA_SLOT0_OP = 20
 R_ARM_BASE_PREL = 25  # aka R_ARM_GOTPC
@@ -562,7 +563,7 @@
         reloc = addr - r_offset
         reloc_type = "xtensa_l32r"

-    elif env.arch.name == "EM_XTENSA" and r_info_type in (R_XTENSA_DIFF32, R_XTENSA_PDIFF32):
+    elif env.arch.name == "EM_XTENSA" and r_info_type in (R_XTENSA_DIFF32, R_XTENSA_PDIFF32, R_XTENSA_ASM_EXPAND):
         if s.section.name.startswith(".text"):
             # it looks like R_XTENSA_[P]DIFF32 into .text is already correctly relocated
             return
```

vshymanskyy, 2024-08-28:
Yes, also bumping into this while [Compiling WebAssembly to MPY](https://github.com/orgs/micropython/discussions/15702).

The suggested patch did help in my case

agatti, 2024-08-28:
If I have something I can easily build on my machine I can also give a shot at fixing the other ARM relocation handling :)

vshymanskyy, 2024-08-28:
@agatti  `wasm2mpy` should be easy to get started if you're on linux:
https://github.com/vshymanskyy/wasm2mpy

I'd be happy to guide you if needed

agatti, 2024-08-28:
@vshymanskyy Thanks, I've just tried it using wabt from the Arch package repository, but it fails to build.  Not sure what to put in WABT_DIR as the binary files are all under /usr.  Leaving that out building apps fails with missing struct entries, but the header files are in /usr/include/wabt/:

```
CC runtime/runtime.c
runtime/runtime.c:10:1: error: unknown type name 'w2c_wasm'
   10 | w2c_wasm module;
      | ^~~~~~~~
runtime/runtime.c: In function 'setup':
runtime/runtime.c:15:5: error: implicit declaration of function 'w2c_wasm_setup'; did you mean 'Z_wasmZ_setup'? [-Wimplicit-function-declaration]
   15 |     w2c_wasm_setup(&module);
      |     ^~~~~~~~~~~~~~
      |     Z_wasmZ_setup
runtime/runtime.c: In function 'loop':
runtime/runtime.c:22:5: error: implicit declaration of function 'w2c_wasm_loop'; did you mean 'Z_wasmZ_loop'? [-Wimplicit-function-declaration]
   22 |     w2c_wasm_loop(&module);
      |     ^~~~~~~~~~~~~
      |     Z_wasmZ_loop
# ....
runtime/runtime.c:86:5: error: implicit declaration of function 'w2c_app_0x5Fstart' [-Wimplicit-function-declaration]
   86 |     w2c_app_0x5Fstart(&module);
      |     ^~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
make: *** [/home/agatti/src/micropython/py/dynruntime.mk:135: build/runtime/runtime.o] Error 1
```

Do I need the WABT source tree instead and point the variable to its root directory?

agatti, 2024-08-28:
Ok, I made it work using a custom-built wabt toolchain.  Unfortunately I cannot reproduce the bug @jonnor reported for armv7m, I only get errors for missing symbols and not for unhandled relocations (the failing assert).

vshymanskyy, 2024-08-28:
Once I start adding libgcc for armv7m, I'm also hitting invalid relocations and other crashes.

agatti, 2024-08-28:
Right, can you please put those files in a zip archive and attach it here? :)

agatti, 2024-08-28:
Ok, for armv6m once I copied `_udivsi3.o` and `_dvmd_tls.o` from /usr/lib/gcc/arm-none-eabi/9.2.1/thumb/v6-m/nofp/libgcc.a and applied this patch to mpy_ld.py:

```diff
diff --git a/tools/mpy_ld.py b/tools/mpy_ld.py
index 45aadadd5..6ba411ca8 100755
--- a/tools/mpy_ld.py
+++ b/tools/mpy_ld.py
@@ -330,7 +330,7 @@ def build_got_generic(env):
         for r in sec.reloc:
             s = r.sym
             if not (
-                s.entry["st_info"]["bind"] == "STB_GLOBAL"
+                s.entry["st_info"]["bind"] in ("STB_GLOBAL", "STB_WEAK")
                 and r["r_info_type"] in env.arch.arch_got
             ):
                 continue
@@ -711,7 +711,7 @@ def load_object_file(env, felf):
             if shndx in sections_shndx:
                 # Symbol with associated section
                 sym.section = sections_shndx[shndx]
-                if sym["st_info"]["bind"] == "STB_GLOBAL":
+                if sym["st_info"]["bind"] in ("STB_GLOBAL", "STB_WEAK"):
                     # Defined global symbol
                     if sym.name in env.known_syms and not sym.name.startswith(
                         "__x86.get_pc_thunk."
```

then mpy files for `armv6m` do build fine but they are more or less all the same size so I'm not sure if that's enough?

vshymanskyy, 2024-08-28:
Probably you need to run `make clean all ARCH=... APP=...`

I'm currently a bit busy but i'll get back in 1.5h

agatti, 2024-08-28:
No worries, `make clean all ...` did the trick.  I am at least able to build `assemblyscript.mpy` once I copy all necessary files (and there's another bug in there if you put too many files as the trampoline needs to be rewritten for larger offsets!).  Working on the rest now.

agatti, 2024-08-28:
Regarding the assertion on the unhandled relocation, it's a R_ARM_ABS32 relocation that is currently supported only on .data segments but not .text/.rodata.  The fix is easy (take the existing word in the file, add the relocation value, find the target it points to and if it's a `STT_FUNC` symbol that's a thumb function turn bit #0 on), but I can continue to look at it tomorrow.

vshymanskyy, 2024-08-28:
Yes, with armv6m I got to a point where this fails:
```
  File "tools/mpy_ld.py", line 93, in asm_jump_thumb
    assert b_off >> 11 == 0 or b_off >> 11 == -1, b_off
```

Probably I was doing something wrong yesterday, I was hitting different types of errors indefinitely

vshymanskyy, 2024-08-28:
With xtensa/ESP8266, I got stuck with this: https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/lib/readme.md
So, relevant files are not even present in `libgcc.a` whipped with the toolchain.

agatti, 2024-08-29:
> Yes, with armv6m I got to a point where this fails:
> 
> ```
>   File "tools/mpy_ld.py", line 93, in asm_jump_thumb
>     assert b_off >> 11 == 0 or b_off >> 11 == -1, b_off
> ```

Sorry for getting back to you this late, was busy with other stuff...  Anyway, that should be fixed with this (not tested on HW, when disassembling a MPY file with ghidra the offsets look OK):

```diff
 def asm_jump_thumb(entry):
-    # Only signed values that fit in 12 bits are supported
     b_off = entry - 4
-    assert b_off >> 11 == 0 or b_off >> 11 == -1, b_off
-    return struct.pack("<H", 0xE000 | (b_off >> 1 & 0x07FF))
+    if b_off >> 11 != 0 and b_off >> 11 != -1:
+        # sub   sp, #4
+        # push  {r6, r7}
+        # ldr   r7, [pc, #8]
+        # add   r6, pc, #0
+        # add   r7, r6
+        # str   r7, [sp, #8]
+        # pop   {r6, r7}
+        # pop   {pc}
+        # .word entry - 8
+        return struct.pack(
+            "<HHHHHHHHI", 0xB081, 0xB4C0, 0x4F02, 0xA600, 0x4437, 0x9702, 0xBCC0, 0xBD00, b_off - 4
+        )
+    return struct.pack("<H9H", 0xE000 | (b_off >> 1 & 0x07FF), *([0xDE00] * 9))
```

I'm not as familiar with ARM's EABI calling convention as I used to be, so maybe R6 and R7 don't need to be saved and restored in this case but that can't hurt I guess.  The extra R_ARM_ABS32 relocations need to still be sorted out, but that's for tomorrow!

vshymanskyy, 2024-08-29:
@agatti thank you, this runs on RPI PICO:
```log
🚀 AssemblyScript is running!
🦀 Rust is running!
🤩 C++ is running!
⚙ WAT is running!
⚡ Zig is running!
```

I'll run more tests later

vshymanskyy, 2024-08-30:
Got another relocation error:

```
make clean all APP=coremark ARCH=armv6m
...
  File "/data/_Business/wasm2mpy/tools/micropython-1.23.0/tools/mpy_ld.py", line 584, in do_relocation_text
    assert 0, r_info_type
AssertionError: 2
```

I pushed all the obj files to `wasm2mpy` repo

agatti, 2024-08-30:
@vshymanskyy glad to know the new trampoline worked :)

The relocation error you see is the one @jonnor mentioned, and the one I plan to sort out today (R_ARM_ABS32).

vshymanskyy, 2024-09-04:
Thanks. Looks like both your and my build (with the supplied fix) is just locking up the device.
This could be due to a different issue though (i.e. specific to my WebAssembly conversion step). It would be best if @jonnor can check it with his scenario.
Also, if you can provide be with a set of your runtime object files, i could play with other aspects of the build to (maybe) get it working.

Also, @agatti maybe you could comment on this:
> With xtensa/ESP8266, I got stuck with this: https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/lib/readme.md
So, relevant files are not even present in libgcc.a shipped with the toolchain.

I have currently no idea on how to progress

agatti, 2024-09-04:
The object files I used to build `coremark.mpy` are here: [libgcc-armv6m.zip](https://github.com/user-attachments/files/16880570/libgcc-armv6m.zip) - I've taken them out from `/usr/lib/gcc/arm-none-eabi/14.1.0/thumb/v6-m/nofp/libgcc.a` provided by Arch Linux's `extra/arm-none-eabi-gcc` package.  Still, I'll see if I can get a cheap RPi board here to eventually try that out with a JTAG debugger attached.

For the ESP8266 bit, I admit I don't have much experience with that but I can see two possible solutions.

If you have object files that reference the symbols present in ROM, you can patch `mpy_ld.py` to fill in the new jump address for those symbols (must be PC-relative) when they are encountered in `R_XTENSA_32` relocations.  The ROM addresses are in `$SDKROOT/ld/eagle.rom.addr.v6.ld`.

The other option would be to create simple Xtensa assembler functions that only contain the jump opcode to the target function's address in the ROM, then compile it with `xtensa-lx106-elf-as` and bring it in your build.  So, for example if you only want to provide `memcpy` and `memset` you can compile this into an object file and link it as part of your MPY file (WARNING: UNTESTED - I'll give it a try tomorrow morning):

```asm
	.file   "stdlib.s"
	.text
	
	.align  4

	.global memcpy
	.type   memcpy, @function
memcpy:
	j       0x4000DF48
	.size   memcpy, .-memcpy

	.align  4       // Not sure if this is needed after the first one but it won't hurt.
	
	.global memset
	.type   memset, @function
memset:
	j       0x4000E190
	.size   memset, .-memset
```

`xtensa-lx106-elf-as` will generate `R_XTENSA_SLOT0_OP` relocations for each of the final addresses, which `mpy_ld.py` should already handle.

Again, none of these approaches have been tested, but rather than starting to patch `mpy_ld.py`, I'd do some experiments with the hand-written trampolines file first.  Once you get a working .mpy file you can get the code block offset via `mpy-tool.py -dx $FILE` (the first blue byte in the colourised output), then cut that bit away from the file and use either `xtensa-lx106-elf-objdump` (cannot remember the flags to let it disassemble a raw binary) to see what's going on or, what I use, open the .mpy file with a build of [Ghidra](https://github.com/nationalsecurityagency/ghidra) taken from Git master.  For Ghidra you can just enter the start offset and pick the Xtensa CPU for disassembly when opening the file for disassembly.  Just don't leave 0 as a starting code address, but pick something like 0x100000 instead (makes it easier to see if offsets have been patched or left alone).

Hope that helps, I'll try this out tomorrow anyway and see how far I get :)

vshymanskyy, 2024-09-05:
Thanks much for the detailed explanation, it makes much more sense now.

The attempt to build Coremark with your gcc runtime failed miserably,
as my `arm-none-eabi-gcc (15:12.2.rel1-1) 12.2.1 20221205` (obviously) does it's own compilation magic
and references a different set of functions :grinning:

agatti, 2024-09-05:
I've just tried my version of coremark for esp8266 (MicroPython built using default configuration):

```
MicroPython v1.23.0 on 2024-09-05; ESP module with ESP8266
Type "help()" for more information.
>>> import coremark
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
MemoryError: memory allocation failed, allocating 18880 bytes for native code
>>> 
```

I need to search in the lab if I have more ESP8266 boards and try with those, but cannot make any guarantees.

Anyway, the ROM softfp code comes from https://gist.github.com/agatti/2b486f925ec535ecf4d475cc96ac881d - I haven't tested the shorter variant though.  The MPY module I built is here: 
[coremark_esp8266.zip](https://github.com/user-attachments/files/16898890/coremark_esp8266.zip)

vshymanskyy, 2024-09-05:
Thank you for a practical example. I think this error makes sense.
Even when importing `wat` (the smallest module), I'm getting the same error:
```
MemoryError: memory allocation failed, allocating 4344 bytes for native code
```

It works for really tiny modules though, so I think the main issue is the lack of IRAM on ESP8266. It only has 32K I believe and this could probably be already occupied by something else.

vshymanskyy, 2024-09-05:
I tried putting this into `boot.py`:
```py
import esp, gc
gc.collect()
esp.set_native_code_location(200*4096, 25*4096)
```

There's no longer a `MemoryError`, but ESP8266 crashes and reboots

agatti, 2024-09-05:
Great!  Can you please let it print something before it attempts to perform any floating point operation that are done through the ROM code?  Just to be sure the calls actually go through - I do have an idea on how to rewrite the trampolines to get a few instructions smaller so I may try that out tomorrow with a JTAG debugger attached to the ESP8266 :)

vshymanskyy, 2024-09-05:
On esp8266, i have a feeling that something fishy is happening, the modules are misbehaving in various weird ways, This could indicate more bugs in `mpy_ld`.

I'll try to create a simpler example which won't require allocating the full wasm memory page (64Kb), and will do some prints and fp ops.

vshymanskyy, 2024-09-06:
On esp8266, I have a feeling that something fishy is happening, the modules are misbehaving in various weird ways, This could indicate more bugs in `mpy_ld`.
Coremark is **pretending to run**, but I believe it just freezes or corrupts the memory.
I think i'll try to narrow it down to a minimal example and file a separate issue.

For `wasm2mpy`, I'll try to create a simpler example which won't require allocating the full wasm memory page (64Kb), and will do some prints and fp ops.

vshymanskyy, 2024-09-09:
I have created a simplified test for sin/cos functions:
[sin_cos_test.zip](https://github.com/user-attachments/files/16932935/sin_cos_test.zip)

Seems to work with the patch from https://github.com/micropython/micropython/issues/14430#issuecomment-2326857677

WASM/Coremark is still not running on `armv6m`, for an unknown reason

agatti, 2024-09-09:
@vshymanskyy colour me surprised, that patch wasn't supposed to work :)

Anyway, I got an RP2040 and I tried building the coremark mpy.  It imports without errors but doesn't do anything.  Is there any particular function I need to call on the object?

On both the pre-built mpy file in your sin/cos test and on the coremark mpy I built, invoking `dir()` on the imported module object is broken - it works on other platforms.

vshymanskyy, 2024-09-09:
Try `coremark.setup()`

agatti, 2024-09-09:
Thanks, although right now I have bigger problems on my end to solve for that - the board crashes when run connected to a black magic probe:

```
(gdb) tar ext /dev/ttyACM0
Remote debugging using /dev/ttyACM0
(gdb) monitor auto_scan
Target voltage: Unknown
JTAG scan found no devices, trying SWD!
Available Targets:
No. Att Driver
 1      RP2040 M0+
 2      RP2040 M0+
 3      RP2040 Rescue (Attach to reset) 
(gdb) attach 1
Attaching to program: /home/agatti/src/micropython/ports/rp2/build-RPI_PICO_W/firmware.elf, Remote target
mp_hal_stdin_rx_chr () at /micropython/ports/rp2/mphalport.c:82
warning: 82     /micropython/ports/rp2/mphalport.c: No such file or directory
(gdb) c
Continuing.

Program received signal SIGTRAP, Trace/breakpoint trap.
HardFault_Handler ()
    at /micropython/lib/pico-sdk/src/rp2_common/pico_standard_link/crt0.S:98
warning: 98     /micropython/lib/pico-sdk/src/rp2_common/pico_standard_link/crt0.S: No such file or directory
(gdb) bt
#0  HardFault_Handler ()
    at /micropython/lib/pico-sdk/src/rp2_common/pico_standard_link/crt0.S:98
#1  <signal handler called>
#2  0x1004bafc in memp_malloc (type=<optimized out>)
    at /micropython/lib/lwip/src/core/memp.c:343
#3  0x1004bb24 in memp_free (type=type@entry=MEMP_SYS_TIMEOUT, 
    mem=<optimized out>) at /micropython/lib/lwip/src/core/memp.c:440
#4  0x10051d28 in sys_check_timeouts ()
    at /micropython/lib/lwip/src/core/timeouts.c:388
#5  0x1003cdfa in mp_network_soft_timer_callback (self=<optimized out>)
    at /micropython/ports/rp2/mpnetworkport.c:107
#6  0x10039ee4 in soft_timer_handler ()
    at /micropython/shared/runtime/softtimer.c:93
#7  0x1003d162 in PendSV_Handler () at /micropython/ports/rp2/pendsv.c:101
#8  <signal handler called>
#9  mp_event_wait_indefinite () at /micropython/py/scheduler.c:265
#10 0x1003ca20 in mp_hal_stdin_rx_chr ()
    at /micropython/ports/rp2/mphalport.c:95
#11 0x1003974a in readline (line=line@entry=0x20041fb0, 
    prompt=<optimized out>) at /micropython/shared/readline/readline.c:560
#12 0x10039cb4 in pyexec_friendly_repl ()
    at /micropython/shared/runtime/pyexec.c:606
#13 0x1003c850 in main (argc=<optimized out>, argv=<optimized out>)
    at /micropython/ports/rp2/main.c:208
(gdb) 
```

I'll see if I can find another SWD debugger around here, or I'll have to get another one somehow :|

agatti, 2024-09-10:
@vshymanskyy I managed to let the debugger work (of course that had to happen *right after* the notification the new SWD/JTAG probe shipped, oh well).

So, what shows up in gdb is the native code runs, then the bytecode vm kicks in, and somehow it ends trashing both the stack and the program counter, with the MCU executing the address of one of the bytecode handlers as if it were Arm opcodes.

```
(gdb) bt

#0  HardFault_Handler () at /micropython/lib/pico-sdk/src/rp2_common/pico_standard_link/crt0.S:98
#1  <signal handler called>
   0xfffffff9:
Cannot access memory at address 0xfffffff8
#2  0x700a3128 in ?? ()
#3  0x100544d0 in __func__.0 ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) x/8a $sp

0x20041c90:     (R0) 0x0     (R1) 0x700a3128      (R2) 0x0     (R3) 0x0
0x20041ca0:     (R12) 0xfffffc01      (Return address) 0x100544d0 <entry_table.1>      (Program counter) 0x700a3128      (Cpu state) 0x41000000
```

To figure out exactly where this happens, rather than stepping in for 30 minutes each time, if you have a debug probe you can attach to the RP2040 you can insert breakpoint opcodes to act as checkpoints in the debugger to see where you're at.

The `BKPT` Thumb instruction does exactly that, also taking an 8 bit value as a marker (0xBExx with xx being the marker byte).  I don't know if gdb can take over and continue after that, but if it doesn't and enters into `HardFault_Handler`, you can see the offending instruction with `x/i $sp + 24` and see if it's one of your BKPTs.

It sucks, I know.  I guess the alternative is to write your own handler for the `DebugMon` exception, but last time I did something like on Arm that was 10+ years ago and I don't remember if just an empty handler will work or not.  https://developer.arm.com/documentation/ddi0419/e/?lang=en has the answer but I can look at that in a couple of days from now :(

vshymanskyy, 2024-09-11:
I have created a PoC for loading `.a` files directly and resolving object files dependencies:
https://github.com/vshymanskyy/wasm2mpy/blob/main/findobj.py

I believe this can be later reused to directly implement this functionality in `mpy_ld`:
```sh
python3 findobj.py --arch=armv6m /usr/lib/arm-none-eabi/newlib/thumb/v6-m/nofp/libm.a /usr/lib/gcc/arm-none-eabi/12.2.1/thumb/v6-m/nofp/libgcc.a sin cos tan
libgcc.a => divdf3.o
libm.a => lib_a-s_fabs.o
libgcc.a => ledf2.o
libm.a => lib_a-k_rem_pio2.o
libgcc.a => eqdf2.o
libgcc.a => muldf3.o
libgcc.a => _udivsi3.o
libgcc.a => _arm_cmpdf2.o
libgcc.a => gedf2.o
libm.a => lib_a-k_cos.o
libgcc.a => _divsi3.o
libgcc.a => _dvmd_tls.o
libgcc.a => adddf3.o
libm.a => lib_a-s_floor.o
libm.a => lib_a-s_sin.o
libm.a => lib_a-k_tan.o
libm.a => lib_a-s_tan.o
libgcc.a => subdf3.o
libm.a => lib_a-k_sin.o
libgcc.a => fixdfsi.o
libgcc.a => floatsidf.o
libm.a => lib_a-s_cos.o
libgcc.a => _clzsi2.o
libm.a => lib_a-e_rem_pio2.o
libm.a => lib_a-s_scalbn.o
```

vshymanskyy, 2024-09-14:
@jonnor while you're here, maybe you could check if this works for you: https://github.com/micropython/micropython/issues/14430#issuecomment-2326857677

jonnor, 2024-09-14:
@vshymanskyy sorry I did not manage to get to testing these fixes today. Had to prioritize testing https://github.com/micropython/micropython/issues/6769 - since that blocks basically any usage of native modules... I will be away for holiday for one week, will check it after that.

vshymanskyy, 2024-09-14:
Thanks for the heads-up!
Have a good holiday 😀

dpgeorge, 2025-03-17:
This is now supported, see #15838  and commit 51976110e2da32b7a5b7035b7d5f17ab49bbd40e

jonnor, 2025-03-21:
I have updated emlearn-micropython to use the new linking support. It looks to fix almost everything. But the relocation issue reported here remains on ARCH=armv6m. I managed to reduce it to a minimal example - a floating point division triggers the same kind of exception as initially reported.

```
  File "/home/jon/projects/micropython/tools/mpy_ld.py", line 707, in do_relocation_text
    assert 0, r_info_type
           ^
AssertionError: 2
```

The other ARM or Xtensawin architectures are no problem. Also, this does not happen with all GCC versions. Here are some I have tested.

```
Ubuntu 24.04. Official package (13.2.1)  FAIL
Ubuntu 24.04. ARM gcc 13.3 FAIL
Ubuntu 24.04. ARM gcc 14.2 FAIL
Arch Linux. Arch distro package.  arm-none-eabi-gcc-14.1.0-1  OK
Arch Linux. ARM gcc from   gcc-arm-none-eabi-bin-14.2.rel1-1 FAIL
```

Perhaps you have an idea how to resolve this @agatti  ? 

Below is the complete build output (with a bit more details added to the assert), the module code and Makefile.
Can also be found at https://github.com/emlearn/emlearn-micropython/tree/minimal-reproduce-armv6m-reloca

```
(venv) [jon@jon-thinkpad emlearn-micropython]$ rm -rf dist && make dist ARCH=armv6m
make -C src/emlearn_arrayutils/ ARCH=armv6m MPY_DIR=/home/jon/projects/micropython  V=1 clean dist
make[1]: Entering directory '/home/jon/projects/emlearn-micropython/src/emlearn_arrayutils'
/bin/rm -rf build emlearn_arrayutils.mpy .mpy_ld_cache
/bin/mkdir -p build/
GEN build/emlearn_arrayutils.config.h
python3 /home/jon/projects/micropython/tools/mpy_ld.py '-vvv' --arch armv6m --preprocess -o build/emlearn_arrayutils.config.h modarrayutils.c
CC modarrayutils.c
arm-none-eabi-gcc -I. -I/home/jon/projects/micropython -std=c99 -Os -Wall -Werror -DNDEBUG -DNO_QSTR -DMICROPY_ENABLE_DYNRUNTIME -DMP_CONFIGFILE='<build/emlearn_arrayutils.config.h>' -fpic -fno-common -U_FORTIFY_SOURCE  -mthumb -mcpu=cortex-m0 -DMICROPY_FLOAT_IMPL=MICROPY_FLOAT_IMPL_NONE  -o build/modarrayutils.o -c modarrayutils.c
LINK build/modarrayutils.o
python3 /home/jon/projects/micropython/tools/mpy_ld.py '-vvv' --arch armv6m --qstrs build/emlearn_arrayutils.config.h -l/usr/lib/gcc/arm-none-eabi/14.2.1/thumb/v6-m/nofp/libgcc.a -l/usr/arm-none-eabi/lib/thumb/v6-m/nofp/libm.a -o build/emlearn_arrayutils.native.mpy build/modarrayutils.o
qstr vals: divide
Loading /usr/lib/gcc/arm-none-eabi/14.2.1/thumb/v6-m/nofp/libgcc.a
Loading /usr/arm-none-eabi/lib/thumb/v6-m/nofp/libm.a
using /usr/lib/gcc/arm-none-eabi/14.2.1/thumb/v6-m/nofp/libgcc.a:_fixunssfsi.o
using /usr/lib/gcc/arm-none-eabi/14.2.1/thumb/v6-m/nofp/libgcc.a:divsf3.o
using /usr/lib/gcc/arm-none-eabi/14.2.1/thumb/v6-m/nofp/libgcc.a:fixsfsi.o
using /usr/lib/gcc/arm-none-eabi/14.2.1/thumb/v6-m/nofp/libgcc.a:subsf3.o
using /usr/lib/gcc/arm-none-eabi/14.2.1/thumb/v6-m/nofp/libgcc.a:_arm_cmpsf2.o
using /usr/lib/gcc/arm-none-eabi/14.2.1/thumb/v6-m/nofp/libgcc.a:_clzsi2.o
using /usr/lib/gcc/arm-none-eabi/14.2.1/thumb/v6-m/nofp/libgcc.a:lesf2.o
using /usr/lib/gcc/arm-none-eabi/14.2.1/thumb/v6-m/nofp/libgcc.a:gesf2.o
using /usr/lib/gcc/arm-none-eabi/14.2.1/thumb/v6-m/nofp/libgcc.a:eqsf2.o
sections:
  00000008 .text size=116
  0000007c .data.rel.ro size=12
  00000088 .text.__fixunssfsi size=48
  000000b8 .text.__aeabi_fdiv size=556
  000002e4 .rodata.__aeabi_fdiv size=124
  00000360 .text.__aeabi_f2iz size=64
  000003a0 .text.__aeabi_fsub size=1156
  00000824 .text size=116
  00000898 .text size=60
  000008d4 .text.__lesf2 size=144
  00000964 .text.__gesf2 size=144
  000009f4 .text.__eqsf2 size=80
  00000a44 GOT size=8
GOT: 00000a44
  00000000 mp_fun_table -> .external+00000000
  00000004 mp_native_qstr_table -> .external.qstr_table+00000000
build/modarrayutils.o: .text relocations via .rel.text:
  00000028 __aeabi_fdiv -> 000000b9
  0000002c __aeabi_f2uiz -> 00000089
  00000038 _GLOBAL_OFFSET_TABLE_ -> 00000a44
  0000003c mp_fun_table -> 00000000
[... 70 lines cut ...]

DIST_DIR := ../../dist/$(ARCH)_$(MPY_ABI_VERSION)

# enable linking of libm etc
LINK_RUNTIME=1

# Name of module
MOD = emlearn_arrayutils

# Source files (.c or .py)
SRC = modarrayutils.c

# Releases
DIST_FILE = $(DIST_DIR)/$(MOD).mpy
$(DIST_DIR):
	mkdir -p $@

$(DIST_FILE): $(MOD).mpy $(DIST_DIR)
	cp $< $@

# Include to get the rules for compiling and linking the module
include $(MPY_DIR)/py/dynruntime.mk

dist: $(DIST_FILE)
```

jonnor, 2025-03-21:
Previously I did try the fix in https://github.com/micropython/micropython/issues/14430#issuecomment-2326857677 - but I then got crashes when running on device. So that does not seem like the right fix

agatti, 2025-03-21:
TL;DR: The armv6m object files reference those functions in an unsupported manner (see the explicit limitation on `example/natmod/features2` not being part of the CI process for that platform).

With that out of the way, all the other platforms reference soft-float functions as relative offsets to a common base address.  `mpy_ld.py` then just embeds the correct offset in the opcodes in question at symbol resolution time, since it already collected all known external references, put all them up together in a single chunk and recorded the starting offset to each of them.  At runtime the base address is then set by the native code loader, execution goes to the jump trampoline and off you go.

On armv6m those references end up pointing to *absolute* addresses, so `mpy_ld.py` has no idea on how to handle them, as the final symbol address cannot be known at linking time, but only at runtime.  I assume `ld` performs some opcode rewriting (as part of the linking optimisation pass) to work around this, but that's not what `mpy_ld.py` is actually able to do.  A possible solution to this problem is to provide an extra section in MPY files containing the relative offset for each referenced symbol, together to all locations where they are referenced from, and embedding the correctly computed address into the appropriate opcodes (since the offset was loaded from the MPY file and you know the relocated code segment's start address) as part of the code loading process.

It is something I may put in my TODO list, but that's after the Xtensa LX6/LX7 inline assembler opcodes support and investigating if SIMD opcodes that are present in the ESP32S3 can be put to some use inside MicroPython.

jonnor, 2025-03-22:
@agatti thanks for the very quick answer. It sounds like armv6m support for soft-float would require a considerable amount of work in the mpy_ld linker, and that it is not really a "bug" per now that it does not work. One question: Do you think it would be possible to convince GCC to output relative relocations for armv6m - same as for the other architectures? Perhaps that could be a path to having this be supported?
The one GCC version I got from Arch Linux packages did not hit this (though I do not know why).

And considering that this is a known limitation, maybe we should put it a bit more user-friendly error message in mpy_ld.py for this case? Right now, it just hits a catch-all AssertionError.

agatti, 2025-03-22:
One thing you may want to try is to use `-fPIC` instead of `-fpic`, as it implies some changes in the memory model (and thus might impact how code is be accessed).  `-fPIC` is the non-optimised version of `-fpic` so it should be something worth trying first.

Another thing worth mentioning is that a few number of floating point stdlib functions are now treated as builtins by GCC, so the way they are accessed may have some implications on what is emitted.  

The first thing that came to mind was that an absolute relocation was emitted in order to reserve enough space to fit a FPU opcode in there if needed when performing the final linking pass, but unless I wade into GCC/binutils source code that's just a (wildly) uneducated guess.

Adding an error message is something that can easily be done, but anyone can do that, just check if the arch is `EM_ARM` and the relocation type is `R_ARM_ABS32`, then print an error message if it's the case :)

jonnor, 2025-03-28:
I tried using fPIC to build the module, and it did not help any. Still get caught on the same assertion due to absolute relocations.

jonnor, 2025-03-28:
Probably this is a wont-fix for now on ARMv6m, ref https://github.com/micropython/micropython/issues/14430#issuecomment-2744431299

clq9920, 2025-05-27:
Has this issue been resolved? I still cannot compile the exp function on MicroPython version 1.25.0, and I am completely unable to pass the test cases ([sin_cos_issues.zip](https://github.com/user-attachments/files/17040919/sin_cos_issues.zip)) mentioned in the previous discussion. My compilation environment is Arch Linux with the following packages:

arm-none-eabi-binutils-2.43-1

arm-none-eabi-gcc-14.2.0-1

arm-none-eabi-newlib-4.5.0.20241231-1

The error message across all ports is: "Absolute relocations not supported on ARM".

dpgeorge, 2025-05-28:
Absolute relocations can actually be supported by `mpy_ld.py`.  I've implemented that in #17377.  With that PR I can use `sin` and `cos` on `ARCH=armv6m` targets, eg RP2040.

dpgeorge, 2025-05-28:
> would it work for cases like [#14430 (comment)](https://github.com/micropython/micropython/issues/14430#issuecomment-2326857677)

Yes, that's exactly the case that now works: when there's a `ldr r1, [pc, #off]` and the data it loads at `pc+#off` has an `R_ARM_ABS32` relocation.  `mpy_ld.py` already supported `R_ARM_ABS32` in the ro-data section, just not in the text section.  The logic is the same though, so all that I needed to do in #17377 was use the existing ro-data relocation logic when in the text section.

The particular case I saw with sin/cos functions was using a constant table of floats (in the text section) whose address needed to be taken and passed to another function.

dpgeorge, 2025-06-10:
Fixed by 718ff4fdd58efd3e853599b489640877070d17be

closed 2025-06-10 by dpgeorge