micropython/micropython #14440: Unix port macOS user_c_module build failure

kdschlosser, 2024-05-07:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

Unix

### MicroPython version

MicroPython: 1.22.2

### Reproduction

Here is a link to the build output.

https://productionresultssa10.blob.core.windows.net/actions-results/78c25ee9-0ff8-473e-b773-3d1be26f09c8/workflow-job-run-ca395085-040a-526b-2ce8-bdc85f692774/logs/job/job-logs.txt?rsct=text%2Fplain&se=2024-05-07T06%3A38%3A12Z&sig=dyeUO9tJU6sXdCVQImosRvvsADf50kjsr5fsZnXCtdI%3D&sp=r&spr=https&sr=b&st=2024-05-07T06%3A28%3A07Z&sv=2021-12-02

I am compiling a user c module and for some reason the compiler arguments are not getting carried over from the main build. None of the include paths and none of the compiler arguments I have supplied in the user c module either. The files to compile are being passed without an issue.

Here is the compiler command that is being used when the user_c_module is being compiled.

```
clang -E  -Wno-unused-function -DNO_QSTR /Users/runner/work/lvgl_micropython/lvgl_micropython/ext_mod/lcd_bus/modlcd_bus.c  /Users/runner/work/lvgl_micropython/lvgl_micropython/ext_mod/lcd_bus/lcd_types.c  /Users/runner/work/lvgl_micropython/lvgl_micropython/ext_mod/lcd_bus/common_src/i2c_bus.c  /Users/runner/work/lvgl_micropython/lvgl_micropython/ext_mod/lcd_bus/common_src/i80_bus.c  /Users/runner/work/lvgl_micropython/lvgl_micropython/ext_mod/lcd_bus/common_src/spi_bus.c  /Users/runner/work/lvgl_micropython/lvgl_micropython/ext_mod/lcd_bus/common_src/rgb_bus.c  /Users/runner/work/lvgl_micropython/lvgl_micropython/ext_mod/lcd_bus/sdl_bus/sdl_bus.c  build-standard/lv_mpy.c  ../../py/mpstate.c  ../../py/malloc.c  ../../py/gc.c  ../../py/pystack.c  ../../py/qstr.c  ../../py/vstr.c  ../../py/mpprint.c  ../../py/unicode.c  ../../py/mpz.c  ../../py/reader.c  ../../py/lexer.c ../../py/parse.c ../../py/scope.c ../../py/compile.c ../../py/emitcommon.c ../../py/emitbc.c ../../py/asmbase.c ../../py/asmx64.c ../../py/emitnx64.c ../../py/asmx86.c ../../py/emitnx86.c ../../py/asmthumb.c ../../py/emitnthumb.c ../../py/emitinlinethumb.c ../../py/asmarm.c ../../py/emitnarm.c ../../py/asmxtensa.c ../../py/emitnxtensa.c ../../py/emitinlinextensa.c ../../py/emitnxtensawin.c ../../py/formatfloat.c ../../py/parsenumbase.c ../../py/parsenum.c ../../py/emitglue.c ../../py/persistentcode.c ../../py/runtime.c ../../py/runtime_utils.c ../../py/scheduler.c ../../py/nativeglue.c ../../py/pairheap.c ../../py/ringbuf.c ../../py/stackctrl.c ../../py/argcheck.c ../../py/warning.c ../../py/profile.c ../../py/map.c ../../py/obj.c ../../py/objarray.c ../../py/objattrtuple.c ../../py/objbool.c ../../py/objboundmeth.c ../../py/objcell.c ../../py/objclosure.c ../../py/objcomplex.c ../../py/objdeque.c ../../py/objdict.c ../../py/objenumerate.c 
```

I am compiling 2 user c modules and what I find strange about this command is it has one source file from another user c module. That other c module compiles some 200 source files but only the one has been added to the build command. I am not sure why this behavior exists. Maybe because that file is auto generated and gets placed into the build folder? IDK.

There is also a mixture of  MicroPython source files in it as well but once again it is not all of the MicroPython source files.

Here is the user c module makefile where this problem seems to be connected to.

```makefile
MOD_DIR := $(USERMOD_DIR)
LVGL_BINDING_DIR = $(subst /ext_mod/lcd_bus,,$(MOD_DIR))

CFLAGS_USERMOD += -I$(MOD_DIR)
CFLAGS_USERMOD += -I$(MOD_DIR)/common_include
CFLAGS_USERMOD += -I$(MOD_DIR)/sdl_bus
CFLAGS_USERMOD += -Wno-missing-field-initializers
CFLAGS_USERMOD += -Wno-unused-function

SRC_USERMOD_C += $(MOD_DIR)/modlcd_bus.c
SRC_USERMOD_C += $(MOD_DIR)/lcd_types.c
SRC_USERMOD_C += $(MOD_DIR)/common_src/i2c_bus.c
SRC_USERMOD_C += $(MOD_DIR)/common_src/i80_bus.c
SRC_USERMOD_C += $(MOD_DIR)/common_src/spi_bus.c
SRC_USERMOD_C += $(MOD_DIR)/common_src/rgb_bus.c
SRC_USERMOD_C += $(MOD_DIR)/sdl_bus/sdl_bus.c

ifneq (,$(findstring unix, $(LV_PORT)))
    CFLAGS_USERMOD += -DMP_PORT_UNIX=1
    CFLAGS_USERMOD += -I$(BUILD)/SDL/include/SDL2
    CFLAGS_USERMOD += -I$(BUILD)/SDL/include-config-release/SDL2
    LDFLAGS_USERMOD += -L$(BUILD)/SDL
    LDFLAGS_USERMOD += -lSDL2
endif
```

when this compiles on Linux (Ubuntu) it compiles without any issues. It only has this issue on macOS (darwin) using clang. Because of the complexity of the MicroPython build system I am not able to isolate where the problem is.

Here is the Unix build. You need to expand the section that says "Build Unix port"
https://github.com/kdschlosser/lvgl_micropython/actions/runs/8980689074/job/24664774356

The same make command is used to compile both ports but the failure only occurs when compiling on macOS using clang

### Expected behaviour

I expect it to complete the build without error.

### Observed behaviour

```shell
In file included from /Users/runner/work/lvgl_micropython/lvgl_micropython/ext_mod/lcd_bus/modlcd_bus.c:2:
In file included from /Users/runner/work/lvgl_micropython/lvgl_micropython/ext_mod/lcd_bus/modlcd_bus.h:4:
Error: /Users/runner/work/lvgl_micropython/lvgl_micropython/ext_mod/lcd_bus/lcd_types.h:7:14: fatal error: 'py/obj.h' file not found
#include "py/obj.h"
^~~~~~~~~~
1 error generated.
Error: ../../extmod/machine_uart.c:27:10: fatal error: 'py/runtime.h' file not found
...
```
There is a bunch of errors and every error is caused by the includes not being passed to the compiler.

### Additional Information

No, I've provided everything above.

kdschlosser, 2024-05-07:
Unix
```shell
gmake -j 3 -C /Users/runner/work/lvgl_micropython/lvgl_micropython/lib/micropython/ports/unix LV_CFLAGS="" LV_PORT=unix USER_C_MODULES="/Users/runner/work/lvgl_micropython/lvgl_micropython/ext_mod" FROZEN_MANIFEST="/Users/runner/work/lvgl_micropython/lvgl_micropython/build/manifest.py" GEN_SCRIPT=python
```

macOS
```shell
make -j 4 -C /home/runner/work/lvgl_micropython/lvgl_micropython/lib/micropython/ports/unix LV_CFLAGS="" LV_PORT=unix USER_C_MODULES="/home/runner/work/lvgl_micropython/lvgl_micropython/ext_mod" FROZEN_MANIFEST="/home/runner/work/lvgl_micropython/lvgl_micropython/build/manifest.py" GEN_SCRIPT=python
```

`GEN_SCRIPT` and `LV_PORT` are used in the makefile for the user c module. They are flags to tell it what code generator to run and to tell it if SDL needs to be linked to.

kdschlosser, 2024-05-07:
The difference in using gmake vs make is because on github actions no matter what version of macOS you use the installed make is version 3.8 and it was compiled for use with darwin 11.3. So I thought that the hiccup might be cause by the macOS image being darwin 23.x. I used Homebrew to install make into the image and because there is already a make installed it installs it as gmake. But none the less it didn't fix the problem anywho.

Here is the version output of the macOS image on github actions and the version if the pre installed make and also the gmake that I installed.

* Darwin version: 
  Darwin Mac-1715070049993.local 23.4.0 Darwin Kernel Version 23.4.0: Fri Mar 15 00:10:50 PDT 2024; root:xnu-10063.101.17~1/RELEASE_ARM64_VMAPPLE arm64

* make version: 
  GNU Make 3.81
  
  Copyright (C) 2006  Free Software Foundation, Inc.
  This is free software; see the source for copying conditions.
  There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

  This program built for i386-apple-darwin11.3.0

* gmake version: 
  GNU Make 4.4.1
  Built for aarch64-apple-darwin23.0.0

  Copyright (C) 1988-2023 Free Software Foundation, Inc.
  License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>
  This is free software: you are free to change and redistribute it.
  There is NO WARRANTY, to the extent permitted by law.

kdschlosser, 2024-05-07:
There are 2 makefiles that get run for the user_c_modules. the path that is pointed to is the parent folder of the user c modules which are in their own folders. The make files are located in each of those folders.

Here is the other makefile that gets used
```make
MOD_DIR := $(USERMOD_DIR)
LVGL_BINDING_DIR = $(subst /ext_mod/lvgl,,$(MOD_DIR))

LVGL_DIR = $(LVGL_BINDING_DIR)/lib/lvgl
LVGL_ADDON_DIR = $(LVGL_BINDING_DIR)/ext_mod/lvgl_addons
LVGL_HEADER = $(LVGL_BINDING_DIR)/build/lvgl_header.h

CFLAGS_USERMOD += -I$(LVGL_BINDING_DIR)
CFLAGS_USERMOD += -I$(LVGL_DIR)
CFLAGS_USERMOD += -I$(LVGL_BINDING_DIR)/include
CFLAGS_USERMOD += -Wno-missing-field-initializers
CFLAGS_USERMOD += -Wno-unused-function

ALL_LVGL_SRC = $(shell find $(LVGL_DIR) -type f -name '*.h') $(LVGL_BINDING_DIR)/lib/lv_conf.h

LVGL_MPY = $(BUILD)/lv_mpy.c
LVGL_MPY_METADATA = $(BUILD)/lv_mpy.json

CFLAGS_USERMOD += $(LV_CFLAGS)

$(LVGL_MPY): $(ALL_LVGL_SRC) $(LVGL_BINDING_DIR)/gen/$(GEN_SCRIPT)_api_gen_mpy.py
	$(ECHO) "LVGL-GEN $@"
	$(Q)mkdir -p $(dir $@)
	$(Q)$(PYTHON) $(LVGL_BINDING_DIR)/gen/$(GEN_SCRIPT)_api_gen_mpy.py $(LV_CFLAGS) --board=$(LV_PORT) --output=$(LVGL_MPY) --include=$(LVGL_BINDING_DIR) --include=$(LVGL_DIR) --include=$(LVGL_BINDING_DIR)/include --module_name=lvgl --module_prefix=lv --metadata=$(LVGL_MPY_METADATA) --header_file=$(LVGL_HEADER)

.PHONY: LVGL_MPY
LVGL_MPY: $(LVGL_MPY)

SRC_USERMOD_LIB_C += $(shell find $(LVGL_DIR)/src -type f -name "*.c")
SRC_USERMOD_LIB_C += $(LVGL_ADDON_DIR)/src/color_addons.c
SRC_USERMOD_C += $(LVGL_MPY)

ifneq (,$(findstring stm32, $(LV_PORT)))
    CFLAGS_USERMOD += -DMP_SOFT_MATH=1
    SRC_USERMOD_LIB_C += $(LVGL_ADDON_DIR)/src/soft_math.c
endif

ifneq (,$(findstring rp2, $(LV_PORT)))
    CFLAGS_USERMOD += -DMP_SOFT_MATH=1
    SRC_USERMOD_LIB_C += $(LVGL_ADDON_DIR)/src/soft_math.c
endif
```

I did want to note the exact error as well.
```
2024-05-07T08:10:17.4354090Z gmake: *** [../../py/mkrules.mk:122: build-standard/genhdr/qstr.i.last] Error 1
```

The error location indicated above is this code. Specifically the last line in the code block below. 
```make
$(HEADER_BUILD)/qstr.i.last: $(SRC_QSTR) $(QSTR_GLOBAL_DEPENDENCIES) | $(QSTR_GLOBAL_REQUIREMENTS)
	$(ECHO) "GEN $@"
	$(Q)$(PYTHON) $(PY_SRC)/makeqstrdefs.py pp $(CPP) output $(HEADER_BUILD)/qstr.i.last cflags $(QSTR_GEN_CFLAGS) cxxflags $(QSTR_GEN_CXXFLAGS) sources $^ dependencies $(QSTR_GLOBAL_DEPENDENCIES) changed_sources $?
```

kdschlosser, 2024-05-07:
and in the unix port makefile there is a force to use clang as the compiler when the OS is darwin.

```
# On OSX, 'gcc' is a symlink to clang unless a real gcc is installed.
# The unix port of MicroPython on OSX must be compiled with clang,
# while cross-compile ports require gcc, so we test here for OSX and
# if necessary override the value of 'CC' set in py/mkenv.mk
ifeq ($(UNAME_S),Darwin)
    ifeq ($(MICROPY_FORCE_32BIT),1)
        CC = clang -m32
    else
        CC = clang
    endif
    # Use clang syntax for map file
    LDFLAGS_ARCH = -Wl,-map,$@.map -Wl,-dead_strip
else
    # Use gcc syntax for map file
    LDFLAGS_ARCH = -Wl,-Map=$@.map,--cref -Wl,--gc-sections
endif
```

kdschlosser, 2024-05-08:
I think I found the problem. I am not sure why it is happening with compiling under macOS but I am not able to set CFLAGS by passing it to make at the command line. It boogers it up for some reason. I have to use CFLAGS_EXTRA instead.

I am working on it at the moment to see if that is the case.

kdschlosser, 2024-05-08:
Thanks for offering to help I appreciate it. Now I just have to figure out why it is getting hunk up when linking.

closed 2024-05-08 by kdschlosser

stinos, 2024-05-08:
> yup that appears to have fixed the issue. crazy why it only causes a problem on darwin.

That's probaby not as intended