micropython/micropython #14447: Using dynamically linked mbedtls

orangepizza, 2024-05-08:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) regarding this feature, and didn't find any.

### Description

after updated to https://github.com/openwrt/packages/blob/master/lang/python/micropython/Makefile it starts to fail to compile because of for some reason mbedtls_config_check running on v2 config file, and when I checked patch for it 
https://github.com/openwrt/packages/blob/master/lang/python/micropython/patches/040-extmod-use-external-mbedtls.patch  It never attempted to get upstream support. so I'd ask how to fix or have defined way to micropython to use dynamically linked system mbedtls

### Implementation

- [ ] I intend to implement this feature and would submit a Pull Request if desirable.
- [ ] I hope the MicroPython maintainers or community will implement this feature.
- [ ] I would like to [Sponsor](https://github.com/sponsors/micropython#sponsors) development of this feature.

jonnor, 2024-09-01:
What is the compilation failure that you are seeing? Which version of MicroPython has the issue? And which prior versions did not have it?

orangepizza, 2024-09-01:
It was because we tried to use mpy1.21 when itbstill used mbedtls 2.28: I just forgot about this:
Btw on newest mbedtls tls 1.3 doesn't allow disable certificate verification: make default ssl context useless, was there a way to load full certificate bundle as default context in unix port? Mbedtls itself should allow but from doc mpy only allows single certificate in context