micropython/micropython #14449: Cannot build example usercmodule in nrf port

microbit-carlos, 2024-05-08:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

nrf port, tested with default PCA10040 board and MICROBIT board

### MicroPython version

8762fe8b4c6a7608d48daa8458e6501fafd3eafc

### Reproduction

```bash
git clone https://github.com/micropython/micropython.git
cd micropython/mpy-cross/
make
cd ../ports/nrf
make submodules
make USER_C_MODULES=../../examples/usercmodule
```

### Expected behaviour

No build errors

### Observed behaviour

Build error:
```
In file included from ../../examples/usercmodule/cppexample/example.cpp:2:
../../examples/usercmodule/cppexample/examplemodule.h:2:10: fatal error: py/runtime.h: No such file or directory
    2 | #include "py/runtime.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
Command '['arm-none-eabi-gcc', '-E', '-I../../examples/usercmodule/cppexample', '-std=c++11', '-DNO_QSTR', '../../examples/usercmodule/cppexample/example.cpp']' returned non-zero exit status 1.
make: *** [build-PCA10040/genhdr/qstr.i.last] Error 1
make: *** Deleting file `build-PCA10040/genhdr/qstr.i.last'
```

<details>
<summary>Full build output</summary>

<pre>

❯ git clone https://github.com/micropython/micropython.git
Cloning into 'micropython'...
remote: Enumerating objects: 120730, done.
remote: Counting objects: 100% (64/64), done.
remote: Compressing objects: 100% (33/33), done.
remote: Total 120730 (delta 31), reused 51 (delta 30), pack-reused 120666
Receiving objects: 100% (120730/120730), 61.44 MiB | 30.37 MiB/s, done.
Resolving deltas: 100% (90584/90584), done.

mbef/micropython/rm took 5s
❯ cd micropython/mpy-cross/

micropython/mpy-cross on  master
❯ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
mkdir -p build/genhdr
GEN build/genhdr/mpversion.h
GEN build/genhdr/qstr.i.last
GEN build/genhdr/qstr.split
GEN build/genhdr/qstrdefs.collected.h
QSTR updated
GEN build/genhdr/qstrdefs.generated.h
GEN build/genhdr/moduledefs.split
GEN build/genhdr/moduledefs.collected
Module registrations updated
GEN build/genhdr/moduledefs.h
GEN build/genhdr/root_pointers.split
GEN build/genhdr/root_pointers.collected
Root pointer registrations updated
GEN build/genhdr/root_pointers.h
mkdir -p build/py
mkdir -p build/shared/runtime
CC ../py/mpstate.c
CC ../py/nlr.c
CC ../py/nlrx86.c
CC ../py/nlrx64.c
CC ../py/nlrthumb.c
CC ../py/nlraarch64.c
CC ../py/nlrmips.c
CC ../py/nlrpowerpc.c
CC ../py/nlrxtensa.c
CC ../py/nlrsetjmp.c
CC ../py/malloc.c
CC ../py/gc.c
CC ../py/pystack.c
CC ../py/qstr.c
CC ../py/vstr.c
CC ../py/mpprint.c
CC ../py/unicode.c
CC ../py/mpz.c
CC ../py/reader.c
CC ../py/lexer.c
CC ../py/parse.c
CC ../py/scope.c
CC ../py/compile.c
CC ../py/emitcommon.c
CC ../py/emitbc.c
CC ../py/asmbase.c
CC ../py/asmx64.c
CC ../py/emitnx64.c
CC ../py/asmx86.c
CC ../py/emitnx86.c
CC ../py/asmthumb.c
CC ../py/emitnthumb.c
CC ../py/emitinlinethumb.c
CC ../py/asmarm.c
CC ../py/emitnarm.c
CC ../py/asmxtensa.c
CC ../py/emitnxtensa.c
CC ../py/emitinlinextensa.c
CC ../py/emitnxtensawin.c
CC ../py/formatfloat.c
CC ../py/parsenumbase.c
CC ../py/parsenum.c
CC ../py/emitglue.c
CC ../py/persistentcode.c
CC ../py/runtime.c
CC ../py/runtime_utils.c
CC ../py/scheduler.c
CC ../py/nativeglue.c
CC ../py/pairheap.c
CC ../py/ringbuf.c
CC ../py/stackctrl.c
CC ../py/argcheck.c
CC ../py/warning.c
CC ../py/profile.c
CC ../py/map.c
CC ../py/obj.c
CC ../py/objarray.c
CC ../py/objattrtuple.c
CC ../py/objbool.c
CC ../py/objboundmeth.c
CC ../py/objcell.c
CC ../py/objclosure.c
CC ../py/objcomplex.c
CC ../py/objdeque.c
CC ../py/objdict.c
CC ../py/objenumerate.c
CC ../py/objexcept.c
CC ../py/objfilter.c
CC ../py/objfloat.c
CC ../py/objfun.c
CC ../py/objgenerator.c
CC ../py/objgetitemiter.c
CC ../py/objint.c
CC ../py/objint_longlong.c
CC ../py/objint_mpz.c
CC ../py/objlist.c
CC ../py/objmap.c
CC ../py/objmodule.c
CC ../py/objobject.c
CC ../py/objpolyiter.c
CC ../py/objproperty.c
CC ../py/objnone.c
CC ../py/objnamedtuple.c
CC ../py/objrange.c
CC ../py/objreversed.c
CC ../py/objset.c
CC ../py/objsingleton.c
CC ../py/objslice.c
CC ../py/objstr.c
CC ../py/objstrunicode.c
CC ../py/objstringio.c
CC ../py/objtuple.c
CC ../py/objtype.c
CC ../py/objzip.c
CC ../py/opmethods.c
CC ../py/sequence.c
CC ../py/stream.c
CC ../py/binary.c
CC ../py/builtinimport.c
CC ../py/builtinevex.c
CC ../py/builtinhelp.c
CC ../py/modarray.c
CC ../py/modbuiltins.c
CC ../py/modcollections.c
CC ../py/modgc.c
CC ../py/modio.c
CC ../py/modmath.c
CC ../py/modcmath.c
CC ../py/modmicropython.c
CC ../py/modstruct.c
CC ../py/modsys.c
CC ../py/moderrno.c
CC ../py/modthread.c
CC ../py/vm.c
CC ../py/bc.c
CC ../py/showbc.c
CC ../py/repl.c
CC ../py/smallint.c
CC ../py/frozenmod.c
CC main.c
CC gccollect.c
CC ../shared/runtime/gchelper_generic.c
LINK build/mpy-cross
__TEXT	__DATA	__OBJC	others	dec	hex
327680	16384	0	4295000928	4295344992	10005c360

micropython/mpy-cross on  master took 28s
❯ cd ../ports/nrf

micropython/ports/nrf on  master
❯ make submodules
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Updating submodules: lib/nrfx lib/tinyusb lib/micropython-lib
Submodule 'lib/micropython-lib' (https://github.com/micropython/micropython-lib.git) registered for path '../../lib/micropython-lib'
Submodule 'lib/nrfx' (https://github.com/NordicSemiconductor/nrfx.git) registered for path '../../lib/nrfx'
Submodule 'lib/tinyusb' (https://github.com/hathach/tinyusb) registered for path '../../lib/tinyusb'
Cloning into '/Users/microbit-carlos/workspace/mbef/micropython/rm/micropython/lib/micropython-lib'...
Cloning into '/Users/microbit-carlos/workspace/mbef/micropython/rm/micropython/lib/nrfx'...
Cloning into '/Users/microbit-carlos/workspace/mbef/micropython/rm/micropython/lib/tinyusb'...
Submodule path '../../lib/micropython-lib': checked out 'ddb1a279578bfff8c1b18aff3baa668620684f64'
Submodule path '../../lib/nrfx': checked out '7a4c9d946cf1801771fc180acdbf7b878f270093'
Submodule path '../../lib/tinyusb': checked out '1fdf29075d4e613eacfa881166015263797db0f6'

micropython/ports/nrf on  master took 8s
❯ make USER_C_MODULES=../../examples/usercmodule
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Including User C Module from ../../examples/usercmodule/cexample
Including User C Module from ../../examples/usercmodule/cppexample
Including User C Module from ../../examples/usercmodule/subpackage
mkdir -p build-PCA10040/genhdr
Create build-PCA10040/genhdr/pins.h
GEN build-PCA10040/genhdr/mpversion.h
GEN build-PCA10040/genhdr/qstr.i.last
In file included from ../../examples/usercmodule/cppexample/example.cpp:2:
../../examples/usercmodule/cppexample/examplemodule.h:2:10: fatal error: py/runtime.h: No such file or directory
    2 | #include "py/runtime.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
Command '['arm-none-eabi-gcc', '-E', '-I../../examples/usercmodule/cppexample', '-std=c++11', '-DNO_QSTR', '../../examples/usercmodule/cppexample/example.cpp']' returned non-zero exit status 1.
make: *** [build-PCA10040/genhdr/qstr.i.last] Error 1
make: *** Deleting file `build-PCA10040/genhdr/qstr.i.last'
</pre>

</details>

### Additional Information

No, I've provided everything above.

microbit-carlos, 2024-05-08:
I guess the nordic port is missing setting the flags like these:

https://github.com/micropython/micropython/blob/8762fe8b4c6a7608d48daa8458e6501fafd3eafc/ports/unix/Makefile#L236
https://github.com/micropython/micropython/blob/8762fe8b4c6a7608d48daa8458e6501fafd3eafc/ports/stm32/Makefile#L164
https://github.com/micropython/micropython/blob/8762fe8b4c6a7608d48daa8458e6501fafd3eafc/ports/samd/Makefile#L90

andrewleech, 2024-05-08:
Good find, yes the nordic port hasn't had as much use as some of the other ports and there are a number of little things like this missing I find. 
You could start with just `CXXFLAGS += $(filter-out -std=c99,$(CFLAGS)) ` in the Makefile and see if that's enough. If the c++ compiler complains about other things it doesn't support simply filter them out too.