micropython/micropython #14450: Make MPY ARCH supports for ESP32C3 RISCV

m24h, 2024-05-08:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) regarding this feature, and didn't find any.

### Description

I tried to use MPY to do real-time tasks on the ESP32C3, but it doesn't support hard interrupts, I can't get the timestamp and Pin state of the IRQ event (why don't MPY give such important informations rather then a useless Pin object), and if I use timer for resampling, the time jitter of the timer event is too bad (also a soft interrupt), so I had to try to use native code mpy, but the RISCV architecture of ESP32C3 is not supported, and there is no way to do it without customizing the firmware. How can I get millimeter-level time processing accuracy?

### Code Size

I'm not sure about the code size requirement for this feature, maybe it's less than several kilo-bytes.

### Implementation

- [ ] I intend to implement this feature and would submit a Pull Request if desirable.
- [X] I hope the MicroPython maintainers or community will implement this feature.
- [ ] I would like to [Sponsor](https://github.com/sponsors/micropython#sponsors) development of this feature.

rkompass, 2024-05-14:
Did you try to use the pin_object.irq() method? It might provide additional information like
pin_object.irq().flags() and pin_object.irq().trigger() in case it is implemented.
In the w600 port there is now also an experimental pin_object.irq().timestamp_us() now, btw..

robert-hh, 2024-05-14:
The ESP port does not support pin_object.irq().flags() or the keyword option `hard`.

agatti, 2024-09-24:
@m24h I'm a bit late to the party, but does https://github.com/micropython/micropython/pull/15603 work for you?