micropython/micropython #14458: Pico-W OSError: [Errno 22] EINVAL bluetooth module

redhead-p, 2024-05-09:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

RP2040 / Pico-W

### MicroPython version

MicroPython v1.23.0-preview.116.g0432f7320 on 2024-02-12; Raspberry Pi Pico W with RP2040

### Reproduction

This fault occurs when setting up a Bluetooth LE peripheral on a Pico-W.  Three services are defined and added to the server configuration using BLE.gatts_register_services(). I've attached a code module to demonstrate the problem

[BLE-OSError22.zip](https://github.com/micropython/micropython/files/15264618/BLE-OSError22.zip)

### Expected behaviour

The expected return from `BLE.gatts_register_services()`  should be similar to that returned on an Arduino RP2040 Nano Connect using MicroPython v1.22.1 on 2024-01-05. E.g

`((21, 24, 26), (29, 32, 34, 37), (41, 44, 46, 49))`

Writes to values using these handles work as expected.

### Observed behaviour

 The returned set of handles from `BLE.gatts_register_services()` contains duplicates e.g. 

`((9, 12, 14), (17, 20, 22, 25), (27, 28, 28, 28))`

Writing a value to handle 17 works OK.  Writing a value to handle 27 causes Error 22.

### Additional Information

The ESP32 port appears OK too.

redhead-p, 2024-05-11:
Some additional information.

1) MicroPython v1.22.2 on 2024-02-22; Raspberry Pi Pico W with RP2040 exhibits the same fault.

2) As well as not allowing server writes, the associated attributes are not discoverable at the BLE central/client end.

Timeinz, 2025-05-25:
Have the same / similar issue on RPI_PICO_W-20241129-v1.24.1

Can't gatts write descriptors into handle 12