micropython/micropython #14475: expanding the ESP32 NVS

kdschlosser, 2024-05-13:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) regarding this feature, and didn't find any.

### Description

Pseudo code of the expansion.

```c

#include <string.h>

#include "py/runtime.h"
#include "py/mperrno.h"
#include "mphalport.h"
#include "modesp32.h"
#include "nvs_flash.h"
#include "nvs.h"

#define NVS_TYPE_FLOAT  (0xFE)

// This file implements the NVS (Non-Volatile Storage) class in the esp32 module.
// It provides simple access to the NVS feature provided by ESP-IDF.

// NVS python object that represents an NVS namespace.
typedef struct _esp32_nvs_obj_t {
    mp_obj_base_t base;
    nvs_handle_t namespace;
} esp32_nvs_obj_t;

// *esp32_nvs_new allocates a python NVS object given a handle to an esp-idf namespace C obj.
STATIC esp32_nvs_obj_t *esp32_nvs_new(nvs_handle_t namespace) {
    esp32_nvs_obj_t *self = mp_obj_malloc(esp32_nvs_obj_t, &esp32_nvs_type);
    self->namespace = namespace;
    return self;
}

// esp32_nvs_print prints an NVS object, unfortunately it doesn't seem possible to extract the
// namespace string or anything else from the opaque handle provided by esp-idf.
STATIC void esp32_nvs_print(const mp_print_t *print, mp_obj_t self_in, mp_print_kind_t kind) {
    // esp32_nvs_obj_t *self = MP_OBJ_TO_PTR(self_in);
    mp_printf(print, "<NVS namespace>");
}

// esp32_nvs_make_new constructs a handle to an NVS namespace.
STATIC mp_obj_t esp32_nvs_make_new(const mp_obj_type_t *type, size_t n_args, size_t n_kw, const mp_obj_t *all_args) {
    // Check args
    mp_arg_check_num(n_args, n_kw, 1, 1, false);

    // Get requested nvs namespace
    const char *ns_name = mp_obj_str_get_str(all_args[0]);
    nvs_handle_t namespace;
    check_esp_err(nvs_open(ns_name, NVS_READWRITE, &namespace));
    return MP_OBJ_FROM_PTR(esp32_nvs_new(namespace));
}

// esp32_nvs_set_i32 sets a 32-bit integer value
STATIC mp_obj_t esp32_nvs_set(size_t n_args, const mp_obj_t *pos_args, mp_map_t *kw_args) {
    enum { ARG_self, ARG_key, ARG_value, ARG_type };
    static const mp_arg_t allowed_args[] = {
        { MP_QSTR_self,   MP_ARG_OBJ  | MP_ARG_REQUIRED, { .u_obj = mp_const_none } },
        { MP_QSTR_key,    MP_ARG_OBJ  | MP_ARG_REQUIRED, { .u_obj = mp_const_none } },
        { MP_QSTR_value,  MP_ARG_OBJ  | MP_ARG_REQUIRED, { .u_obj = mp_const_none } },
        { MP_QSTR_type,   MP_ARG_INT  | MP_ARG_REQUIRED, { .u_int = 1 } },
    };

    mp_arg_val_t args[MP_ARRAY_SIZE(allowed_args)];
    mp_arg_parse_all(n_args, pos_args, kw_args, MP_ARRAY_SIZE(allowed_args), allowed_args, args);
[... 202 lines cut ...]
STATIC MP_DEFINE_CONST_DICT(esp32_nvs_locals_dict, esp32_nvs_locals_dict_table);

MP_DEFINE_CONST_OBJ_TYPE(
    esp32_nvs_type,
    MP_QSTR_NVS,
    MP_TYPE_FLAG_NONE,
    make_new, esp32_nvs_make_new,
    print, esp32_nvs_print,
    locals_dict, &esp32_nvs_locals_dict
);
```

### Code Size

I think the code size increase would not be very large but it would be worth doing to provide better use of the NVS and at the same time having the possibility of a reduction in memory use by having smaller python source files.

It would be nice to use all of the data types that are supported by the NVS and while this won't directly reduce the size it has the potential to reduce the ram use. Python code takes up memory when the files are loaded. The smaller the source file the lower amount of memory is used. currently there are only 2 data types that are supported. int32_t and blob. storing a uint8_t is going to take up 4 bytes of NVS space instead of 1. sure I can use bit shifting at the expense of increase memory use do to a larger python code footprint to do that. The other thing is storing negative and positive numbers together, it just won't work because there is no way to know what is negative and what is positive when separating them. 

I also added support for storing floats as well. This is easily done using type punning.

### Implementation

- [ ] I intend to implement this feature and would submit a Pull Request if desirable.
- [X] I hope the MicroPython maintainers or community will implement this feature.
- [ ] I would like to [Sponsor](https://github.com/sponsors/micropython#sponsors) development of this feature.

glenn20, 2024-07-03:
An alternative approach to use the NVS storage efficiently for complex data is to store packed struct data as an NVS blob, eg:

```py
buf = bytearray(36)
blob = esp32.NVS("data").get_blob("config")
b1, b2, i1, name, i2, i3, u1, u2 = struct.unpack("<bbL20sLLBB", blob)
```

NB. I make no comment on the value of implementing the requested implementation, just offering advice on an alternative approach to solving the identified problem.

glenn20, 2024-07-03:
Packing data into blobs also uses the NVS storage much more efficiently (every key_value pair uses at least 32 bytes regardless of  the stored int size).

kdschlosser, 2024-07-03:
Well how about that. You are correct about the 32 bytes but that 32 bytes is used for more than just the key and value. 16 bytes gets used for the key and 8 bytes are what is used for the value.

A blob gets broken into pieces and still get stored using the same mechanics. It ends up getting split across one or more items which means that the most data that is able to be stored is going to be 8 bytes. So if you pass a blob of 32 bytes the amount of NVS space used is going to be `32 / 8 * 32 = 128`. That's a whole lot better than storing 32 separate 8 bit integers which would take up `32 * 32 = 1024` bytes.

I am going to work up something to handle storing the data using blobs. The one thing I need to be able to do is to store floats and I don't know if there is a way to cast a `float` to a `uint32_t` in MicroPython from the Python side. I know there is in CPython using ctypes but it is only good to a float precision of 7 decimal places. I may have to write a C helper function to handle this.

glenn20, 2024-07-03:
Actually, I believe it is 64 bytes for a 32 byte blob (ie. 2 entries). First entry includes all the metadata. The blob itself is stored across the next n entries required to hold it (32 bytes per entry).

kdschlosser, 2024-07-03:
If you look at the code that stores the blob it calls `Page::writeItem` which is what writes the data. That is what creates the `Item` union which is what gets written. 

This is the `Item` union

```
union {
        struct {
            uint8_t  nsIndex;
            ItemType datatype;
            uint8_t  span;
            uint8_t  chunkIndex;
            uint32_t crc32;
            char     key[NVS_KEY_NAME_MAX_SIZE];
            union {
                struct {
                    uint16_t dataSize;
                    uint16_t reserved;
                    uint32_t dataCrc32;
                } varLength;
                struct {
                    uint32_t   dataSize;
                    uint8_t    chunkCount; // Number of children data blobs.
                    VerOffset  chunkStart; // Offset from which the chunkIndex for children blobs starts
                    uint16_t   reserved;
                } blobIndex;
                uint8_t data[8];
            };
        };
        uint8_t rawData[32];
    };
```

since this is a variable length thing when looking at the code located here

https://github.com/espressif/esp-idf/blob/a6e89b466fc42a1b3a68b091ddea82ae32f7ebc4/components/nvs_flash/src/nvs_page.cpp#L234

You can see here

https://github.com/espressif/esp-idf/blob/a6e89b466fc42a1b3a68b091ddea82ae32f7ebc4/components/nvs_flash/src/nvs_page.cpp#L243

That `rawData[32]` is not being used. So that makes the actual data that is getting stored to have a length of 8 bytes at a time.

glenn20, 2024-07-03:
> That `rawData[32]` is not being used. So that makes the actual data that is getting stored to have a length of 8 bytes at a time.

Thanks for pointing me to the code, I was going off the docs.

Actually, at line 255, `writeEntryData` is used to copy the contents of data into the next N entries on the flash. Then, on line 265, any residual data is copied into the next entry.

Effectively the blob is written across the next N entries after the entry record for the item.