micropython/micropython #14488: ESP32 NVS OverflowError

kdschlosser, 2024-05-15:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

ESP32

### MicroPython version

1.22.2

### Reproduction

```python
import esp32

config = esp32.NVS('test')
config.set_i32('testkey', -2147483648)
config.commit()
```

### Expected behaviour

for the value to get set properly. The minimum allowed value for a 32 bit integer is `-2147483648` and I should be able to set that value without an issue.

### Observed behaviour

```
OverflowError: overflow converting long int to machine word
```

### Additional Information

No, I've provided everything above.

agatti, 2024-09-08:
@kdschlosser I've bumped into this the other day as well.  The workaround I've used was this:

```diff
diff --git a/ports/esp32/esp32_nvs.c b/ports/esp32/esp32_nvs.c
index daeec4c26..1781c2595 100644
--- a/ports/esp32/esp32_nvs.c
+++ b/ports/esp32/esp32_nvs.c
@@ -72,7 +72,7 @@ static mp_obj_t esp32_nvs_make_new(const mp_obj_type_t *type, size_t n_args, siz
 static mp_obj_t esp32_nvs_set_i32(mp_obj_t self_in, mp_obj_t key_in, mp_obj_t value_in) {
     esp32_nvs_obj_t *self = MP_OBJ_TO_PTR(self_in);
     const char *key = mp_obj_str_get_str(key_in);
-    int32_t value = mp_obj_get_int(value_in);
+    int32_t value = mp_obj_get_int_truncated(value_in);
     check_esp_err(nvs_set_i32(self->namespace, key, value));
     return mp_const_none;
 }
```

However if you pass values that don't fit in 32 bits it won't detect the overflow and will still write the lower 32 bits of the value to the NVS key.  It worked for my case, but this fix may not be good enough to submit as a PR :)

kdschlosser, 2024-09-08:
it's strange that the fix you have done is what fixed it because `mp_obj_get_int_truncated` should be returning an unsigned int. an int has a range from -2,147,483,648 to 2,147,483,647 where as an unsigned 32 bit integer  has a range of 0 to 4,294,967,295. Both are 4 bytes. 

If you look at the command... `config.set_i32('testkey', -2147483648)` and then look at the original code `int32_t value = mp_obj_get_int(value_in);` this would infer that I should be able to pass a negative value. even the function being used from the IDF `nvs_set_i32` is the correct function being used for passing a 4 byte signed integer. There is `nvs_set_u32` for passing a 4 byte unsigned integer.  There is definitely something wrong somewhere. 

what I have not tried to do is see if I am able to set the value to `-1` or to `-2,147,483,647`.. Maybe you want to test that to see if it will work without an overflow. if it will not take a negative value at all then there is something that is broken somewhere. I went a different route because I wanted to store floats and I also wanted to use as little space in the NVS as possible so I decided to use the `struct` library to pack multiple values into a byte array and I am then setting the data as a blob instead. storing in the NVS has a bunch of added data so using a bunch of different keys eats up the space really fast. While I know that I am able to increase the size of the NVS by altering the partition sizes I didn't want to do that seeing as how I am not storing all that much data.

agatti, 2024-09-08:
> it's strange that the fix you have done is what fixed it because `mp_obj_get_int_truncated` should be returning an unsigned int. an int has a range from -2,147,483,648 to 2,147,483,647 where as an unsigned 32 bit integer has a range of 0 to 4,294,967,295. Both are 4 bytes.

Well, whether a number is signed or unsigned is just a matter of how you want to look at the bits. -2147483648 as a signed integer has the same bit representation as 2147483648 as an unsigned integer after all.

> There is `nvs_set_u32` for passing a 4 byte unsigned integer. There is definitely something wrong somewhere.

Python (and by extension MicroPython) integers are signed, so it makes sense for `nvs_set_i32` being used by default.

> what I have not tried to do is see if I am able to set the value to `-1` or to `-2,147,483,647`.. Maybe you want to test that to see if it will work without an overflow.

It did work for me indeed.

kdschlosser, 2024-09-08:
> It did work for me indeed.

If it worked without making any modifications then there is a range issue with `mp_obj_get_int`. it is not handling -2147483648 correctly.
 
> -2147483648 as a signed integer has the same bit representation as 2147483648 as an unsigned integer after all.
2147483648 as a unsigned integer is out of the range for an signed integer.

where you run into the issue is -1 is stored as `11111111111111111111111111111111` for a signed integer where as `11111111111111111111111111111111` for an unsigned integer ends up being 4294967295. If you attempt to push `4294967295` into a signed int it is not going to fit.

I don't know how MicroPython is handling the conversions from one to the other but it should be done like so...

```c
union { uint32_t u; int32_t i; } __c;
__u.i = -1;

// the value of __c.u would be 4294967295
```

this is called type punning and is how the conversion should be handled.

the `mp_obj_get_int_truncated` macro converts a python integer to an unsigned integer. and this would be incorrect because as you said the storage used in python is signed not unsigned. The reason why it works when wanting to store `-2147483648` is because the bits are the same as `2147483648` but when you want to store `-1` it is not going to work. The reason why is you are setting the value returned by `mp_obj_get_int_truncated` which evaluates into this...

```c
int32_t value = 4294967295;
```
which is an overflow.

There are several problems...
```c
#if MICROPY_OBJ_REPR == MICROPY_OBJ_REPR_A || MICROPY_OBJ_REPR == MICROPY_OBJ_REPR_C

#define MP_SMALL_INT_MIN ((mp_int_t)(((mp_int_t)MP_OBJ_WORD_MSBIT_HIGH) >> 1))
```

for the ESP32 `MICROPY_OBJ_REPR == MICROPY_OBJ_REPR_A` evaluates to `true`

```c
// Number of bytes in an object word: mp_obj_t, mp_uint_t, mp_uint_t
#ifndef MP_BYTES_PER_OBJ_WORD
#define MP_BYTES_PER_OBJ_WORD (sizeof(mp_uint_t))
#endif

// Number of bits in a byte
#ifndef MP_BITS_PER_BYTE
#define MP_BITS_PER_BYTE (8)
#endif
// mp_int_t value with most significant bit set
#define MP_OBJ_WORD_MSBIT_HIGH (((mp_uint_t)1) << (MP_BYTES_PER_OBJ_WORD * MP_BITS_PER_BYTE - 1))
```

`MP_OBJ_WORD_MSBIT_HIGH` evaluates to `-2147483648`

The problem is this ` ((mp_int_t)MP_OBJ_WORD_MSBIT_HIGH) >> 1`

That evaluates to `-1073741824` or `11000000 00000000 00000000 00000000`

and even more important is this is not correct either...
```c
#if MICROPY_OBJ_REPR == MICROPY_OBJ_REPR_A

static inline bool mp_obj_is_small_int(mp_const_obj_t o) {
    return (((mp_int_t)(o)) & 1) != 0;
}
#define MP_OBJ_SMALL_INT_VALUE(o) (((mp_int_t)(o)) >> 1)
```

```c
mp_obj_is_small_int((const mp_const_obj_t)((mp_int_t)-2147483648))
```
evaluates to `false` which is incorrect. This should be evaluating to `true`

the `MP_OBJ_SMALL_INT_VALUE` is also incorrect as well. 

```
MP_OBJ_SMALL_INT_VALUE(-2147483648)
```
evaluates as `-1073741824`

agatti, 2024-09-09:
Well, at least on an ESP3C3 `-2147483648` is not evaluated as a small int, but as an arbitrary precision number, with some limits on the amount of digits of course.  This is on the clean sourcetree with no modifications made, when evaluating `config.set_i32('testkey', -2147483648)`:

```
Thread 2 "mp_task" hit Breakpoint 1, mpz_as_int_checked (i=0x3fca8f24, value=value@entry=0x3fca7eac) at /home/agatti/src/micropython/py/mpz.c:1553
1553        mpz_dig_t *d = i->dig + i->len;
(gdb) bt
#0  mpz_as_int_checked (i=0x3fca8f24, value=value@entry=0x3fca7eac)
    at /home/agatti/src/micropython/py/mpz.c:1553
#1  0x42011c78 in mp_obj_int_get_checked (self_in=<optimized out>)
    at /home/agatti/src/micropython/py/objint_mpz.c:435
#2  0x4200ce1c in mp_obj_get_int_maybe (value=<synthetic pointer>, 
    arg=<optimized out>) at /home/agatti/src/micropython/py/obj.c:340
#3  0x42054d86 in esp32_nvs_set_i32 (self_in=0x3fca8dc0, 
    key_in=<optimized out>, value_in=0x3fca8f20)
    at /home/agatti/src/micropython/ports/esp32/esp32_nvs.c:75
#4  0x4038305c in mp_execute_bytecode (
    code_state=code_state@entry=0x3fca7f80, inject_exc=inject_exc@entry=0x0)
    at /home/agatti/src/micropython/py/vm.c:1042
#5  0x42010708 in fun_bc_call (self_in=<optimized out>, n_args=0, n_kw=0, 
    args=0x0) at /home/agatti/src/micropython/py/objfun.c:267
#6  0x4202b344 in parse_compile_execute (source=source@entry=0x3fca8070, 
    input_kind=<optimized out>, exec_flags=exec_flags@entry=22)
    at /home/agatti/src/micropython/shared/runtime/pyexec.c:119
#7  0x4202b770 in pyexec_friendly_repl ()
#8  0x420072dc in mp_task (pvParameter=<optimized out>) at /home/agatti/src/micropython/ports/esp32/main.c:161
#9  0x00000000 in ?? ()
Backtrace stopped: frame did not save the PC
```

and still triggers the overflow exception.

With my changes instead of going through `mpz_as_int_checked` it goes through `mpz_hash`, where it just picks up as many digits as it can to fill up a 32 bits integer, no exceptions raised.

Now, not being able to convert `-2147483648` from an arbitrary precision integer to a valid 32-bits signed integer does look like a bug indeed, but I think it's rather unlikely that it's been there the whole time.  Could be the product of a regression, couldn't it?

agatti, 2024-09-09:
@kdschlosser do you remember on which ESP32 board and on which ESP-IDF version you saw the issue?

On an ESP32C3 using ESP-IDF 5.0.4 and `gcc version 11.2.0 (crosstool-NG esp-2022r1)`, I maybe found some invalid code being generated that triggers the issue.

In the source for `mpz_as_int_checked` the check that triggers an exception is `if (val > (~(MP_OBJ_WORD_MSBIT_HIGH) >> DIG_SIZE)) {`, but the generated code performs a greater-or-equal comparison instead:

```
   ...
   // Load `val`
   0x4204dcac <mpz_as_int_checked+16>:  li      a5,0
   // Load `~(MP_OBJ_WORD_MSBIT_HIGH) >> DIG_SIZE`
   0x4204dcae <mpz_as_int_checked+18>:  lui     a6,0x8
   0x4204dcb0 <mpz_as_int_checked+20>:  j   0x4204dcb8 <mpz_as_int_checked+28>
   0x4204dcb2 <mpz_as_int_checked+22>:  addi    a4,a4,-2
   // Return false if a5 (val) is *greater or equal* than a6 (overflow marker)
   // This should probably have been "bltu a6, a5, 0x4204dcd4" instead?
   0x4204dcb4 <mpz_as_int_checked+24>:  bgeu    a5,a6,0x4204dcd4 <mpz_as_int_checked+56>
   // val = (val << DIG_SIZE) | *d
   0x4204dcb8 <mpz_as_int_checked+28>:  lhu     a3,0(a4)
   0x4204dcbc <mpz_as_int_checked+32>:  slli    a5,a5,0x10
   0x4204dcbe <mpz_as_int_checked+34>:  or      a5,a5,a3
   // while (*d = i->dig + i->len)
   0x4204dcc0 <mpz_as_int_checked+36>:  bltu    a2,a4,0x4204dcb2 <mpz_as_int_checked+22>
   ...
   // Return true
   0x4204dcd0 <mpz_as_int_checked+52>:  li      a0,1
   0x4204dcd2 <mpz_as_int_checked+54>:  ret
   // Return false
   0x4204dcd4 <mpz_as_int_checked+56>:  li      a0,0
   0x4204dcd6 <mpz_as_int_checked+58>:  ret
   ...
```

The first segment of `-2147483648` ends up being `0x8000`, which is also the value used for partial overflow checks when building the 32-bits integer, and you know the rest.

agatti, 2024-09-09:
As far as I can tell, smallints are 31-bits wide (from mpconfig.h on master, line 127):

```
// A MicroPython object is a machine word having the following form (called R):
//  - iiiiiiii iiiiiiii iiiiiiii iiiiiii1 small int with 31-bit signed value
...
```

hence why when attempting to store a 32-bits integer it will go through the arbitrary-precision code operate on its value.

kdschlosser, 2024-09-09:
can't store a 32 bit integer with only 31 bits..

agatti, 2024-09-10:
> can't store a 32 bit integer with only 31 bits..

Glad we've got a common understanding on that :)

I believe that you've probably missed is the argument type of `mp_obj_is_small_int`.  That function doesn't receive a raw 32 bits number but a tagged pointer instead, which also contains its type encoded in it.  In this case, if the least significant bit of the object pointer is set (which can't normally occur on most MCUs, where addresses are usually aligned on a 16-bits boundary at least) then it means the pointer value isn't an address but instead the upper 31 bits form a stand-alone integer that is up to 31 bits in length (smallint).

You're passing a number that's one bit too big to fit in a tagged pointer where a smallint could be placed in, and therefore MicroPython switches to arbitrary precision integers (all that mpz stuff) to store the number in question.

That aside, there's the case of the invalid check in the mpz code, but I'm still not sure it's a compiler issue, unless I'm missing something as well - all the versions of GCC for RV32 I have access to generate the apparently invalid check (greater-than-or-equal instead of plain greater-than).  I'll have to try on Clang later today to see if that's still the case.

kdschlosser, 2024-09-10:
I believe the compiler that is being used is the Xtensa gcc compiler.

agatti, 2024-09-10:
Indeed, but I saw that issue also on an ESP32C3, which is RV32 based.  Since I've got way more experience with RISC-V assembler than with Xtensa, I'm focusing on that platform.  Plus there isn't another Xtensa compiler besides GCC to test things against, and I can try this with Clang for RV32 later on today.

kdschlosser, 2024-09-10:
OK cool. At least I know that I am not going crazy and there is an actual problem..... I cannot tell you how long I was staring at the code trying to figure out why that error was occurring.

agatti, 2024-09-10:
This is weird but now I think it wasn't a compiler issue as I thought but more a logic error as you suggested.

I know it's a bit too late as you chose a different approach in your code but can you please try this on your MCU and see if it doesn't have issues:

```diff
diff --git a/py/mpz.c b/py/mpz.c
index 750664ad9..d755e648a 100644
--- a/py/mpz.c
+++ b/py/mpz.c
@@ -1553,7 +1553,7 @@ bool mpz_as_int_checked(const mpz_t *i, mp_int_t *value) {
     mpz_dig_t *d = i->dig + i->len;
 
     while (d-- > i->dig) {
-        if (val > (~(MP_OBJ_WORD_MSBIT_HIGH) >> DIG_SIZE)) {
+        if (val > (MP_OBJ_WORD_MSBIT_HIGH >> DIG_SIZE)) {
             // will overflow
             return false;
         }
@@ -1578,7 +1578,7 @@ bool mpz_as_uint_checked(const mpz_t *i, mp_uint_t *value) {
     mpz_dig_t *d = i->dig + i->len;
 
     while (d-- > i->dig) {
-        if (val > (~(MP_OBJ_WORD_MSBIT_HIGH) >> (DIG_SIZE - 1))) {
+        if (val > (MP_OBJ_WORD_MSBIT_HIGH >> (DIG_SIZE - 1))) {
             // will overflow
             return false;
         }
```

I've tested a few edge cases within the -2<sup>31</sup> ≤ x < 2<sup>31</sup> range and they seem to work, but now my ESP32C3 is busy running this just to be sure:

```python
>>> a = -2147483648
>>> while a <= 2147483647:
...     config.set_i32('testkey', a)
...     b = config.get_i32('testkey')
...     if a != b:
...         raise Exception(a, b)
...     if a % 0x1000 == 0:
...         print(".", end='')
...     a += 1
```

If that's successful I'll see how it behaves on numbers outside the `int32_t` range and if it looks ok I'll send it off as a PR.

Still, this never surfaced until this May it'd be truly wild.

agatti, 2024-09-10:
Scratch that, not sure this can be easily fixed in the first place.  Maybe a specific issue should be submitted about `mpz_as_int_checked` (and possibly about `mpz_as_uint_checked` too) not handling your case rather than ESP32's NVS behaviour, as this issue applies on any port and not just on ESP32.

kdschlosser, 2024-09-10:
at the time I opened this bug report the only thing I had to work from is it was happening when I was trying to set a value to the NVS.

closed 2025-03-03 by kdschlosser