micropython/micropython #14492: Does not build for ESP32-S3 with IDF 5.2 despite claiming support

vladkvit, 2024-05-15:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

ESP32-S3

### MicroPython version

v1.22.2

### Reproduction

I'm building Micropython for an ESP32. It works fine with micropython v1.22.2 and IDF v5.1.2. It does not work with IDF v5.2, despite claimed support here: https://github.com/micropython/micropython/tree/master/ports/esp32/#setting-up-esp-idf-and-the-build-environment

Here's my Dockerfile to reproduce:

```
FROM ubuntu:22.04

ARG DEBIAN_FRONTEND=noninteractive

# Set a working directory
WORKDIR /code

# Install system dependencies
RUN apt-get update && apt-get install -y \
    build-essential \
    libffi-dev \
    git \
    pkg-config \
    python3.10-venv \
    libusb-1.0-0-dev \
    cmake \
    && rm -rf /var/lib/apt/lists/*

# Clone the micropython repository
RUN git clone https://github.com/micropython/micropython.git
WORKDIR /code/micropython
RUN git checkout -b v1.22.2 v1.22.2

# Clone ESP-IDF
WORKDIR /code
RUN git clone -b v5.2 --recursive https://github.com/espressif/esp-idf.git
WORKDIR /code/esp-idf
RUN git checkout v5.2
RUN git submodule update --init --recursive

SHELL ["/bin/bash", "-c"] 

RUN ./install.sh

# Build MicroPython
WORKDIR /code/micropython/ports/esp32

RUN . /code/esp-idf/export.sh && \
    make BOARD=ESP32_GENERIC_S3 submodules && \
    make BOARD=ESP32_GENERIC_S3

ENTRYPOINT ["/bin/bash"]

```

Replace 'v5.2" with "v5.1.2" to have it run successfully.

The error is here:

```
32.51 /code/micropython/ports/esp32/network_common.c:172:1: error: static assertion failed: "Synchronize WIFI_AUTH_XXX constants with the ESP-IDF. Look at esp-idf/components/esp_wifi/include/esp_wifi_types.h"
32.51   172 | _Static_assert(WIFI_AUTH_MAX == 11, "Synchronize WIFI_AUTH_XXX constants with the ESP-IDF. Look at esp-idf/components/esp_wifi/include/esp_wifi_types.h");   
32.51       | ^~~~~~~~~~~~~~             
```                                                                                                                                           

### Expected behaviour

Expected a finished build

### Observed behaviour

Build errors out due to a static assert.

### Additional Information

No, I've provided everything above.

ricksorensen, 2024-05-15:
I believe the 1.22.2 does not support IDF 5.2 -- the docs default to the latest/development.  This version of the documents for esp32 is from 7 Mar 2024 with the 5.2 reference which is after 1.22.2

Try building the latest development version with IDF 5.2

vladkvit, 2024-05-15:
Thank you. Checking out latest and using IDF v5.2 works.

closed 2024-05-15 by vladkvit

mattytrentini, 2024-05-16:
Just an additional note - espressif [publish docker containers](https://hub.docker.com/r/espressif/idf), there's no need to build one yourself. Also, you can avoid building your container with your project code inside it; the canonical way to do this is to clone your project to your host and volume-mount your code into your container. The container should really only contain your build environment.