micropython/micropython #14498: get_x_aligned returns wrong value in case of percentage unit

leidto, 2024-05-17:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

emsdk

### MicroPython version

MicroPython v1.20.0-1052-gca138ed90

### Reproduction

import display_driver
import lvgl as lv

scr = lv.obj()
btn = lv.button(scr)
btn.align(lv.ALIGN.CENTER, 0, 0)

btn.set_x(lv.pct(10))

label = lv.label(btn)
label.set_text('Hello World!')

lv.screen_load(scr)

print('btn x value:', btn.get_x_aligned())

### Expected behaviour

btn.get_x_aligned() returns the correct x value

### Observed behaviour

btn.get_x_aligned() returns incorrect x value, for example: 536870922

### Additional Information

No, I've provided everything above.

robert-hh, 2024-05-17:
Are you sure that you posted your issue in the rigth repository? It seems to be a problem with lvgl. https://github.com/lvgl/

leidto, 2024-05-17:
Wrong repo.

closed 2024-05-17 by leidto