micropython/micropython #14536: docs: Windows port README should include `make submodule` as a build step

mattytrentini, 2024-05-19:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) and didn't find any that matched.

### Documentation URL

https://github.com/micropython/micropython/blob/master/ports/windows/README.md

### Description

The build steps for the Windows port are missing `make submodule`. Build errors occur because micropython-lib (specifically the ssl module) isn't found.

PolarGoose, 2024-07-06:
Indeed, if I execute `msbuild ports/windows/micropython.vcxproj /p:PyVariant=standard` I get the following error:
```
FreezeModules:
  python C:\my\gitrepos\micropython-1.23.0\tools\makemanifest.py -v MPY_DIR=C:\my\gitrepos\micropython-1.23.0\ -v MPY_LIB_DIR=C:\my\gitrepos\micropython-1.23.0\/lib/micropython-lib -v PORT_DIR=C:\my\gitrepos\micropython-1.23.0\ports\wi
  ndows\ -o C:\my\gitrepos\micropython-1.23.0\ports\windows\build-standard\frozen_content.c -b C:\my\gitrepos\micropython-1.23.0\ports\windows\build-standard\ C:\my\gitrepos\micropython-1.23.0\ports\windows\\variants\manifest.py
  freeze error executing "C:\my\gitrepos\micropython-1.23.0\ports\windows\\variants\manifest.py": Error in manifest file: C:\my\gitrepos\micropython-1.23.0\ports\windows\variants\manifest.py: Package 'ssl' not found in any known librar
  y.
C:\my\gitrepos\micropython-1.23.0\ports\windows\msvc\genhdr.targets(175,5): error MSB3073: The command "python C:\my\gitrepos\micropython-1.23.0\tools\makemanifest.py -v MPY_DIR=C:\my\gitrepos\micropython-1.23.0\ -v MPY_LIB_DIR=C:\my\g
itrepos\micropython-1.23.0\/lib/micropython-lib -v PORT_DIR=C:\my\gitrepos\micropython-1.23.0\ports\windows\ -o C:\my\gitrepos\micropython-1.23.0\ports\windows\build-standard\frozen_content.c -b C:\my\gitrepos\micropython-1.23.0\ports\
windows\build-standard\ C:\my\gitrepos\micropython-1.23.0\ports\windows\\variants\manifest.py" exited with code 1. [C:\my\gitrepos\micropython-1.23.0\ports\windows\micropython.vcxproj]
Done Building Project "C:\my\gitrepos\micropython-1.23.0\ports\windows\micropython.vcxproj" (default targets) -- FAILED.
```

MSVC toolchain doesn't contain `make`. How can I do `make submodule` on Windows?

stinos, 2024-07-06:
You need to run `git submodule update --init lib/micropython-lib` or similar

I have a commit in this PR which adds support for doing this automatically when building in VS and adds an msbuild target: https://github.com/micropython/micropython/pull/12810/commits

 Maybe that commit should be split off in a separate PR?

PolarGoose, 2024-07-06:
@stinos,

Thank you for the hint. Yes, with this command, the build works.
I have created a repository that creates a Windows binary: [MicroPython-for-Windows](https://github.com/PolarGoose/MicroPython-for-Windows)