micropython/micropython #15098: mpremote mount fails reading binary file

peterhinch, 2024-05-22:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

RP2, Pyboard 1.1

### MicroPython version

MicroPython v1.22.0 on 2023-12-27; Raspberry Pi Pico with RP2040

### Reproduction

Create a file `rats15.py` on the PC:
```py
import os
fn = "delete_me"
with open(fn, "wb") as f:
    f.write(b"hello\n\xde\xad\xbe\xef")
with open(fn, "rb") as f:
    print(f.readline())
    print(f.read(4))
os.unlink(fn)
```
Run `mpremote`
```bash
$ mpremote mount .
```
At the REPL issue
```py
import rats15
```

### Expected behaviour

```py
>>> import rats15
b'hello\n'
b'\xde\xad\xbe\xef'
>>> 
```
This occurs if the script is run under CPython, under the Unix build, or if run locally on a MP target.

### Observed behaviour

When run as described above via `mpremote mount .`:
```
MicroPython v1.22.2 on 2024-02-22; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>> import rats15
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "rats15.py", line 7, in <module>
  File "<stdin>", line 147, in readline
TypeError: unsupported types for __add__: 'str', 'bytes'
>>> 
```

### Additional Information

mpremote is V1.22.0. The fault occurs on `readline()`. It originally became evident accessing a pgm graphics file which contains four lines of \n terminated ASCII text followed by binary data

dpgeorge, 2024-05-22:
Thanks for the report.  It looks like `mpremote` can only handle a `readline()` call on a file opened in text mode, ie not bytes mode.

CPython allows it (as you've shown), so we need to improve `mpremote` so it also supports reading lines from binary files.

cwalther, 2024-05-22:
Possible fix suggested in https://github.com/orgs/micropython/discussions/15091#discussioncomment-9511516.