micropython/micropython #15115: const declatarion: real constant, not accepting other assignments

massimosala, 2024-05-25:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) regarding this feature, and didn't find any.

### Description

Example:
```
A = const(5)
print(A)
A = 10
print(A)
```

If I understand well, now **const** is only a hint to the compiler to use the immediate value (in this case, 5), without referencing the variable A.
The compiled code is equivalent to
```
print(5)
A = 10
print(A)
```

My suggestion: an assignment using **const** declares an immutable object.
So the compiler will abort on
```
A = 10
```

It seems to me intuitive to enforce the **const** semantic, aligned with other languages and the traditional thinking about **const** effects.
I don't see rational cases to declare a variable **const** and later change its value.
It is a pain to review source code using this bad habit.

### Implementation

- [ ] I intend to implement this feature and would submit a Pull Request if desirable.
- [X] I hope the MicroPython maintainers or community will implement this feature.
- [X] I would like to [Sponsor](https://github.com/sponsors/micropython#sponsors) development of this feature.

andrewleech, 2024-05-25:
It's true that const is just used by the compiler to replace any usages of the variable with the literal value. 

It's used as a size reduction rather than code enforcement. 

This confused me the first few times I used it!

My initial thought was that any kind of effort to make a variable name (eg `A` above) immutable would possibly be a breach of python language rules, but in reality classes can have immutable attributes if you override setattr and disallow updates! Why not modules too.
I'm not sure how much effort it would be to detect and enforce this in the compiler though, I haven't worked on that codebase.

peterhinch, 2024-05-26:
> classes can have immutable attributes if you override setattr and disallow updates

Or define a property with no setter. But (as far as I know) you can't do this at module level, so I'm not sure that subverting Python language rules is a good idea.

massimosala, 2024-05-26:
I appreciate your feedback.

Andrew wrote:

> This confused me the first few times I used it!

Yes, it confuses every programmers, both novice and expert ones !

MP already broke Python compatibility, with the introduction of the statement

> from micropython import const

There isn't any _const_ inside _micropython_, it is just a hint for the compiler.

---

Regarding the effort to make _const_ immutable: I think it is simple.
Every time the compiler parses a statement
`varname = const(...)
`
it adds varname to a list of immutables names. In python code:
`IMMUTABLE_VARS.append(varname)
`

Every time the compiler parses a statement
`varname = ...
`
if _varname_ is present in the list, the compiler aborts with a specific error.

Furthermore, declaring objects as immutable can lead to further optimizations.

massimosala, 2024-05-26:
Ah, I am thinking to enforce _const_ semantic only for primitive types like: int, str, bytes.
IMHO this covers the majority of usage cases.

If, for any reason, it is desiderable to preserve the actual _const_ semantic, we can use two different tags:

- rename _const_ as _literal_ (or something better)

- new _const_ to declare immutable ones.

dpgeorge, 2024-05-27:
I think the confusion here is the use of `const` at the REPL versus in a file (or in paste mode).

At the REPL each line is executed independently and in particular the `const` variables are not remembered between lines.

If you use paste mode at the REPL (ctrl-E) then `const` will work as expected because in paste mode all lines are executed in the same context.  It also works when executed as part of a file.

See previous discussion here: https://github.com/micropython/micropython/issues/3232

massimosala, 2024-05-27:
Hi Damien. Thanks for your reply.

I read #3232 but I don't understand: I am not talkin about differences between REPL or modules, but to enforce the _const_ semantic to declare immutable objects.

From a perspective  of language geatures IMHO is an important one and I am ready to sponsor its development.

dpgeorge, 2024-05-27:
> I am not talkin about differences between REPL or modules, but to enforce the _const_ semantic to declare immutable objects.

The `const()` feature in MicroPython already declares read-only variables.

For example the following code:
```py
from micropython import const

A = const(5)
print(A)
A = 10
print(A)
```
If you run that in MicroPython (eg the unix port, or via `mpremote run <file.py>` on a board, or pasting using ctrl-E paste mode) you get:
```
Traceback (most recent call last):
  File "test.py", line 5, in <module>
SyntaxError: can't assign to expression
```

peterhinch, 2024-05-28:
@dpgeorge There are limits to this, though. Unlike a tuple, constancy is only enforced within a module. If you create a file `rats16.py` containing
```py
a = const(42)
```
and then import it, the value of `rats16.a` can be changed:
```py
>>> import rats16
>>> rats16.a
42
>>> rats16.a = 99
>>> rats16.a
99
>>> 
```
I'm not suggesting this is a bug: merely something of which people need to be aware.

dpgeorge, 2024-05-28:
@peterhinch yes you are correct, a global/module-level variable is still created for the "constant".

You can instead prefix your constant variable names with underscore, eg `_a = const(42)` and then they are not visible outside the module (they are not even included in the bytecode, they are fully optimised out).

fojie, 2024-05-29:
@dpgeorge Is it necessary to use "from machine import const" for this optimisation to occur?

I ask because you have it in your example in previous mail, but i have noticed others saying it not necessary to have the import statement, and I have started to remove them in my code.

Since I have projects with very limited RAM, and therefore use const() a lot, I am interested to know.

dpgeorge, 2024-05-29:
> Is it necessary to use "from machine import const" for this optimisation to occur?

No.  It is not strictly necessary to have that line.  If `const` optimisation support is built in to the compiler on your port (it is on almost all ports, and also `mpy-cross` on the desktop) then it will work regardless of that import line.

But, if you want to make linters happy, and run the same code under CPython, you need that line.  If you add that line then it does incur a small cost at runtime to load the `const` function from the `micropython` module and store it in the dict of that module.

In summary: if you're really trying to optimise down to the last byte of RAM, then exclude this import.  Otherwise I recommend including it.

massimosala, 2024-05-30:
@dpgeorge
_You can instead prefix your constant variable names with underscore, eg _a = const(42) and then they are not visible outside the module (they are not even included in the bytecode, they are fully optimised out)._

I think we should not mix visibility with immutability.

Like written by @peterhinch, there are cases where the constant declared in a module
- are immutable
- should by visible outside the module (and immutable)

dpgeorge, 2024-05-31:
@massimosala can you please state the issue clearly, what you see as the problem / possible improvement.

From your original report:

> My suggestion: an assignment using const declares an immutable object.
> So the compiler will abort on
>
> A = 10

This is currently the case, the compiler raises `SyntaxError` when trying to assign to the const-declared `A`.

So in my view this issue can be closed, for that particular point.

massimosala, 2024-05-31:
@dpgeorge 

>>> import rats16
>>> rats16.a
42
>>> rats16.a = 99
>>> rats16.a
99
>>>

closed 2024-06-10 by massimosala