micropython/micropython #15116: implement os.sep

massimosala, 2024-05-25:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) regarding this feature, and didn't find any.

### Description

To facilitate the reuse of libraries written for Python and to test Micropython libraries on regular Python,
I ask to implement in the module **os** the attribute **sep**

I am not sure about all ports and filesystems... but perhaps it can be just implemented as
`
sep = '/'
`
for all.

### Implementation

- [ ] I intend to implement this feature and would submit a Pull Request if desirable.
- [X] I hope the MicroPython maintainers or community will implement this feature.
- [X] I would like to [Sponsor](https://github.com/sponsors/micropython#sponsors) development of this feature.

dpgeorge, 2024-05-27:
This is actually implemented already, it's just not enabled on all ports.  It can be enabled via the compile-time option `MICROPY_PY_OS_SEP`.

Probably it should be enabled more consistently across ports that have the "full features" enabled.

massimosala, 2024-05-27:
Hi Damien

IMHO for such a basic and minimal feature, it is premature optimization to have it conditional enabled.

Developing on different boards it a pain to remember where _os.sep_ is available or not.
And recompiling the firmware just to enable it... mah...

I see it's optional availability more as a burden than a gain (how many bytes saved?).

Proposal: enable _os.sep_ on all firmwares.

jonnor, 2024-07-18:
All ports with a virtual file systems (VFS) might be a good choice? If one does not have a VFS, then os.sep seems less likely to be useful.

dpgeorge, 2024-07-19:
See #15497 for a solution to this issue.

dpgeorge, 2024-07-20:
Fixed by 8159dcc276dc07bcb28d6ab3a3b7d2a5c1d90ab5

closed 2024-07-20 by dpgeorge