micropython/micropython #15117: os.rmdir() delete files

massimosala, 2024-05-25:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

rp2, esp32, esp8266

### MicroPython version

Micropython 1.22.2

### Reproduction

open("prova.tmp", 'w').write('1')
1

import os
os.rmdir("prova.tmp")

open("prova.tmp", 'r')

### Expected behaviour

rmdir() should raise OSError

### Observed behaviour

rmdir() succeeds and the file is deleted.

`open("prova.tmp", 'r')
`
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 2] ENOENT

### Additional Information

No, I've provided everything above.

massimosala, 2024-05-26:
Related to this commit ?
https://github.com/micropython/micropython/commit/d02f3a57f47b44bf31769fb089f1f18060ea3b4d

Is it pending or merged in the official MP release ?

dpgeorge, 2024-05-27:
Thanks for the report.

This issue was fixed for FAT filesystems, but you probably tested a littlefs filesystem, for which `os.remove` and `os.rmdir` are the same thing.

massimosala, 2024-05-27:
Hi Damien

Looking for _rmdir_ and litlefs, I found this file:
[micropython](https://github.com/micropython/micropython/tree/8a8c65f309a39e99b3bd5a74c5da313d8a5ec5fd)/[extmod](https://github.com/micropython/micropython/tree/8a8c65f309a39e99b3bd5a74c5da313d8a5ec5fd/extmod)
/vfs_lfsx.c

Is it the right file?

I can try to fix it; is it sufficient I follow the changes made in  this commit ?
https://github.com/micropython/micropython/commit/d02f3a57f47b44bf31769fb089f1f18060ea3b4d