micropython/micropython #15118: network.hostname: max length of hostname

massimosala, 2024-05-25:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) regarding this feature, and didn't find any.

### Description

Form the docs: the max length depends on the port (and its tcp/ip stack).

So, without spelunking into each network low-level libraries or make tests on each specific board, we don't know the limit.

Proposal: calling hostname(None) returns the accepted max length of hostname.

### Implementation

- [ ] I intend to implement this feature and would submit a Pull Request if desirable.
- [X] I hope the MicroPython maintainers or community will implement this feature.
- [X] I would like to [Sponsor](https://github.com/sponsors/micropython#sponsors) development of this feature.

massimosala, 2024-05-27:
Here is the implementation: just two lines to add to the function _mod_network_hostname_ in in _modnetwork.c_

```
mp_obj_t mod_network_hostname(size_t n_args, const mp_obj_t *args) {
    if (n_args == 0) {
        return mp_obj_new_str(mod_network_hostname_data, strlen(mod_network_hostname_data));
    } else {
++		if (args[0] == mp_const_none)
++               return MP_OBJ_NEW_SMALL_INT(MICROPY_PY_NETWORK_HOSTNAME_MAX_LEN);

        size_t len;
        const char *str = mp_obj_str_get_data(args[0], &len);

        if (len > MICROPY_PY_NETWORK_HOSTNAME_MAX_LEN) {
            mp_raise_ValueError(NULL);
        }
        memcpy(mod_network_hostname_data, str, len);
        mod_network_hostname_data[len] = 0;
        return mp_const_none;
    }
}
```

---

Update the [doc](https://docs.micropython.org/en/latest/library/network.html)

> network.hostname([name])
> Get or set the hostname that will identify this device on the network. It will be used by all interfaces.
> 
> **If _name_ is None, this function return the maximum length admitted for the hostname.**

---

Please, can some expert tell me how publish these two changes (source code and doc) ?
I don't know github, so I will start from scratch, be patient.

Gadgetoid, 2024-06-06:
> Proposal: calling hostname(None) returns the accepted max length of hostname.

I'm going to go out on a limb and suggest this would violate the principle of least surprise.

Or, to put it simply; it's too *weird.*

Possibly a better approach would be to do the aforementioned spelunking and simply document the maximum lengths accepted by common ports/stack combos.

Doing some basic spelunking (I'm lazy so grep does it for me) suggests that it's always 32, and that no ports override it:

```
grep -r MICROPY_PY_NETWORK_HOSTNAME_MAX_LEN
./extmod/modnetwork.h:#ifndef MICROPY_PY_NETWORK_HOSTNAME_MAX_LEN
./extmod/modnetwork.h:#define MICROPY_PY_NETWORK_HOSTNAME_MAX_LEN (32)
./extmod/modnetwork.h:extern char mod_network_hostname_data[MICROPY_PY_NETWORK_HOSTNAME_MAX_LEN + 1];
./extmod/modnetwork.c:char mod_network_hostname_data[MICROPY_PY_NETWORK_HOSTNAME_MAX_LEN + 1] = MICROPY_PY_NETWORK_HOSTNAME_DEFAULT;
./extmod/modnetwork.c:        if (len > MICROPY_PY_NETWORK_HOSTNAME_MAX_LEN) {
```

But I guess it's *possible* that `MICROPY_PY_NETWORK_HOSTNAME_MAX_LEN` is either larger or smaller than some underlying TCP/IP stack implementation which the hostname is passed to... making returning that value (always 32) useless anyway.

massimosala, 2024-06-10:
Hi @Gadgetoid Philip

Scenario: deploy at a customer site a batch of MP boards.
Every board has a serial number, 4 digits (written on the file system by the board manufacturer).
The board at start-up initializes the network (wired or Wifi) and sets its hostname.

IMHO this is the most elegant solution:

```
model = "X001 My custom board with xxx yyy zzz"
...
max_len = hostname(None)
if max_len < 4 + 1 + 4:
    raise RuntimeError("Cannot set hostname")
i = max_len - 5
hostname( model[: i] + '-' + serial)
```

1) It is really a pain to search in the **C sources** the defines

2) Some of my colleagues are fluent in Python but ignorant in C.

3) At any time the defines could change values, so hard-coding their values in Python is a big fallacy.

Gadgetoid, 2024-06-10:
In your case the most _Pythonic_ way to handle this case would be for hostname to raise a ValueError if the name exceeds the maximum length. Getting the length could just be a board constant? or something else that doesn’t require overloading hostname in a non intuitive way.

mattytrentini, 2024-06-14:
>     * hostname(None) returns the accepted max length

No, thanks. Pragmatic, perhaps, but extremely _ugly_. 

> In my past 30 years as programmers, I see a wide-spread usage of the convention _return the required 'size' then the API is called with a NULL parameter_ especially in the C world. At this level of simplicity, the API isn't weird : it is only necessary to document its interface.

...and it is an ugly pattern in the C domain too IMHO. 😜 

I do support the desire to be able to query the max length - either by querying a constant or calling a function for the max length. Or, perhaps, even solely by documentation. In any case an error should be raised if the max length is exceeded.

massimosala, 2024-06-18:
Ciao Matt

@mattytrentini 
> No, thanks. Pragmatic, perhaps, but extremely ugly.

Ok, I get it.

Now, what do I need to do to migrate this to a PR?
I don't know if I have the rights / how to do it.