micropython/micropython #15120: RP2: Hard I2C not working

massimosala, 2024-05-25:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

rp2

### MicroPython version

MicroPython v1.22.2

### Reproduction

from machine import Pin, I2C
from ssd1306 import SSD1306_I2C
bus = I2C(id = 0, scl = Pin(9), sda = Pin(8), freq = 100_000)
display = SSD1306_I2C(width, height, bus)

### Expected behaviour

ssd1306 working

### Observed behaviour

Failure with errors EIO / ETIMEDOUT.

Tested with id 0 and 1, on different pis.
Works like a charm using SoftI2C.

### Additional Information

Could be related to this ticket ?
RP2: Hard I2C is intolerant of clock stretching
https://github.com/micropython/micropython/issues/8167

peterhinch, 2024-05-26:
I have used the following with two different designs of I2C hardware:
```py
i2c = I2C(1, scl=Pin(27), sda=Pin(26), freq=100_000)
```
You might want to try that setup and pins and run `I2C.scan()`.

massimosala, 2024-05-26:
Hi Peter.

As per your request, I make a new test.
Setup: official Raspberry rp2 and generic ssd1306 display. No other components.

I used these displays from over a year, they have fixed i2c addr 0x3c.
The consume very little, so I wired their Vdd to rp2 pin 36 (3.3 Vout).

Testing I2C:
`bus = I2C(1, sda = Pin(26), scl = Pin(27), freq = 100_000)
`

It fails:
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "web_display_ex.py", line 302, in <module>
>   File "ssd1306.py", line 1, in __init__
>   File "ssd1306.py", line 1, in __init__
>   File "ssd1306.py", line 1, in init_display
>   File "ssd1306.py", line 1, in show
>   File "ssd1306.py", line 1, in write_data
> OSError: [Errno 110] ETIMEDOUT

Testing Soft i2c:
`bus = SoftI2C(sda = Pin(26), scl = Pin(27), freq = 100_000)
`

No errors, the display works.

Peter, are you trying with the same configuration: rp2 with ssd1306 display?

Perhaps the _ssd1306_ is a bit finicky... but on esp8266 and esp32 the I2C works fine.

Flipje1955, 2024-05-27:
use various brands rp2's with ssd1306 and have no issues using "hard" fast i2c but have pullups on the i2c pins. U2 ?

massimosala, 2024-05-30:
> use various brands rp2's with ssd1306 and have no issues using "hard" fast i2c but have 4k7 pullups on the i2c pins. U2 ?

Certainly I have pullups, otherwise SoftI2C doesn't work too.
I use 5.1k resistors.

robert-hh, 2025-02-02:
About scan and RP2: scan() just sends the address. In that case there is no data and SoftI2C is used, even when the Hard I2C API is called.