micropython/micropython #15122: docs: incorrect link on the web page Wiznet W5500-EVB-Pico

massimosala, 2024-05-25:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) and didn't find any that matched.

### Documentation URL

https://micropython.org/download/W5500_EVB_PICO/

### Description

Wrong link:
More info: [Website](https://www.wiznet.io/product-item/w5500-evb-pico/)

Correct one:
https://docs.wiznet.io/Product/iEthernet/W5500/w5500-evb-pico

mattytrentini, 2024-05-26:
Thanks for reporting this, it should be resolved by #15125.

massimosala, 2024-05-26:
> Thanks for reporting this, it should be resolved by #15125.

Hi Matty
Great!

Pls help: I incorrectly assigned BUG to the issue 15119. How can I change it to ENH ?

closed 2024-05-26 by massimosala

mattytrentini, 2024-05-26:
> Pls help: I incorrectly assigned BUG to the issue 15119. How can I change it to ENH ?

Removed. :)