micropython/micropython #15127: wiznet: spi baudrate declared as 2 and 20 MHz ?!

massimosala, 2024-05-26:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

rp2

### MicroPython version

Micropython 1.22.2

### Reproduction

[network_wiznet5k.c](https://github.com/micropython/micropython/blob/master/extmod/network_wiznet5k.c)
`#define MICROPY_HW_WIZNET_SPI_BAUDRATE  (2000000)
`
[mpconfigboard.h](https://github.com/micropython/micropython/blob/master/ports/rp2/boards/W5500_EVB_PICO/mpconfigboard.h)
`#define MICROPY_HW_WIZNET_SPI_BAUDRATE      (20 * 1000 * 1000)
`

### Expected behaviour

Which is the right speed?
Should I try with 20 MHz ?

### Observed behaviour

Initializing the SPI bus at 2 MHz works.

### Additional Information

No, I've provided everything above.

robert-hh, 2024-05-26:
As far as I understand it, both settings are fine. The setting in network_wiznet5k.c is the default value, which can be overridden by a board definition. This approach of default + specific defines is used at many place in the code. By itself this is not a bug.

massimosala, 2024-05-26:
Hi Robert

Thanks for your fast reply.

I see. However the online examples doesn't help regular Python users to understand this topic.

The [micropython official doc](https://docs.micropython.org/en/latest/library/network.WIZNET5K.html)
says

> class WIZNET5K ... allows you to control WIZnet5x00 Ethernet adaptors based on the W5200 and W5500 chipsets

Unfortunately there is only one example... referring to the Pyboard:
```
import network
nic = network.WIZNET5K(pyb.SPI(1), pyb.Pin.board.X5, pyb.Pin.board.X4)
```

---

I am using the
[Wiznet W5500-EVB-Pico](https://micropython.org/download/W5500_EVB_PICO/)
firmware with a regular rp2 (non WiFi) board, wired to an external W5500 module.

The rp2 and the w5500 are mounted on a breadboard, with plain wires to connect the signals.
I tested up to 20 MHz and it works (the program on the rp2 uses up to 3 tcp connections; running over 1 hour without problems).

---

I suggest to improve the documentation on these web pages:

[Firmware download](https://micropython.org/download/W5500_EVB_PICO/)

This firmware isn't limited to Wiznet W5500-EVB-Pico boards, it works also with rp2 plus external W5500 module.

[Wiznet W5500-EVB-Pico](https://micropython.org/download/W5500_EVB_PICO/)

To use a Raspberry Pico with an external Wiznet 5500 module, you can flash the
[W5500 EVB PICO](https://micropython.org/download/W5500_EVB_PICO/) firmware.

Example to initialize the network card:

```
from machine import Pin, SPI
from network import WIZNET5K, hostname
spi = SPI(0, spi_mhz * 1_000_000, mosi = Pin(19), miso = Pin(16), sck = Pin(18))
nic = WIZNET5K(spi, Pin(17), Pin(20))
```

Recommendations:
The speed of the spi bus wired to the W5500 can be in the range 2 .. 20 MHz.
The maximum speed can be used with the chips soldered on a PCB and a good routing of the signal traces.
Using a breadboard, it is safer to stay under 8 MHz.

massimosala, 2024-05-26:
Robert, can you pls add the tag "Documentation" to this issue? I cannot see how to do.

jonnor, 2024-07-13:
Improvements to the documentation would be much appreciated!