micropython/micropython #15129: if uart receives 0xf0 byte then mcu after entering lightsleep never wakes up

f1andrew, 2024-05-26:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

RP2, STM32L4, STM32H7

### MicroPython version

MicroPython v1.22.2 on 2024-02-22; Raspberry Pi Pico with RP2040

### Reproduction

Example for Rpi Pico:
```
from machine import Pin, UART, lightsleep
from time import sleep

uart = UART(0, baudrate=115200, bits=8, tx=Pin(0), rx=Pin(1), parity=None, stop = 1, timeout=2000)

## Main Loop ##
uart.write("Start \n")

while True:
    sleep(1)
    buff=bytearray()
    while uart.any():
        buff.extend(bytearray(uart.read(uart.any())))
    # to exit program send '9'
    if buff.find(b"9")>=0:
        break 
    
    if len(buff)>0:
        buff.append(10)
        uart.write(buff)
    uart.write("before lightsleep \n")
    lightsleep(100)
    sleep(0.05)
    uart.write("after lightsleep\n")
```

### Observed behaviour

If uart receives 0xf0 byte then mcu after entering lightsleep never wakes up.
If message doesn't have 0xf0 byte then it continues working ok.

I tried it on RPi Pico, and STM32 boards they have same behavior.

### Additional Information

No, I've provided everything above.

robert-hh, 2024-05-26:
That's interesting. A few more observations testing with a RP2 board.
- dropping lightsleep it works.
- when b"\f0" was sent, the code at the receiver side continues, echoes data and stops after receiving b'9', but it does not return to REPL.

Since you have seen that on two very different ports it may not be related directly to lightsleep, but just something that affects lightsleep. 

If I replace `buff.append()` by buffer addition, it seems to work:
```py
from time import sleep

uart = UART(0, baudrate=115200, bits=8, tx=Pin(0), rx=Pin(1), parity=None, stop = 1, timeout=2000)

## Main Loop ##
uart.write("Start \n")

while True:
    sleep(1)
    buff=b''
    while uart.any():
        buff += uart.read(uart.any())
    # to exit program send '9'
    if buff.find(b"9")>=0:
        break

    if len(buff)>0:
        buff += b'\n'
        uart.write(buff)
    uart.write("before lightsleep \n")
    lightsleep(100)
    sleep(0.2)
    uart.write("after lightsleep\n")
```

f1andrew, 2024-05-27:
Thanks for quick answer.
Unfortunately, it still hangs.
Only difference that now it can work first time but hangs after second sending of 0xf0 byte.
I tried it on Rpi Pico and STM32.

robert-hh, 2024-05-27:
How do you send the bytes?

f1andrew, 2024-05-27:
I used Modbus terminal.
Now I tried it with Realterm terminal:
- Rpi Pico works ok.
- STM32L4 always hangs after 0xf0, even if uart data read without assignment to any variable.

f1andrew, 2024-05-27:
Also, when this happen mcu not going into lightsleep correctly, power consumption remains high.